Koenkk/zigbee2mqtt #2850: Feature: bulb on/off transition

Koenkk, 2020-01-29:
Currently the device specific `transition` option allows to control the bulb brightness, colour temperature and colour transitions. 

With the latest Zigbee2mqtt 1.9.0-dev this will also be applied the on/off state.

Documentation: https://www.zigbee2mqtt.io/devices/9290011370.html#device-type-specific-configuration

TheDodger, 2020-01-30:
What to do with bulbs transitioning to `brightness: 1`?

This happens at least to innr RB214 and Mueller Light (some E14 kind). When switching back on they are basically off, but report ON.

Koenkk, 2020-01-30:
When sending `{"state": "OFF", "brightness": 1}` the brightness should be ignored and the bulb should turn off.

TheDodger, 2020-01-30:
I'm only sending `{"state": "OFF"}`. The bulb then transitions to `brightness: 1`.

Koenkk, 2020-01-30:
Is reporting enabled?

TheDodger, 2020-01-30:
Yes.

Koenkk, 2020-01-31:
Should be fixed in latest dev.

TheDodger, 2020-01-31:
Thanks a lot, I'll give it a try when I'm back home later.

Underknowledge, 2020-02-01:
Had hopes that it fixes my transition issue with the [Innr RB](https://www.zigbee2mqtt.io/devices/RB_175_W.html) but that thing will sill just flash on to the set brightness when it isn't on already. It could also be  that the convertor is pissed that home assistant doesn't send a real int. that is not obvious yet. example mqtt message:
```
lights/innr/set {"state": "ON", "transition": 10.0, "brightness": 117}
``` 
but it seems working well with other lights

groenmarsmannetje, 2020-02-05:
I recently switched to dev version and notice that some lights are acting strange.?it might be related to this change. What I see is that some IKEA lights might stay on in a very dimmed state (brightness 1?) when they are turned off. Or these are very dimmed when you turn them on again while they were much brighter when you turned them off. I also have seen lights suddenly turn on and off unexpected. I think that could be triggered by a turn-off command while they were already turned off or maybe had brightness 0.

Koenkk, 2020-02-05:
@groenmarsmannetje do you use the `transition` option for these devices in `configuration.yaml`?

groenmarsmannetje, 2020-02-05:
Yes I do. It is set to 6 seconds

Koenkk, 2020-02-05:
@groenmarsmannetje did you enable the reporting feature?

groenmarsmannetje, 2020-02-05:
Yes I did

Koenkk, 2020-02-05:
thanks, will look into it.

Koenkk, 2020-02-07:
@groenmarsmannetje I cannot reproduce this on zigbee2mqtt 1.10.0, can you? If so please provide the logging when running with debug logging enabled.

groenmarsmannetje, 2020-02-08:
Okay I will install new master too and check. I already removed transition for the bulbs that acted weird. And by doing that the experience was much better. Maybe it is also related to routing or timeouts because all lights that had this strange behavior were lights that are most distant from the coordinator. I will check again with 1.10.0 and let you know.

groenmarsmannetje, 2020-02-09:
I have not been able yet to manually reproduce this with version 1.10.0 but I still experience these states after a while. Will keep you updated when I know more.

TheDodger, 2020-02-09:
I'm still experiencing problems when setting multiple values at once (on, brightness, color_temp).

Wasn't able to make up some test cases as I'm somewhat ill at the moment.

groenmarsmannetje, 2020-02-10:
Okay I have been able to reproduce it.

1. turn group of lights with brightness 100%
2 turn the group of lights off
3 trigger a scene with these settings:

light.group:
      state: on
      brightness: 0

4. The IKEA lights turn on with brightness 1. Other branded bulbs remain off.

Without this change the lights would have stayed turned off after step 3 and when turned on the situation of step 1 would have been restored.

groenmarsmannetje, 2020-02-12:
Changing the scenes did not have the effect that I expected. I still have group of bulbs that are very dimmed when they are turned on. And I also notice that some bulbs in groups that are not turned off. Which is strange because normally a group will turn off as a group almost instant. Will see what happens when I remove the transition from part of the groups and bulbs.

ndfred, 2020-04-13:
@groenmarsmannetje: try and only set the brightness, that will implicitly send the ON state change to the bulbs and may avoid sending the Ikea bulb two commands, which might avoid the issues you are seeing. Here is what I am sending the bulb:

{"brightness": 77, "transition": 0.4} // from OFF state, brightness fades in
{"state": "OFF", "transition": 0.4} // to OFF state, brightness fades out
{"brightness": 254, "transition": 0.4} // from OFF state, brightness fades in

Could you also share which bulb firmware version you have? I have the latest firmware version for the _TRADFRI bulb E14 WS opal 400lm_ bulb: 2.3.007.

closed 2020-06-20 by stale[bot]

wzbfyb, 2020-08-30:
Is there still work going on on this issue?
My Innr 285 ignite the transition setting and always go on/off etc immediately.