Koenkk/zigbee2mqtt #2851: Failed to interview: 7146060PU

TomMettam, 2020-01-29:
# Bug Report
## What happened
Attempting to pair a Philips Hue Go (7146060PU) fails with "Interview failed".

This device (at least the PU variant) is not in the supported devices list, however it should still pass the interview stage since zigbee2mqtt can pair with all zigbee devices.

Log attached.

## Debug Info
zigbee2mqtt version:  287f843bbbc5265f
CC253X firmware version:  CC2531_DEFAULT_20190608

Koenkk, 2020-01-29:
The latest zigbee2mqtt dev branch contains a fix related to Hue pairing, please try with that.

If it doesn't fix please provide the logging again.

TomMettam, 2020-02-03:
The dev branch had the same issue. However, after the 12th(!!!!) attempt, it finally paired.

I guess there's some protocol unreliability here somewhere.

closed 2020-04-11 by stale[bot]