Koenkk/zigbee2mqtt #2852: Unsupported Blitzwolf BW-IS4 Temp and humidity sensor

JoseAlcerreca, 2020-01-29:
I think this sensor is new, and I don't have time right now to follow the instructions to add it. Until I do, here is the log:

```
zigbee2mqtt:debug 2020-01-29 20:12:13: Device '0xec1bbdfffee4bf6x' announced itself
zigbee2mqtt:debug 2020-01-29 20:12:14: Received Zigbee message from '0xec1bbdfffee4bf6x', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltage":32,"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-01-29 20:12:14: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-01-29 20:12:14: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-01-29 20:12:17: Received Zigbee message from '0xec1bbdfffee4bf6x', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2710}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2020-01-29 20:12:17: Received message from unsupported device with Zigbee model 'undefined'
zigbee2mqtt:warn  2020-01-29 20:12:17: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:debug 2020-01-29 20:12:18: Received Zigbee message from '0xec1bbdfffee4bf6x', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2710}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-01-29 20:12:21: Received Zigbee message from '0xec1bbdfffee4bf6x', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5880}' from endpoint 1 with groupID 0
```
Looks very similar to other temp&humidity sensors.

1Topf, 2020-02-15:
There is a similar tuya sensor:

 {
        zigbeeModel: ['TS0201'],
        model: 'TS0201',
        vendor: 'TuYa',
        description: 'Temperature & humidity sensor with display',
        supports: 'temperature and humidity',
        fromZigbee: [fz.battery_percentage_remaining, fz.temperature, fz.humidity],
        toZigbee: [],
    },

I just copied the text above to the BlitzWolf - section and made changes:

{
        zigbeeModel: ['TS0201'],
        model: 'BW-IS4',
        vendor: 'BlitzWolf',
        description: 'Temperature & humidity sensor with display',
        supports: 'temperature and humidity',
        fromZigbee: [fz.battery_percentage_remaining, fz.temperature, fz.humidity],
        toZigbee: [],
    },

Not the pro way, but works for me. :-)

[image]
[image]

orimate, 2020-02-28:
How long is the interval between 2 measurements?

orimate, 2020-02-29:
Hi @Koenkk! 

I added to my system a Blitzwolf-IS4 in the device.js.
{
        zigbeeModel: ['TS0201'],  
        model: 'BW-IS4',
        vendor: 'Blitzwolf',
        description: 'Blitzwolf temperature & humidity sensor with display',
        supports: 'temperature and humidity',										
	fromZigbee: [fz.battery_200, fz.temperature, fz.humidity],			
        toZigbee: [],
    },
The sensor publishing only if the temperaure or humidity values are changed. Example a 15:09 22.00 C and 17:01 22.40 C. 
Is that sensor specified or need I set something to investigate the sensor? The xiaomi sensors are publishing several times in an hour. 

{"version":"1.7.1","commit":"b459c35","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}
 
Thank you for your work!

Koenkk, 2020-03-01:
If this device uses `TS0201` as the zigbeeModel, this devices is actually already supported. It's just a rebrand of the already supported TuYa sensor so it will work fine in Zigbee2mqtt 1.11.0

JoseAlcerreca, 2020-03-02:
Yup, it connected at some point automatically and it works fine.

closed 2020-03-02 by JoseAlcerreca

sjorge, 2020-08-05:
> How long is the interval between 2 measurements?
It depends, I have one in a room that has very low temp/humidity variance and it sometimes takes up to 6h before a new measurement is send.

Others in rooms with higher variance update every few minutes... personally I'd love if they would periodically report even if there was no or a tiny change... but they do not support reporting unfortunately. I haven't figured out the exact thresholds but it looks to be around 0.5C or 2% humidity before it will send an update.

I currently have 6 and they work rather well, aside from ugly holes in my graphs. I also have/had a HS1HT-E from Heiman, they do support reporting and I was able to adjust them to port minimal 1x hour. But the CR2450 battery was a deal breaker, I much prefer AAA.

stefanfoulis, 2020-08-30:
🤔  I tried connecting the device, but it says `supported: false`. Do I need to do anything in addition than just pairing it?

[image]

(`Zigbee2MQTT version 1.14.3 (commit #f8066e8)`)

Koenkk, 2020-08-30:
@stefanfoulis could you share the `data/database.db` entry of your device

stefanfoulis, 2020-08-30:
@Koenkk 
```json
{
    "id": 4,
    "type": "EndDevice",
    "ieeeAddr": "0xec1bbdfffe91c427",
    "nwkAddr": 9603,
    "manufId": 4098,
    "powerSource": "Battery",
    "epList": [
        1
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 770,
            "inClusterList": [
                0,
                1,
                1026,
                1029
            ],
            "outClusterList": [
                25
            ],
            "clusters": {
                "msTemperatureMeasurement": {
                    "attributes": {
                        "measuredValue": 2760
                    }
                },
                "genBasic": {
                    "attributes": {
                        "powerSource": 3,
                        "zclVersion": 3,
                        "appVersion": 67,
                        "stackVersion": 0,
                        "hwVersion": 1,
                        "dateCode": ""
                    }
                },
                "msRelativeHumidity": {
                    "attributes": {
                        "measuredValue": 4960
                    }
                }
            },
            "binds": []
        }
    },
    "appVersion": 67,
    "stackVersion": 0,
    "hwVersion": 1,
    "dateCode": "",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {},
    "lastSeen": 1598747823312
}
```

Koenkk, 2020-08-30:
@stefanfoulis thanks, for some reason the `modelID` is missing. Try to remove the device through https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove and pair it again and provide the herdsman debug logging of this.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

stefanfoulis, 2020-08-30:
🤔  oh... I enabled the herdsman debugging, force removed the device and re-paired it. But this time it just worked 🤷‍♂️.
Must have been some other issue then.

Thanks for helping me out and sorry for the noise.