Koenkk/zigbee2mqtt #2854: Interfaz web Administration

jondiez, 2020-01-29:
I would like to propose the implementation of a web interface for proper administration and monitoring of zigbee devices.  Something that greatly helps the end user for the correct configuration and implementation of the zigbee2mqtt system.
 Thank you!!

Koenkk, 2020-01-29:
Its already there: https://github.com/yllibed/Zigbee2MqttAssistant

closed 2020-01-29 by Koenkk

jondiez, 2020-01-29:
Pero es compatible con Raspbian? En raspberry sin docker? No veo forma de instalarlo.

jondiez, 2020-01-29:
But is it compatible with Raspbian? In raspberry without docker? I see no way to install it.

> Its already there: https://github.com/yllibed/Zigbee2MqttAssistant

Koenkk, 2020-01-30:
For further support please open an issue here https://github.com/yllibed/Zigbee2MqttAssistant/issues