Koenkk/zigbee2mqtt #2855: About the log file...

antti365, 2020-01-30:
@Koenkk

Is it possible to make a "normal" rotation of the log file, the same as in linux? That is, so that the name of the current log file always remains unchanged, and only rotation files change? And now, if log_directory: data / log is specified, then the current log file changes all the time - log.txt, then log1.txt, log2.txt, etc.
The second option is simpler - you can optionally disable rotation in configuration.yaml, who needs it, will configure the rotation by the operating system ...

pcb1962, 2020-01-31:
You can specify the log file location in your configuration.yaml. The default is 
  log_directory: data/log/%TIMESTAMP%
If you remove the TIMESTAMP part then the file location won't change when you restart.
I haven't done this so I'm not sure whether the log file will be appended to or overwritten each time you restart, hopefully appended.
https://www.zigbee2mqtt.io/information/configuration.html

antti365, 2020-01-31:
@Koenkk 
This is all clear, but I wrote about the rotation of the log file, and not about its location.  I probably put it not entirely clear the costs of google translator.  :) Initially, at startup, the log.txt file is created, when a certain size is reached, the next log1.txt file is created and logging continues in it, etc.  In my opinion, this is not correct, the file name log.txt should be unchanged, the same way that the rotation of the log files in the linux system occurs.

Koenkk, 2020-01-31:
That indeed makes sense, the latest log will now always be in `log.txt`. Implemented in latest dev brach.

closed 2020-01-31 by Koenkk