Koenkk/zigbee2mqtt #2856: Eurotronic SPZB0001 one-way communication

AndrzejOlender, 2020-01-30:
# Bug Report
## What happened
Recently I migrated from CC253X to CC1352P-2. Everything worked out after re-pairing (30 devices) except all my Eurotronic. I can send out heat statuses, off, auto. I can send out temperatures, the Eurotronic is receiving it almost instantaneously. 

But the reception from Eurotronic is not working. Local temperature is still the same (value from before migration). If I manually force the Eurotronic temperature (+ or -) it is not transmitted either, the reading after mqtt shows the last value sent.
## What did you expect to happen
I reset the Eurotronic, pairing again, deleting from the database, upgeade to 1.9-dev etc. nothing helped.
## Debug Info
zigbee2mqtt version: 1.9-dev
CC26X2R1 firmware version: 20191106
database.db https://pastebin.com/8ZyD2T8d

AndrzejOlender, 2020-02-05:
If I send the json to mqtt as below, it shows the actual temperature (example 21,5)

`mosquitto_pub -h localhost -p 1883 -t zigbee2mqtt/ThermoChildRoom/get -m '{"local_temperature": "24,0"}'`

But it works one time. If the temperature even changes in the apartment, the isoon shows the last value.
I added it to the cron every minute, but it's ugly.

TilmanK, 2020-02-10:
I have exactly the same problem. I can get the values quite easily when sending a get message on MQTT, configuration works, I can set everything I want.

The thermostat just doesn't report anything back. Also, not after receiving and confirming a set request on MQTT. I sniffed the traffic and indeed the thermostat isn't sending anything.

I cannot see anything that would explain this behavior, but I can pinpoint this down to reporting. I have another thermostat working fine. I just configured it and it stopped reporting anything right after configuration. I have on thermostat left, that hasn't been configured yet and it is still reporting values.

@sjorge: Can you please check the two commits you made (8824dcf9890dae9f6c0e17cdc0f57ccc4fe573a0 and cdc71e7503a4ae2f003dc720125bf3d2bcd09d74) if they somehow broke reporting? As written I wasn't able to find anything.

@Koenkk: Have there been any changes in configuration, especially since 1.9?

sjorge, 2020-02-10:
Only this 8824dcf9890dae9f6c0e17cdc0f57ccc4fe573a0 does anything to do with reporting, and it actually increases the interval to what eurotronic recommends in there docs.

Koenkk, 2020-02-12:
Which of the configureReport configures the local temperature (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L5408). Perhaps the reportableChange should be reduced?

sjorge, 2020-02-12:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L5415 this one does, it should update every 10 minutes (600s) or at 0.25 degrees C change.

TilmanK, 2020-02-12:
I didn't see any problem in the reportableChange values configured. 

@sjorge On what hardware/software of the SPZB0001 did you test the changes? Could you check if there's any difference to mine?

{"id":7,"type":"EndDevice","ieeeAddr":"0x00158d000192dc4d","nwkAddr":31553,"manufId":4151,"manufName":"Eurotronic","powerSource":"Battery","modelId":"SPZB0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,513,25,10],"outClusterList":[0,1,3,4,513,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"SPZB0001","manufacturerName":"Eurotronic","powerSource":3,"zclVersion":2,"appVersion":18,"stackVersion":5,"hwVersion":34,"dateCode":"20190408","swBuildId":"18181120"}},"hvacThermostat":{"attributes":{"16386":0,"16387":2300,"16392":1,"localTemp":1954,"pIHeatingDemand":251,"occupiedHeatingSetpoint":2000,"unoccupiedHeatingSetpoint":1600}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":150}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001caabcf6","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00124b001caabcf6","endpointID":1}]}},"appVersion":18,"stackVersion":5,"hwVersion":34,"dateCode":"20190408","swBuildId":"18181120","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1581360448914}

sjorge, 2020-02-12:
```json
{
  "id": 13,
  "type": "EndDevice",
  "ieeeAddr": "0x00158d0001ffaffc",
  "nwkAddr": 49490,
  "manufId": 4151,
  "manufName": "Eurotronic",
  "powerSource": "Battery",
  "modelId": "SPZB0001",
  "epList": [
    1
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 769,
      "inClusterList": [
        0,
        1,
        3,
        513,
        25,
        10
      ],
      "outClusterList": [
        0,
        1,
        3,
        4,
        513,
        25,
        10
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "SPZB0001",
            "manufacturerName": "Eurotronic",
            "powerSource": 3,
            "zclVersion": 2,
            "appVersion": 22,
            "stackVersion": 5,
            "hwVersion": 35,
            "dateCode": "20191014",
            "swBuildId": "22190930"
          }
        },
        "hvacThermostat": {
          "attributes": {
            "16386": 0,
            "16387": 1900,
            "16392": 1,
            "localTemp": 1750,
            "pIHeatingDemand": 95,
            "occupiedHeatingSetpoint": 1900,
            "unoccupiedHeatingSetpoint": 1600,
            "systemMode": 4,
            "localTemperatureCalibration": 0
          }
        },
        "genPowerCfg": {
          "attributes": {
            "batteryPercentageRemaining": 150,
            "batteryAlarmState": 0
          }
        }
      },
      "binds": [
        {
          "cluster": 1,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b001938a7e5",
          "endpointID": 1
        },
        {
          "cluster": 513,
          "type": "endpoint",
          "deviceIeeeAddress": "0x00124b001938a7e5",
          "endpointID": 1
        }
      ]
    }
  },
  "appVersion": 22,
  "stackVersion": 5,
  "hwVersion": 35,
  "dateCode": "20191014",
  "swBuildId": "22190930",
  "zclVersion": 2,
  "interviewCompleted": true,
  "meta": {
    "configured": 3
  },
  "lastSeen": 1581503013236
}
```

I did a diff with yours mine seems newer, things of note:
- appVersion is 22 vs 18
- hwVersion 35 vs 34 (A new hardware revision?)
- swBuildId 22190930 vs 18181120 (wow that is a huge jump)
- **configured 3 vs 2**  (try running zigbee2mqtt/bridge/configure with the friendly name as payload)

Aside from my hardware being newer, it looks like you are a config revision behind so you might not have all the reporting changes applied yet.

@sti0 had some older revision devices too and they worked fine for him IIRC

TilmanK, 2020-02-12:
The configured value is explainable easy. I altered the value locally in the code to see what it does (without success). Someone might want to explain it to me, since my JS is very limited :/.

But the other differences might explain why mine isn't working anymore. Does anyone know how to upgrade the firmware? Might need to find out...

sjorge, 2020-02-12:
The value should come from: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L5410

Everytime the configuration block is change it should be raised by 1, so zigbee2mqtt knows it needs to reconfigure the device or not.

sti0, 2020-02-12:
@sjorge I returned the old ones because the don't work properly and got two new ones with the newer revision. They work quite well (sometimes Mac errors #715)

sjorge, 2020-02-12:
I'm a bit surprise on how many software revision and even a hardware revision happened in the span of less than half a year.

Maybe the older ones did have a big issue. Sadly I haven't seen any firmwares for these flooting around either.

AndrzejOlender, 2020-02-13:
My version numbers are ok from what I see.

`{
    "id": 42,
    "type": "EndDevice",
    "ieeeAddr": "0x00158d0001ffafc5",
    "nwkAddr": 10071,
    "manufId": 4151,
    "manufName": "Eurotronic",
    "powerSource": "Battery",
    "modelId": "SPZB0001",
    "epList": [
        1
    ],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 769,
            "inClusterList": [
                0,
                1,
                3,
                513,
                25,
                10
            ],
            "outClusterList": [
                0,
                1,
                3,
                4,
                513,
                25,
                10
            ],
            "clusters": {
                "hvacThermostat": {
                    "attributes": {
                        "16387": 2200,
                        "16392": 17,
                        "unoccupiedHeatingSetpoint": 1600,
                        "occupiedHeatingSetpoint": 2200,
                        "pIHeatingDemand": 0,
                        "localTemp": 1950,
                        "localTemperatureCalibration": 0
                    }
                },
                "genBasic": {
                    "attributes": {
                        "modelId": "SPZB0001",
                        "manufacturerName": "Eurotronic",
                        "powerSource": 3,
                        "zclVersion": 2,
                        "appVersion": 22,
                        "stackVersion": 5,
                        "hwVersion": 35,
                        "dateCode": "20191014",
                        "swBuildId": "22190930"
                    }
                }
            },
            "binds": [
                {
                    "cluster": 1,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00124b0018e25dbc",
                    "endpointID": 1
                },
                {
                    "cluster": 513,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x00124b0018e25dbc",
                    "endpointID": 1
                }
            ]
        }
    },
    "appVersion": 22,
    "stackVersion": 5,
    "hwVersion": 35,
    "dateCode": "20191014",
    "swBuildId": "22190930",
    "zclVersion": 2,
    "interviewCompleted": true,
    "meta": {
        "configured": 3
    },
    "lastSeen": 1581446584808
}`

sjorge, 2020-02-13:
Yeah your device looks the same as mine, I wonder if it is because the CC1352P-2 is zigbee 3.1 and the other one was 1.2 🤔

AndrzejOlender, 2020-02-14:
@TilmanK  do you also use CC1352P-2?

TilmanK, 2020-02-14:
I have a CC26X2R1 with Zigbee 3 too. But is this a likely cause?

sti0, 2020-02-17:
Today one of my devices stopped working like this. I receive updates if I set the temperature at the device but I can't send commands by mqtt...

zigbee2mqtt 1.10 and CC26X2R1...

sti0, 2020-02-17:
Today 1 of my devices stopped working with HA. I receive messages but can't set temperature / hvac mode in HA. If I set it directly via mqtt it works fine.

Maybe something broken in HA?

TilmanK, 2020-02-17:
@sti0 What appVersion, hwVersion and swBuildId does your device report? Just for the record...

sti0, 2020-02-17:
All 3 devices:
"hardwareVersion": 35,
"softwareBuildID": "22190930",
"dateCode": "20191014"

AndrzejOlender, 2020-02-18:
@sti0 My thermostat behaves the opposite. I can send statuses but I can't receive them from the thermostat.

TilmanK, 2020-03-29:
Just some update on this: I've upgraded to the latest dev branch and accidentally deleted my database. After repairing the thermostat it works. Not that I've not tried that before, but whatever.

Will try with the other ones today...

AndrzejOlender, 2020-03-29:
Great news. It still doesn't work for me. I didn't just delete my base. Let me know how the other thermostats work.

AndrzejOlender, 2020-03-29:
I deleted the database, downloaded the latest dev... works perfectly both ways! (tested on two devices).

closed 2020-05-30 by AndrzejOlender