Koenkk/zigbee2mqtt #2858: Hue power_on_color doesn't work with philips bulb 929001953101

kitopopo, 2020-01-30:
Dear @Koenkk ,

Throughout today I have tried many commands to try to store the lighting behavior of the bulb but none works.  According to the documentation I think the messages are correct have tried the following commands:

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/0x0017880104dc1794/set -m '{"hue_power_on_brightness": 125, "hue_power_on_color": "#37ff00" , "hue_power_on_behavior": "on"}'    
`  
_(After send the previosly command no info appears in the log. and after a power cut the bulb powers on in blue color)_

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/lamps/set -m "{\"hue_power_on_brightness\": \"255\", \"hue_power_on_color\": \"#37ff00\" , \"hue_power_on_behavior\":\"on\"}"`

_(after send the previosly command to the group, the log is the following:

_ene 30 18:49:31 raspberrypi npm[505]: zigbee2mqtt:error 2020-01-30 18:49:31: No converter available for 'hue_power_on_behavior' (on)
ene 30 18:49:31 raspberrypi npm[505]: zigbee2mqtt:error 2020-01-30 18:49:31: No converter available for 'hue_power_on_color' (#37ff00)
ene 30 18:49:31 raspberrypi npm[505]: zigbee2mqtt:error 2020-01-30 18:49:31: No converter available for 'hue_power_on_brightness' (255)_

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/bridge/config/0x0017880104dc1794/set -m '{"hue_power_on_brightness": 125, "hue_power_on_color": "37ff00" , "hue_power_on_behavior":"on"}'`

_(After send the previosly command no info appears in the log. and after a power cut the bulb powers on in blue color)_

I also test the followitns paths in the topics but no results for me:

zigbee2mqtt/config/lamps/set
zigbee2mqtt/config/lamps/set
zigbee2mqtt/bridge/config/lamps/set
zigbee2mqtt/bridge/config/lamps/add

Assuming these messages will work I find an added problem. for these messages to work the power on behavior must be set to "on". If you have a power loss at home for a few minutes, the lamps will always come back on. Unless we use the wall switch as off. but if we use mqtt or domoticz messages as off, the bulbs will remain on after the loss of tension in the home.

I wait new from you, best regards and thanks in advanced. 

## What happened

The bulb powers on in blue colour after power on. the color is not storage in the bulb/group

## What did you expect to happen

the bulb should be powers on in the desired color

## How to reproduce it (minimal and precise)

Sending the previosly topics with mosquito

## Debug Info

zigbee2mqtt version: version 1.9.0 Lastet dev brach
CC2531 firmware version: 20190608.zip

Koenkk, 2020-01-30:
You cannot send this to a group, you need to set it for each individual bulb.

kitopopo, 2020-01-30:
Hi dear friend,

Ok, i know it now, thanks. But the first message is sended only to 1 bulb and also doensn't  work.

Thanks

Koenkk, 2020-01-31:
Found the issue, fixed in latest dev branch!

closed 2020-01-31 by Koenkk

kitopopo, 2020-01-31:
Thanks dear friend, i will test. Best regards

kitopopo, 2020-01-31:
Dear @Koenkk ,

I test, Now works correctly. nice job. regards