Koenkk/zigbee2mqtt #2864: State of devices incorrect

michaeladam3010, 2020-01-31:
Hello,
I own one of the Müller Licht 404021 smart switches.
First, I used the switch with the IKEA Tradfri gateway and realized, that the IKEA app was always showing the correct state of the switch meaning, that it was showing that the light was switched on (or off), even when I switched it on (or off) at the switch.
zigbee2mqtt seem to not provide the actual state of the switch.

Environment:
zigbee2mqtt  1.9.0
hass.io

Koenkk, 2020-01-31:
Try enabling the report feature: https://www.zigbee2mqtt.io/information/report.html

michaeladam3010, 2020-01-31:
Enabling the report feature fixed the problem.
Thank you very much for your help and the really short response time :)

closed 2020-01-31 by michaeladam3010