Koenkk/zigbee2mqtt #2866: Cannot rebuild my zigbee network - pairing failed

johnnychicago, 2020-01-31:
#Bug Report
## What happened
Hello --
I had to rebuild my docker host and was offline for a few days. Bringing the docker back up, devices failed to connect to zigbee2mqtt. After playing around with re-pairing, I decided to restart, and created a new container (using a different channel on the zigbee controller to force re-pairing.) I am still not able to pair any device.

I have Ikea, Philips, and Xiaomi Mi devices here, but none of them pairs.

## What did you expect to happen
Devices would pair when factory resetting them.

## How to reproduce it (minimal and precise)
If I knew :)

## Debug Info
log: https://pastebin.com/yYdkSdBm
config: https://pastebin.com/6CswEmUH
zigbee2mqtt version: latest docker
CC253X firmware version:

Koenkk, 2020-01-31:
Please try the steps from: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

johnnychicago, 2020-02-01:
@Koenkk  - thanks for pointing me to the obvious, that's what I needed. :-) Let me document for the next guy trying to google this.
I did of course look at this before filing the issue, but went back. Yes, my dongle was nicely pulled from behind the computer with an USB cable. And it was sitting on top of the box, just as it used to be.
The only difference was that it was currently sitting on a case that had its covers removed, since I wasn't bothering closing it again, waiting for a new hard drive to be delivered. Turns out, quite surprisingly, that all the noise from an open box can actually overpower the low power zigbee network.
I'll let myself out now.

closed 2020-02-01 by johnnychicago