Koenkk/zigbee2mqtt #2867: Error: libnode.so.64: cannot open shared object file

switchtower, 2020-01-31:
# Bug Report
## What happened
Unable to start application using `npm start`
## What did you expect to happen
The application to start
## How to reproduce it (minimal and precise)
npm start
## Debug Info
```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

zigbee2mqtt@1.9.0 start /opt/zigbee2mqtt
 node index.js

zigbee2mqtt:info  2020-01-31 20:37:36: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-01-31.20-37-36'
zigbee2mqtt:info  2020-01-31 20:37:37: Starting zigbee2mqtt version 1.9.0 (commit #287f843)
zigbee2mqtt:info  2020-01-31 20:37:37: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-01-31 20:37:37: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-01-31 20:37:37: Failed to start zigbee
zigbee2mqtt:error 2020-01-31 20:37:37: Exiting...
zigbee2mqtt:error 2020-01-31 20:37:37: Error: libnode.so.64: cannot open shared object file: No such file or directory
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1021:18)
    at Module.load (internal/modules/cjs/loader.js:811:32)
    at Function.Module._load (internal/modules/cjs/loader.js:723:14)
    at Module.require (internal/modules/cjs/loader.js:848:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at bindings (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:2:36)
    at Module._compile (internal/modules/cjs/loader.js:955:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
    at Module.load (internal/modules/cjs/loader.js:811:32)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.9.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.9.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-01-31T20_37_37_259Z-debug.log
```
zigbee2mqtt version: 1.9.0
CC253X firmware version: 1.2 I believe
Raspberry Pi 4
Raspbian 10.1

sjorge, 2020-01-31:
Looks like a broken node install?

switchtower, 2020-01-31:
perhaps, i installed it on a fresh install, and everything appears to be working

closed 2020-01-31 by switchtower