Koenkk/zigbee2mqtt #2868: Unsuccessful Pairing of IM6001-BTP01

switchtower, 2020-01-31:
# Bug Report
## What happened
Unable to pair SmartThings IM6001-BTP01 Smart Button 
## What did you expect to happen
the button to pair
## How to reproduce it (minimal and precise)
Attempted to reset or  click button to force pairing
## Debug Info
```
zigbee2mqtt:info  2020-01-31 21:58:59: Starting interview of '0x286d9700010a2792'
zigbee2mqtt:info  2020-01-31 21:58:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x286d9700010a2792"}}'
zigbee2mqtt:error 2020-01-31 22:00:19: Failed to interview '0x286d9700010a2792', device has not successfully been paired
zigbee2mqtt:info  2020-01-31 22:00:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x286d9700010a2792"}}'
zigbee2mqtt:info  2020-01-31 22:01:52: Starting interview of '0x286d9700010a2792'
zigbee2mqtt:info  2020-01-31 22:01:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x286d9700010a2792"}}'
zigbee2mqtt:error 2020-01-31 22:03:12: Failed to interview '0x286d9700010a2792', device has not successfully been paired
zigbee2mqtt:info  2020-01-31 22:03:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x286d9700010a2792"}}'
```
zigbee2mqtt version: 1.9
CC253X firmware version: 1.2

Koenkk, 2020-01-31:
Please provide herdsman debug log when pairing this device on the latest **dev** branch.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

switchtower, 2020-01-31:
Thanks for the quick responses!
```
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":35863,"extaddr":"0x286d9700010a2792","parentaddr":0} +2m
  zigbee-herdsman:controller:log Device '0x286d9700010a2792' joined +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:controller:log Device '0x286d9700010a2792' accepted by handler +1ms
  zigbee-herdsman:controller:log New device '0x286d9700010a2792' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x286d9700010a2792' +0ms
zigbee2mqtt:info  2020-01-31 22:28:09: Device '0x286d9700010a2792' joined
zigbee2mqtt:info  2020-01-31 22:28:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x286d9700010a2792"}}'
  zigbee-herdsman:controller:log Interview '0x286d9700010a2792' start +15ms
zigbee2mqtt:info  2020-01-31 22:28:09: Starting interview of '0x286d9700010a2792'
zigbee2mqtt:info  2020-01-31 22:28:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x286d9700010a2792"}}'
  zigbee-herdsman:controller:device Interview - start device '0x286d9700010a2792' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":35863,"nwkaddrofinterest":35863} +2m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,23,140,23,140,35] +2m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +34ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,252,165,1,1,0,5,0,204,105,222,0,0,7,24,46,10,0,0,16,1,252,165,29,145] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,252,165,1,1,0,5,0,204,105,222,0,0,7,24,46,10,0,0,16,1,252,165,29,145] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,252,165,1,1,0,5,0,204,105,222,0,0,7,24,46,10,0,0,16,1,252,165,29] - 145 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":42492,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":5,"securityuse":0,"timestamp":14576076,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,46,10,0,0,16,1]}} +3s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":46,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":1}],"Cluster":{"ID":6,"attributes":{"onOff":{"ID":0,"type":16,"name":"onOff"},"globalSceneCtrl":{"ID":16384,"type":16,"name":"globalSceneCtrl"},"onTime":{"ID":16385,"type":33,"name":"onTime"},"offWaitTime":{"ID":16386,"type":33,"name":"offWaitTime"}},"name":"genOnOff","commands":{"off":{"ID":0,"parameters":[],"name":"off"},"on":{"ID":1,"parameters":[],"name":"on"},"toggle":{"ID":2,"parameters":[],"name":"toggle"},"offWithEffect":{"ID":64,"parameters":[{"name":"effectid","type":32},{"name":"effectvariant","type":32}],"name":"offWithEffect"},"onWithRecallGlobalScene":{"ID":65,"parameters":[],"name":"onWithRecallGlobalScene"},"onWithTimedOff":{"ID":66,"parameters":[{"name":"ctrlbits","type":32},{"name":"ontime","type":33},{"name":"offwaittime","type":33}],"name":"onWithTimedOff"}},"commandsResponse":{}}},"address":42492,"endpoint":1,"linkquality":5,"groupID":0}' +3s
zigbee2mqtt:debug 2020-01-31 22:28:12: Received Zigbee message from '0x00158d00044d9310', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-01-31 22:28:12: MQTT publish: topic 'zigbee2mqtt/0x00158d00044d9310', payload '{"battery":100,"voltage":3055,"contact":false,"linkquality":5}'
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +18ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,252,165,1,1,0,5,0,2,106,222,0,0,7,24,47,10,0,0,16,0,252,165,29,92] +5ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,252,165,1,1,0,5,0,2,106,222,0,0,7,24,47,10,0,0,16,0,252,165,29,92] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,252,165,1,1,0,5,0,2,106,222,0,0,7,24,47,10,0,0,16,0,252,165,29] - 92 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":42492,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":5,"securityuse":0,"timestamp":14576130,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,47,10,0,0,16,0]}} +24ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":47,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}],"Cluster":{"ID":6,"attributes":{"onOff":{"ID":0,"type":16,"name":"onOff"},"globalSceneCtrl":{"ID":16384,"type":16,"name":"globalSceneCtrl"},"onTime":{"ID":16385,"type":33,"name":"onTime"},"offWaitTime":{"ID":16386,"type":33,"name":"offWaitTime"}},"name":" [... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +58ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
^Czigbee2mqtt:debug 2020-01-31 22:28:42: Saving state to file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-01-31 22:28:42: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
zigbee2mqtt:info  2020-01-31 22:28:42: Disconnecting from MQTT server
```

Koenkk, 2020-01-31:
Please post the log until you see `interview_failed`.

switchtower, 2020-02-03:
@Koenkk This appeared to be an issue with either distance from the CC2531 being too great or channel interference.  I updated the channel, since my WiFi was using channel 9, to channel 20, and tried to be more line-of-site to the coordinator and was able to get everything working.

I'm a bit disappointed about the range of the CC2531, but that's not a coding issue, that's an issue with either placement of the board, or interference, which is my issue to figure out. 

Thanks for the help!

closed 2020-02-03 by switchtower