Koenkk/zigbee2mqtt #2869: Cannot pair xbee router

Exhortae, 2020-01-31:
Hi,

I'm trying to pair unsuccessfully an xbee s2c pro with zigbee2mqtt 

This is the message I get :

```
zigbee2mqtt:info  2020-01-31 23:14:19: Device '0x0013a20041a77278' joined
zigbee2mqtt:info  2020-01-31 23:14:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0013a20041a77278"}}'
zigbee2mqtt:info  2020-01-31 23:14:19: Starting interview of '0x0013a20041a77278'
zigbee2mqtt:info  2020-01-31 23:14:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0013a20041a77278"}}'
zigbee2mqtt:info  2020-01-31 23:14:19: MQTT publish: topic 'zigbee2mqtt/0x0013a20041a77278/availability', payload 'online'
zigbee2mqtt:error 2020-01-31 23:15:39: Failed to interview '0x0013a20041a77278', device has not successfully been paired
zigbee2mqtt:info  2020-01-31 23:15:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x0013a20041a77278"}}'
zigbee2mqtt:info  2020-01-31 23:15:42: Starting interview of '0x0013a20041a77278'
```

zigbee2mqtt version: 1.8.0
Coordinator : cc26x2r1

Koenkk, 2020-01-31:
Please provide herdsman debug log when pairing this device on the latest **dev** branch.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Exhortae, 2020-01-31:
OK here is the debut log using edge

```
2020-01-31T22:47:04.900Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,103,84,120,114]
2020-01-31T22:47:04.900Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,103,84,120,114]
2020-01-31T22:47:04.901Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [167,65,0,162,19,0,0,0,237]
2020-01-31T22:47:04.901Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,103,84,120,114,167,65,0,162,19,0,0,0,237]
2020-01-31T22:47:04.901Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [103,84,120,114,167,65,0,162,19,0,0,0] - 237
2020-01-31T22:47:04.901Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":21607,"extaddr":"0x0013a20041a77278","parentaddr":0}
2020-01-31T22:47:04.902Z zigbee-herdsman:controller:log Device '0x0013a20041a77278' joined
2020-01-31T22:47:04.902Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-31T22:47:04.902Z zigbee-herdsman:controller:log Device '0x0013a20041a77278' accepted by handler
2020-01-31T22:47:04.902Z zigbee-herdsman:controller:log New device '0x0013a20041a77278' joined
2020-01-31T22:47:04.902Z zigbee-herdsman:controller:log Creating device '0x0013a20041a77278'
zigbee2mqtt:info  2020-01-31 23:47:04: Device '0x0013a20041a77278' joined
zigbee2mqtt:info  2020-01-31 23:47:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0013a20041a77278"}}'
2020-01-31T22:47:04.910Z zigbee-herdsman:controller:log Interview '0x0013a20041a77278' start
zigbee2mqtt:info  2020-01-31 23:47:04: Starting interview of '0x0013a20041a77278'
zigbee2mqtt:info  2020-01-31 23:47:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0013a20041a77278"}}'
2020-01-31T22:47:04.912Z zigbee-herdsman:controller:device Interview - start device '0x0013a20041a77278'
2020-01-31T22:47:04.915Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":21607,"nwkaddrofinterest":21607}
2020-01-31T22:47:04.915Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,103,84,103,84,35]
2020-01-31T22:47:04.929Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-01-31T22:47:04.930Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-01-31T22:47:04.930Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-01-31T22:47:04.930Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-01-31T22:47:04.930Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-01-31T22:47:14.920Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x0013a20041a77278', attempt 1
2020-01-31T22:47:14.920Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":21607,"nwkaddrofinterest":21607}
2020-01-31T22:47:14.920Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,103,84,103,84,35]
2020-01-31T22:47:14.996Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2020-01-31T22:47:14.996Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2020-01-31T22:47:14.996Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2020-01-31T22:47:14.996Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2020-01-31T22:47:14.996Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

Koenkk, 2020-02-01:
Thanks, please provide the log until you see `interview_failed`

Exhortae, 2020-02-01:
Sorry,

```
2020-02-01T17:07:36.843Z zigbee-herdsman:controller:log Permit joining,
2020-02-01T17:07:36.843Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0},
2020-02-01T17:07:36.843Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228],
2020-02-01T17:07:37.146Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82,254,3],
2020-02-01T17:07:37.146Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82,254,3],
2020-02-01T17:07:37.146Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82,
2020-02-01T17:07:37.146Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0},
2020-02-01T17:07:37.147Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3],
2020-02-01T17:07:37.147Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [69,182,0,0,0,240],
2020-02-01T17:07:37.147Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240],
2020-02-01T17:07:37.147Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240,
2020-02-01T17:07:37.147Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0},
2020-02-01T17:07:37.147Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [],
2020-02-01T17:07:48.637Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,224,108,120,114],
2020-02-01T17:07:48.637Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,224,108,120,114],
2020-02-01T17:07:48.639Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [167,65,0,162,19,0,0,0,82],
2020-02-01T17:07:48.639Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,224,108,120,114,167,65,0,162,19,0,0,0,82],
2020-02-01T17:07:48.639Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [224,108,120,114,167,65,0,162,19,0,0,0] - 82,
2020-02-01T17:07:48.639Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":27872,"extaddr":"0x0013a20041a77278","parentaddr":0},
2020-02-01T17:07:48.639Z zigbee-herdsman:controller:log Device '0x0013a20041a77278' joined,
2020-02-01T17:07:48.639Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [],
2020-02-01T17:07:48.639Z zigbee-herdsman:controller:log Device '0x0013a20041a77278' accepted by handler,
2020-02-01T17:07:48.639Z zigbee-herdsman:controller:log Interview '0x0013a20041a77278' start,
2020-02-01T17:07:48.641Z zigbee-herdsman:controller:device Interview - start device '0x0013a20041a77278',
2020-02-01T17:07:48.642Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":27872,"nwkaddrofinterest":27872},
2020-02-01T17:07:48.642Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,224,108,224,108,35],
zigbee2mqtt:info  2020-02-01 18:07:48: Starting interview of '0x0013a20041a77278',
zigbee2mqtt:info  2020-02-01 18:07:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0013a20041a77278"}}',
2020-02-01T17:07:48.654Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102],
2020-02-01T17:07:48.655Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102],
2020-02-01T17:07:48.655Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102,
2020-02-01T17:07:48.655Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0},
2020-02-01T17:07:48.655Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [],
2020-02-01T17:07:58.643Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x0013a20041a77278', attempt 1,
2020-02-01T17:07:58.644Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":27872,"nwkaddrofinterest":27872},
2020-02-01T17:07:58.644Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,224,108,224,108,35],
2020-02-01T17:07:58.727Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102],
2020-02-01T17:07:58.727Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102],
2020-02-01T17:07:58.727Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102,
2020-02-01T17:07:58.727Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0},
2020-02-01T17:07:58.727Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [],
2020-02-01T17:08:08.646Z zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x0013a20041a77278', attempt 2,
2020-02-01T17:08:08.646Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":27872,"nwkaddrofinterest":27872},
2020-02-01T17:08:08.646Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,224,108,224,108,35],
2020-02-01T17:08:08.799Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102],
2020-02-01T17:08:08.799Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102],
2020-02-01T17:08:08.799Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102,
2020-02-01T17:08:08.799Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0},
2020-02-01T17:08:08.799Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [],
[... cut ...]
2020-02-01T17:09:08.657Z zigbee-herdsman:controller:error Interview failed for '0x0013a20041a77278 with error 'Error: Interview failed because can not get active endpoints ('0x0013a20041a77278')',
zigbee2mqtt:error 2020-02-01 18:09:08: Failed to interview '0x0013a20041a77278', device has not successfully been paired,
zigbee2mqtt:info  2020-02-01 18:09:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x0013a20041a77278"}}',

```

Koenkk, 2020-02-01:
Looks like the device does not respond with the endpoints. Note that if you only want to use for router purposes it already works (interview does not have to succeed for that).

Exhortae, 2020-02-01:
I will only use it as a router, so I guess I'm good.

Is there a way to disable the multiple interviews retry ?

Thanks

Koenkk, 2020-02-01:
Edit database.db manually and set `interviewCompleted` to true

Exhortae, 2020-02-02:
I did it but it keeps trying to pair.

1/Stopped zigbee2mqtt

2/edited database.db

`{"id":9,"ieeeAddr":"0x0013a20041a77278","nwkAddr":2906,"epList":[],"endpoints":{},"interviewCompleted":true,"meta":{}}`

3/started zigbee2mqtt

4/ device is not in the list of device

```
zigbee2mqtt:info  2020-02-02 03:04:14: Device 'Xbee Routeur - Haut' joined
zigbee2mqtt:info  2020-02-02 03:04:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"Xbee Routeur - Haut"}}'
zigbee2mqtt:info  2020-02-02 03:04:14: Starting interview of 'Xbee Routeur - Haut'
zigbee2mqtt:info  2020-02-02 03:04:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"Xbee Routeur - Haut"}}'
zigbee2mqtt:error 2020-02-02 03:05:34: Failed to interview 'Xbee Routeur - Haut', device has not successfully been paired
zigbee2mqtt:info  2020-02-02 03:05:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"Xbee Routeur - Haut"}}'
zigbee2mqtt:info  2020-02-02 03:05:51: Starting interview of 'Xbee Routeur - Haut'
zigbee2mqtt:info  2020-02-02 03:05:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"Xbee Routeur - Haut"}}'
```

5/ Device is trying to pair again

```
zigbee2mqtt:info  2020-02-02 03:04:14: Device 'Xbee Routeur - Haut' joined
zigbee2mqtt:info  2020-02-02 03:04:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"Xbee Routeur - Haut"}}'
zigbee2mqtt:info  2020-02-02 03:04:14: Starting interview of 'Xbee Routeur - Haut'
zigbee2mqtt:info  2020-02-02 03:04:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"Xbee Routeur - Haut"}}'
zigbee2mqtt:error 2020-02-02 03:05:34: Failed to interview 'Xbee Routeur - Haut', device has not successfully been paired
zigbee2mqtt:info  2020-02-02 03:05:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"Xbee Routeur - Haut"}}'
zigbee2mqtt:info  2020-02-02 03:05:51: Starting interview of 'Xbee Routeur - Haut'
zigbee2mqtt:info  2020-02-02 03:05:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"Xbee Routeur - Haut"}}'
```

6/ New lines are added to database.db

`{"id":10,"ieeeAddr":"0x0013a20041a77278","nwkAddr":46806,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1580609297622}`

If I set this last line to true it keeps getting set automatically to false every 60 seconds

Also the network graph shows the xbee router not connected to the coordinator

`zigbee2mqtt:error 2020-02-02 03:15:22: Failed to execute LQI for 'Xbee Routeur - Haut'`

[image]

Koenkk, 2020-02-02:
Did you stop zigbee2mqtt before doing this?

Exhortae, 2020-02-02:
Yes I stopped it.

the xbee device is definitly not behaving like a router (apart from a few seconds at the beginning, no device connects to it, even those who are out of reach of the coordinator)

vmanuel, 2020-03-09:
Same issue

vmanuel, 2020-03-10:
If I usr the configuration in
https://github.com/Koenkk/zigbee2mqtt/issues/880

ZS = 2
NJ = 5A
NO = 3
SN = AF0
SP = 64

in XBee module ( firmware XBP24-ZB, ZigBee Router API mode, version 23A7), it seems to work

alekslyse, 2020-04-05:
For any xbee issues with connecting I suggest using serial (+++) to enable, then check with
ATAI (https://www.digi.com/resources/documentation/Digidocs/90002002/Reference/r_cmd_AI.htm?tocpath=AT%20commands%7CDiagnostic%20commands%7C_____3)

Its a life saver to debug xbee joining

closed 2020-06-12 by stale[bot]

Exhortae, 2020-08-30:
> If I usr the configuration in
> #880
> 
> ZS = 2
> NJ = 5A
> NO = 3
> SN = AF0
> SP = 64
> 
> in XBee module ( firmware XBP24-ZB, ZigBee Router API mode, version 23A7), it seems to work

Still cannot join even if using these values.

XBP24C
ZIGBEE TH Pro
Firm. 4061

Exhortae, 2020-08-30:
I followed the configuration of this article and at last it is working. Only thing that I didn't follow is setting the pan id (I left it 0)

https://community.hubitat.com/t/everything-xbee/2328