Koenkk/zigbee2mqtt #2871: Trouble again with OSRAM lightstrip

marcgarciamarti, 2020-02-01:
`zigbee2mqtt:info  2020-02-01 15:44:11: MQTT publish: topic 'zigbee2mqtt/cube1xiaomi', payload '{"battery":86,"voltage":2975,"linkquality":144,"angle":55.49,"side":0,"action":"rotate_right"}'
zigbee2mqtt:info  2020-02-01 15:44:11: MQTT publish: topic 'zigbee2mqtt/cube1xiaomi', payload '{"battery":86,"voltage":2975,"linkquality":144,"angle":55.49,"side":0,"action":""}'
zigbee2mqtt:info  2020-02-01 15:45:15: MQTT publish: topic 'zigbee2mqtt/occupancys1xiaomi', payload '{"occupancy":true,"linkquality":147,"battery":100,"voltage":3005}'
zigbee2mqtt:error 2020-02-01 15:45:19: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Data request failed with error: 'No network route' (205)'
zigbee2mqtt:info  2020-02-01 15:45:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"ledstrip1osram"}}'
zigbee2mqtt:error 2020-02-01 15:45:34: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Data request failed with error: 'No network route' (205)'
zigbee2mqtt:info  2020-02-01 15:45:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"ledstrip1osram"}}'
zigbee2mqtt:error 2020-02-01 15:45:48: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Data request failed with error: 'No network route' (205)'
zigbee2mqtt:info  2020-02-01 15:45:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"ledstrip1osram"}}'
zigbee2mqtt:error 2020-02-01 15:45:53: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Data request failed with error: 'No network route' (205)'
zigbee2mqtt:info  2020-02-01 15:45:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"ledstrip1osram"}}'
zigbee2mqtt:info  2020-02-01 15:46:07: MQTT publish: topic 'zigbee2mqtt/plug1xiaomi', payload '{"state":"OFF","power":0,"voltage":234.5,"consumption":55.74,"temperature":34,"linkquality":76}'
zigbee2mqtt:info  2020-02-01 15:46:32: MQTT publish: topic 'zigbee2mqtt/occupancys1xiaomi', payload '{"occupancy":true,"linkquality":144,"battery":100,"voltage":3005}'
zigbee2mqtt:info  2020-02-01 15:46:34: MQTT publish: topic 'zigbee2mqtt/cube1xiaomi', payload '{"battery":86,"voltage":2975,"linkquality":144,"angle":55.49,"side":0,"action":"wakeup"}'
zigbee2mqtt:info  2020-02-01 15:46:34: MQTT publish: topic 'zigbee2mqtt/cube1xiaomi', payload '{"battery":86,"voltage":2975,"linkquality":144,"angle":55.49,"side":0,"action":""}'
zigbee2mqtt:info  2020-02-01 15:46:35: MQTT publish: topic 'zigbee2mqtt/cube1xiaomi', payload '{"battery":86,"voltage":2975,"linkquality":144,"angle":55.49,"side":0,"action":"shake"}'
zigbee2mqtt:info  2020-02-01 15:46:35: MQTT publish: topic 'zigbee2mqtt/cube1xiaomi', payload '{"battery":86,"voltage":2975,"linkquality":144,"angle":55.49,"side":0,"action":""}'
zigbee2mqtt:error 2020-02-01 15:46:36: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Data request failed with error: 'No network route' (205)'
zigbee2mqtt:info  2020-02-01 15:46:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"ledstrip1osram"}}'
zigbee2mqtt:info  2020-02-01 15:46:36: MQTT publish: topic 'zigbee2mqtt/cube1xiaomi', payload '{"battery":86,"voltage":2975,"linkquality":144,"angle":55.49,"side":0,"action":"shake"}'
zigbee2mqtt:info  2020-02-01 15:46:36: MQTT publish: topic 'zigbee2mqtt/cube1xiaomi', payload '{"battery":86,"voltage":2975,"linkquality":144,"angle":55.49,"side":0,"action":""}'
zigbee2mqtt:error 2020-02-01 15:46:37: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Data request failed with error: 'No network route' (205)'
zigbee2mqtt:info  2020-02-01 15:46:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Data request failed with error: 'No network route' (205)'","meta":{"friendly_name":"ledstrip1osram"}}' `

Koenkk, 2020-02-01:
What coordinator device and firmware are you using?

marcgarciamarti, 2020-02-01:
> What coordinator device and firmware are you using?

Here it is:
`zigbee2mqtt:info  2020-02-01 17:04:01: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190223}} `

Koenkk, 2020-02-01:
You might want to try: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing 

It has the zigbee source routing functionality enabled which allows it to remember routes  (to prevent `No network route' (205)`).

marcgarciamarti, 2020-02-01:
> You might want to try: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

Is that a firmware for the coordinator? My coordinator is the CC2530

> It has the zigbee source routing functionality enabled which allows it to remember routes (to prevent `No network route' (205)`).

So, you think it is not that the network intensity is not enough but rather a matter or routing?

Koenkk, 2020-02-02:
@marcgarciamarti I think due to network quality routing can have some hiccups, remembering the routes might solve it. You will find the CC2530 coordinator in the same directory. (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

marcgarciamarti, 2020-02-02:
> @marcgarciamarti I think due to network quality routing can have some hiccups, remembering the routes might solve it. You will find the CC2530 coordinator in the same directory. (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

@Koenkk you rock!! Does this scenario fall in the category which allows me to update my current coordinator without repairing? 

Also, I have a spare stick lying around and I wonder if I could somehow backup my current coordinator prior to updating my “productive one”? May I ask what extend do sticks hold relevant information about my setup (devices, pairing hashes, etc.)?

Koenkk, 2020-02-02:
- You can switch freely between z-stack 1.2 firmwares, so yes
- You don't have to backup anything, just swap the stick and start (but the one you switch it with should also run a z-stack 1.2 firmware).

marcgarciamarti, 2020-02-08:
> * You can switch freely between z-stack 1.2 firmwares, so yes
> * You don't have to backup anything, just swap the stick and start (but the one you switch it with should also run a z-stack 1.2 firmware).

Hi again,
I just upgraded the firmware and tried to turn on the device with no results either:

`'{"battery":100,"voltage":3015,"linkquality":0,"angle":-18.53,"side":0}'
zigbee2mqtt:info  2020-02-08 01:04:22: MQTT publish: topic 'zigbee2mqtt/plug1xiaomi', payload '{"state":"OFF","power":0,"voltage":234.5,"consumption":56.12,"temperature":33,"linkquality":0}'
zigbee2mqtt:info  2020-02-08 01:04:58: MQTT publish: topic 'zigbee2mqtt/plug3xiaomi', payload '{"state":"ON","power":16.77,"voltage":null,"consumption":14.43,"temperature":47,"linkquality":0}'
zigbee2mqtt:info  2020-02-08 01:05:24: MQTT publish: topic 'zigbee2mqtt/doors1xiaomi', payload '{"contact":true,"linkquality":65,"battery":80,"voltage":2965}'
zigbee2mqtt:error 2020-02-08 01:06:57: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
zigbee2mqtt:info  2020-02-08 01:06:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"ledstrip1osram"}}'
zigbee2mqtt:error 2020-02-08 01:07:03: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
zigbee2mqtt:info  2020-02-08 01:07:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"ledstrip1osram"}}' `

this is the current version:

`./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-02-08T01:03:29: PM2 log: Launching in no daemon mode
2020-02-08T01:03:30: PM2 log: App [npm:0] starting in -fork mode-
2020-02-08T01:03:30: PM2 log: App [npm:0] online
> zigbee2mqtt@1.10.0 start /zigbee2mqtt-1.10.0
> node index.js
zigbee2mqtt:info  2020-02-08 01:03:36: Logging to console and directory: '/share/zigbee2mqtt/log/2020-02-08.01-03-35' filename: log.txt
zigbee2mqtt:info  2020-02-08 01:03:37: Starting zigbee2mqtt version 1.10.0 (commit #unknown)
zigbee2mqtt:info  2020-02-08 01:03:37: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-02-08 01:03:39: zigbee-herdsman started
zigbee2mqtt:info  2020-02-08 01:03:39: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}}' `

any more ideas? 

Thanks a million @Koenkk

marcgarciamarti, 2020-02-08:
More details below:

`zigbee2mqtt:info  2020-02-08 14:07:34: MQTT publish: topic 'zigbee2mqtt/cube1xiaomi', payload '{"battery":100,"voltage":3015,"linkquality":0,"angle":13.14,"side":0,"action":""}'
zigbee2mqtt:error 2020-02-08 14:07:35: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
zigbee2mqtt:info  2020-02-08 14:07:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"ledstrip1osram"}}'
zigbee2mqtt:info  2020-02-08 14:07:46: MQTT publish: topic 'zigbee2mqtt/temperatures3xiaomi', payload '{"temperature":22.41,"linkquality":5,"humidity":50.14,"pressure":1002.5,"battery":80,"voltage":2965}'
zigbee2mqtt:info  2020-02-08 14:07:46: MQTT publish: topic 'zigbee2mqtt/temperatures3xiaomi', payload '{"temperature":22.41,"linkquality":0,"humidity":50.26,"pressure":1002.5,"battery":80,"voltage":2965}'
zigbee2mqtt:info  2020-02-08 14:07:46: MQTT publish: topic 'zigbee2mqtt/temperatures3xiaomi', payload '{"temperature":22.41,"linkquality":0,"humidity":50.26,"pressure":1001,"battery":80,"voltage":2965}'
zigbee2mqtt:error 2020-02-08 14:07:57: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
zigbee2mqtt:info  2020-02-08 14:07:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"ledstrip1osram"}}'
zigbee2mqtt:error 2020-02-08 14:08:14: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
zigbee2mqtt:info  2020-02-08 14:08:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"ledstrip1osram"}}'
zigbee2mqtt:error 2020-02-08 14:08:19: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
zigbee2mqtt:info  2020-02-08 14:08:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"ledstrip1osram"}}'
zigbee2mqtt:error 2020-02-08 14:08:24: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
zigbee2mqtt:info  2020-02-08 14:08:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"ledstrip1osram"}}'
zigbee2mqtt:info  2020-02-08 14:08:30: MQTT publish: topic 'zigbee2mqtt/plug3xiaomi', payload '{"state":"ON","power":9.84,"voltage":null,"consumption":14.61,"temperature":47,"linkquality":52}'
zigbee2mqtt:info  2020-02-08 14:08:30: MQTT publish: topic 'zigbee2mqtt/light11000lmikea', payload '{"state":"ON","brightness":255,"linkquality":0}'
zigbee2mqtt:error 2020-02-08 14:08:34: Publish 'set' 'state' to 'ledstrip1osram' failed: 'Error: Command 0x8418260000caa3af/3 genOnOff.on({}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'MAC no ack' (233))' `

Koenkk, 2020-02-08:
You can try sniffing the traffic (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

marcgarciamarti, 2020-02-08:
You won't believe it but I'm pretty sure it is a matter of reception after all. Just moved around some stuff in the house and so far it seems to be working back!!!!