Koenkk/zigbee2mqtt #2872: Interview failed Gledopto

huahe, 2020-02-02:
Hello !

I am trying to add two Gledopto RGBW controllers, GL-C-007/GL-C-008.

Every time I try to add them (reseting with the power cycle) I get a "Interview failed" error.

It pairs perfectly with Hue hub, but not con zigbee2mqtt.

I tried to pair it just next to it (<1 meter) with no luck.

I am using HASSIO addon.

This is the debug log I got.

https://pastebin.com/wB1X5t9T

Thanks !!

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Koenkk, 2020-02-02:
- First test if it still happens with the latest zigbeee2mqtt dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)
- If it still happens after that provide the herdsman debug logging

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

huahe, 2020-02-02:
Thanks for the quick response !
I installed the edge version and same result. The strange thing is that the device seems to join the network as a router...

https://controlc.com/ce850889/fullscreen.php?hash=d0b8c29c7109e900129c270608fe11d2&toolbar=true&linenum=false

Koenkk, 2020-02-02:
I don't see the herdsman debug logging, note that it's written to the screen and not the the log files

> The strange thing is that the device seems to join the network as a router...

This device is a router so this is expected.

huahe, 2020-02-02:
Oh sorry, I didn't know that.

This is it the log rom screen:

https://pastebin.com/G9YjNrU9

Koenkk, 2020-02-03:
Probably this will be fixed after #2844 will be fixed (https://github.com/Koenkk/zigbee2mqtt/issues/2844#issuecomment-581133226)

For know you can workaround it by manually setting the `modelID` in `database.db` to `GLEDOPTO`.

huahe, 2020-02-04:
Thanks !! It worked... partially.

I can see the device now in Home Assistant, but I can't control anything. This is what I get in the log.

Thanks !!!

https://pastebin.com/RzhDSEV7

Koenkk, 2020-02-04:
Instead of `GLEDOPTO` can you try `GL-C-008`.

Koenkk, 2020-02-07:
Pushed some fixes to the latest dev branch, can you check and provide the log?

closed 2020-04-14 by stale[bot]