Koenkk/zigbee2mqtt #2874: Zigbee2mqtt over ESP8266 with serial port cc2530

roesl, 2020-02-02:
I want to connect a cc2530 over a software transparent serial port with ESP8266 ( e.g. ESPLink ) to build a independent Zigbee hub for use with my Synology NAS.
There is no VSP for Synology so I have to add a network connection to config file.
Is this possible with zwigbee2mqtt ?

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

roesl, 2020-04-02:
no Idea ??

Koenkk, 2020-04-02:
https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

roesl, 2020-04-03:
thank you for the link.

markuspeter, 2020-04-06:
Hi @roesl,
were you able to make this work ? Was there any additional configuration on the Synology NAS needed besides defining the port in the z2m configuration ?

roesl, 2020-04-06:
I will order the cc2530 device this week in china, so I think it will arrive the next 6 week.
Then I will try to install

roesl, 2020-04-24:
Hi,

 

Thank you for the link, but it doesnt work on my Synology NAS with zigbee2mqtt installed as docker container.

Is there a solution ?

 

Regards

Georg

 

https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

roesl, 2020-04-24:
I got the devices today, but it doesnt work on my Synology with zigbee2mqtt installed in the docker container

roesl, 2020-04-25:
I got it work on a virtual machine with Ubuntu 18 where I also installed IP-Symcon.
But now I have problems with the switches from IKEA - the can be connected to the coordinator but they dont work.
All other devices I use are working ( Aquara sensors, Aldi LEDs )
Maybe its a problem of the coordinator firmware for the SOC with 2530+2592

roesl, 2020-04-25:
Now I installed the firmware from 
Z-Stack-firmware/coordinator/Z-Stack_Home_1.2/bin/default/
and it works fine.
Maybe someone get it work for a Synology Docker container - this would be easier to install.
Thanks to Koenkk for the great work !

closed 2020-07-03 by stale[bot]