Koenkk/zigbee2mqtt #2876: zigbee_publish_error with CC2531 and Iluminize 511.202

ochstobi, 2020-02-02:
Hello,
I'm subscribing to the topic of my Iluminize Switch with MQTT.fx. When I turn on/off my switch via home assistant, the switch is working well but the state can't be saved. After looking at the /bridge/log I saw this message:
{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'switch_buero' failed: 'Error: Data request failed with error: 'MAC no ack' (233)'","meta":{"friendly_name":"switch_buero"}}
What can cause this Problem even the switch can be toggled on/off?
I tried to set a higher qos für this device, but I don't know how. I have the edited the devices.yaml like below, restartet the zigbee2mqtt container (and mosquitto and home assistant) but the qos displayed in MQTT.fx (and home assistant, too) shows qos: 0. Why isn't it updated?
'0xec1bbdfffe93fa38':
  friendly_name: 'switch_buero'
  qos: 1
I'm new to zigbee in general and only got it working today. Hope someon can help me.

ochstobi, 2020-02-02:
The first problem is solved when using an usb extender cable. Sorry for the noise.
But I can't update the configuration regarding QoS. Can anyone help me out?

Koenkk, 2020-02-02:
@ochstobi good, I will close this as the original issue has been solved

For `qos` see https://www.zigbee2mqtt.io/information/configuration.html

To set it for all devices add the following to `configuration.yaml`.

```yaml
device_options:
  qos: 1
```

closed 2020-02-02 by Koenkk