Koenkk/zigbee2mqtt #2877: New Device -  Heiman - Model ID: WarningDevice-EF-3.0

joseluisgodinho, 2020-02-02:
Hello

I would appreciate your support for this new device.
Thank you
José Godinho

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Koenkk, 2020-02-02:
Please provide a link to the device and your `database.db` when paired.

joseluisgodinho, 2020-02-02:
{"id":7,"type":"Router","ieeeAddr":"0xccccccfffedc1c91","nwkAddr":52068,"manufId":4619,"manufName":"HEIMAN","modelId":"WarningDevice-EF-3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1027,"inClusterList":[0,1,3,1280,1282,2821],"outClusterList":[4,25],"clusters":{"genBasic":{"attributes":{"modelId":"WarningDevice-EF-3.0","manufacturerName":"HEIMAN","powerSource":129,"zclVersion":2,"appVersion":19,"stackVersion":2,"hwVersion":16,"dateCode":"2019.5.23"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}},"ssIasWd":{"attributes":{"61440":0}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":19,"stackVersion":2,"hwVersion":16,"dateCode":"2019.5.23","zclVersion":2,"interviewCompleted":false,"meta":{},"lastSeen":1580672779309}

joseluisgodinho, 2020-02-02:
https://www.aliexpress.com/item/4000185176198.html?scm=1007.22893.149154.0&pvid=ab531f9c-6cd3-4a23-aa2f-1309b94e1ee8&onelink_page_from=ITEM_DETAIL&onelink_item_to=4000185176198&onelink_publisherid=172076246&onelink_memberseq=786288887&onelink_duration=1.788834&onelink_status=noneresult&onelink_item_from=4000185176198&onelink_subid=324395&onelink_page_to=ITEM_DETAIL&af=324395&afref=&cv=1578323&dp=1578323%253A%253A324395%253A%253ACjwKCAiAg9rxBRADEiwAxKDTusNDaUhSzJoIOsstSXuUDk1pdN.Xts.zjT8gbdPQjHxCReQj62gvwhoCcboQAvD.BwE%253A%253A%253A%253A1580674437&cn=27294&aff_request_id=489896062e1249f0861efacd85c3bf3f-1580674437486-02568-Y7bAZbY&aff_platform=aaf&cpt=1580674437486&sk=Y7bAZbY&aff_trace_key=489896062e1249f0861efacd85c3bf3f-1580674437486-02568-Y7bAZbY&terminal_id=cb4a0ca0f5ac412bb4c2d50a0f99bfb6

Koenkk, 2020-02-02:
Try adding the following to `devices.js` (https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html)

```js
{
    zigbeeModel: ['WarningDevice-EF-3.0'],
    model: 'M420',
    vendor: 'HEIMAN',
    description: 'Zigbee 3.0 smart siren',
    supports: 'warning',
    fromZigbee: [fz.battery_200],
    toZigbee: [tz.warning],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await configureReporting.batteryPercentageRemaining(endpoint);
    },
},
```

Now you should be able to control it by sending to `zigbee2mqtt/0xccccccfffedc1c91/set` with payload `{"warning": {"mode": "fire", "level": "medium"}}`

joseluisgodinho, 2020-02-02:
WORKING!

Thank you

closed 2020-02-02 by joseluisgodinho

Koenkk, 2020-02-03:
Good, will be supported in next zigbee2mqtt version out of the box.

mehmetu61, 2025-05-01:
> Frage dazu: Obwohl es in Zigbee2MQTT angezeigt wird und ich dort sogar einen Alarm auslösen kann, scheint es in Home Assistant keine Möglichkeit zu geben, einen Alarm auszulösen. Es werden lediglich Verbindungsqualität und Akku angezeigt und ich habe die Möglichkeit, die Einstellung für die maximale Dauer zu ändern. Übersehe ich hier etwas?

I have the same problem