Koenkk/zigbee2mqtt #2878: Error SRSP -AF datarequest after 6000ms  CC1352P-2

ozer59, 2020-02-02:
Sometimes i have this message 
on network i have 27 devices aqara door/ temp/sensor and Legrand netatmo

debug 2020-02-02 20:57:44: Error: SRSP - AF - dataRequest after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
info  2020-02-02 20:57:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'prise salon' failed: 'Error: SRSP - AF - dataRequest after 6000ms'","meta":{"friendly_name":"prise salon"}}'

i see on : https://github.com/Koenkk/zigbee2mqtt/issues/1429#issuecomment-581126407
an other firmware i can try it ? 

## Debug Info
zigbee2mqtt  version 1.9.0 (commit #287f843) on docker (xpenology) without USB extension cable
firmwareversion:'{"type":"zStack3x0","meta{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'

configuration.yaml:
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: null
advanced:
  pan_id: 6755
  baudrate: 115200
  report: true
  rtscts: false
  log_level: debug

Koenkk, 2020-02-02:
You shouldn't set `rtscts: false` for this coordinator, try removing it from the `configuration.yaml`.

ozer59, 2020-02-02:
thank you for quick response 
i remove rtscts: false
but i have same error 

debug 2020-02-02 22:22:57: Error: SRSP - AF - dataRequest after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
info  2020-02-02 22:22:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00047400001d9b3d' failed: 'Error: SRSP - AF - dataRequest after 6000ms'","meta":{"friendly_name":"0x00047400001d9b3d"}}'

Koenkk, 2020-02-03:
Can you provide the complete logging?

tedsluis, 2020-02-12:
I think I have the same issue. 

# Bug Report
## What happened

Currently I have a good functioning zigbee2mqtt with all kinds of zigbee sensors, mostly xiaomi sensors. No issues so far.

I tried to pair an unsupported device HUE 9290002593D (white bulb, E27, 11,5Watt, 2700K, 1055lm, 105mA), that used to be paired to a HUE bridge, using the touchlink methode:

* permit_join: true
* Placed the blub near (1cm) the CC1352P-2 coordinator during pairing.
* Published a MQTT *zigbee2mqtt/bridge/config/touchlink/factory_reset* without payload from home-assistant
* I see this in the zigbee2mqtt logging (DEBUG loglevel):

```
feb 12 22:31:27 : zigbee2mqtt:debug 2020-02-12 22:31:27: Received MQTT message on 'zigbee2mqtt/bridge/config/touchlink/factory_reset' with data 'null'
feb 12 22:31:27 : winston:file written true false
feb 12 22:31:27 : 2020-02-12T21:31:27.666Z zigbee-herdsman:controller:touchlink Set InterPAN channel to '11'
feb 12 22:31:27 : 2020-02-12T21:31:27.666Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - interPanCtl - {"cmd":1,"data":[11]}
feb 12 22:31:27 : winston:file logged 179078 debug 2020-02-12 22:31:27: Received MQTT message on 'zigbee2mqtt/bridge/config/touchlink/factory_reset' with data 'null'
feb 12 22:31:27 : zigbee2mqtt:info  2020-02-12 22:31:27: Starting touchlink factory reset...
```
* 44 seconds of silence later, I see this in the logging:

```
feb 12 22:31:33 : zigbee2mqtt:error 2020-02-12 22:31:33: Failed to call 'BridgeConfig' 'onMQTTMessage' (Error: SRSP - AF - interPanCtl after 6000ms
feb 12 22:31:33 :     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
feb 12 22:31:33 :     at listOnTimeout (internal/timers.js:531:17)
feb 12 22:31:33 :     at processTimers (internal/timers.js:475:7))
```
* I tried this procedure over and over again with a couple of bulbs that used to be paired with a HUE bridge.

full zigbee2mqtt log at https://pastebin.com/JxwDCF5p

## What did you expect to happen

I would aspect the bulb to blink, to pair and to show up in device.yaml. Non of this happened.

## How to reproduce it (minimal and precise)

Just follow the steps I described above.

## Debug Info

zigbee2mqtt version: docker image koenkk/zigbee2mqtt:latest-dev (3 days ago)
CC1352P-2 coordinator firmware version: {'type': 'zStack3x0', 'meta': {'transportrev': 2, 'product': 1, 'majorrel': 2, 'minorrel': 7, 'maintrel': 1, 'revision': 20191106}}

### Configuration.yaml

```yaml
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://**********'
  user: *********
  password: *********************
  reject_unauthorized: false
  include_device_information: true
serial:
  port: /dev/ttyACM0
advanced:
  network_key: ************************
  pan_id: *******
  ext_pan_id: *********************************
  channel: 26
  cache_state: true
  log_level: debug
  last_seen: ISO_8601
  elapsed: false
  availability_timeout: 0
  availability_blacklist:
    - DEVICE_FRIENDLY_NAME
  report: true
  homeassistant_discovery_topic: homeassistant
  homeassistant_status_topic: hass/status
map_options:
  graphviz:
    colors:
      fill:
        enddevice: '#fff8ce'
        coordinator: '#e04e5d'
        router: '#4ea3e0'
      font:
        coordinator: '#ffffff'
        router: '#ffffff'
        enddevice: '#000000'
      line:
        active: '#009900'
        inactive: '#994444'
queue:
  delay: 250
  simultaneously: 5
device_options: null
devices: devices.yaml
groups: groups.yaml
```

ozer59, 2020-02-13:
@Koenkk
https://pastebin.com/hzsupRVX

this is log with error

ozer59, 2020-02-13:
https://pastebin.com/zmbt7Ft5

Koenkk, 2020-02-14:
I cannot really explain why it happens, does it only occur with `applique salon`?

ozer59, 2020-02-15:
no all devices 
i have legrand devices and aqara 
i lose aqara door sensor and vibration after few hours need repairing

closed 2020-04-25 by stale[bot]

Smanar, 2020-04-27:
@ozer59 Do you still have the problem ?

I have same problem than you (but on another project) It seem Xiaomi don't support Legrand device as router, but I m not sure yet, collecting information.

ozer59, 2020-04-27:
i have this problem only with zigbee2mqtt on docker 
if a use CC1352P-2 with pi no problem 
on my network map no xiaomi device connected on legrand router 

Sorry for my poor english

Smanar, 2020-04-28:
Ok so I don't think it's the same problem.

So atm you have touchlink issue and some aquara disconnection but only on a special hardware and not caused by legrand device  ?

ozer59, 2020-04-29:
I switch zigbee2mqtt from xpenology (docker) to rasberrypi and now it work correctly 
i think like you Xiaomi don't support Legrand device as router on my system i never see  aqara device connected on legrand routeur

https://github.com/Koenkk/zigbee2mqtt/issues/2399#issuecomment-580410792

Smanar, 2020-04-29:
Yes, I have return from other zigbee project, It have nothing to see with your issue, but yes, same reaction, Legrand can work as router but not for Xiaomi, but It's not the only one with this problem.