Koenkk/zigbee2mqtt #2879: Cannot connect new devices to a TRADFRI repeater

ArkadiuszNiemiec, 2020-02-02:
# Bug Report
## What happened
I am not able to add any new device to a TRADFRI repeater. I was able to connect few devices when I started to configure the network but after weeks I am not able to add anything. Everything pairs to a coordinator (devices few centimeters from repeater) and now that the coordinator is full I am not able to connect anything at all.
I have tried rebooting the server, unplugging the CC2531 etc. Nothing helped.
## What did you expect to happen
I expect the devices to connect to a TRADFRI repeater if they are close to it or if coordinator is full.
## How to reproduce it (minimal and precise)
Network like this:
[image]
I try to add a device that is close to a repeater but it does not connect.
## Debug Info
zigbee2mqtt version: 1.9.0
CC253X firmware version: 20190608 default

ArkadiuszNiemiec, 2020-02-03:
Seems to be related to #2843.
@Koenkk you asked there for a startup log, it might be useful here as well:

```
zigbee2mqtt:info  2020-02-03 16:51:35: Currently 23 devices are joined:
zigbee2mqtt:info  2020-02-03 16:51:35: office_repeater (0x000d6ffffed2b6d7): E1746 - IKEA TRADFRI signal repeater (Router)
zigbee2mqtt:info  2020-02-03 16:51:35: bedroom_sensor (0x00158d0003d4378b): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: office_wall_switch (0x00158d000302f075): QBKG03LM - Xiaomi Aqara double key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (Router)
zigbee2mqtt:info  2020-02-03 16:51:35: bedroom_wall_switch (0x00158d000345426c): QBKG03LM - Xiaomi Aqara double key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: office_radiator (0x00158d0003c47319): SPZB0001 - Eurotronic Spirit Zigbee wireless heater thermostat (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: bathroom_wall_switch (0x00158d00034542a8): QBKG03LM - Xiaomi Aqara double key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (Router)
zigbee2mqtt:info  2020-02-03 16:51:35: bedroom_cover (0x04cf8cdf3c74a41b): ZNCLDJ12LM - Xiaomi Aqara B1 curtain motor  (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: office_desk_switch (0x00158d00028a9064): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: corridor_wall_switch (0x00158d00030a21ba): QBKG04LM - Xiaomi Aqara single key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (Router)
zigbee2mqtt:info  2020-02-03 16:51:35: corridor_motion_sensor (0x00158d00041451af): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: bedroom_night_switch_karolina (0x00158d0001e8724b): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: bedroom_radiator (0x00158d0003c3fc7a): SPZB0001 - Eurotronic Spirit Zigbee wireless heater thermostat (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: bedroom_night_switch_arek (0x00158d000205ae96): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: kitchen_wall_switch (0x00158d000302f092): QBKG03LM - Xiaomi Aqara double key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (Router)
zigbee2mqtt:info  2020-02-03 16:51:35: diningroom_wall_switch (0x00158d0002a9ba45): QBKG04LM - Xiaomi Aqara single key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (Router)
zigbee2mqtt:info  2020-02-03 16:51:35: livingroom_wall_switch (0x00158d000343c9a4): QBKG03LM - Xiaomi Aqara double key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (Router)
zigbee2mqtt:info  2020-02-03 16:51:35: livingroom_sensor (0x00158d00023ce81a): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: outdoor_sensor (0x00158d00023ac220): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: livingroom_radiator (0x00158d0003c4602e): SPZB0001 - Eurotronic Spirit Zigbee wireless heater thermostat (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: diningroom_radiator (0x00158d0003c45fea): SPZB0001 - Eurotronic Spirit Zigbee wireless heater thermostat (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: bathroom_sensor (0x00158d000423ec9c): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: office_sensor (0x00158d000407469e): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
zigbee2mqtt:info  2020-02-03 16:51:35: office_cover (0x04cf8cdf3c74ad87): ZNCLDJ12LM - Xiaomi Aqara B1 curtain motor  (EndDevice)
```

Koenkk, 2020-02-03:
I think the devices somehow prefer the coordinator over the TRADFRI repeaters, Zigbee2mqtt cannot influence this behaviour.

However what we can do is to only open the network at a specific node so that the device only can join through that. Let's continue in #2888.

closed 2020-02-03 by Koenkk