Koenkk/zigbee2mqtt #2880: CC2531 usb stick model E18-2G4U04B support?

mauro-borg, 2020-02-03:
I'm trying to flash the firmware on a CC2531 usb stick, using a NodeMCU (ESP-12E) with Arduino IDE and CCLoader on Linux.
The stick is a blue pcb, with no headers - had to do some soldering to get the connections to a breadboard (there are five connection holes, identified by one letter: V=3v3, D=DD, C=DC, R=Reset, G=GND).
CCLoader seems to complete (Program successfully! File closed! Comport closed!), however when plugging in the usb stick there is no sign of activity (and no messages from the kernel).

Prior to the flashing attempt, when plugged in, the kernel reported:
usb 2-1.2: New USB device found, idVendor=0451, idProduct=16ae
usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
usb 2-1.2: Product: CC2531 USB Dongle
usb 2-1.2: Manufacturer: Texas Instruments

Shell command used for flashing: ./CCLoader /dev/ttyUSB0 CC2531ZNP-Prod.bin 1

Is it ok to use this firmware image?
Was anyone able to use this stick? (https://www.aliexpress.com/item/4000511888519.html)

closed 2020-04-10 by stale[bot]

ortigosano, 2020-05-19:
did you got it?

mauro-borg, 2020-05-26:
> did you got it?

No, I even tried to connect it to a raspberry pi zero, and do the flashing in an alternative way, but I couldn't get it to work. Now I suspect I have a defective item. Or, the blue pcb could be wired in a slightly different way or requires a different image?

Pravaya, 2022-11-20:
have you done it ?

simaoafonso, 2023-12-29:
Try again with the Raspberry and make sure you have wiringpi installed, ir worked for me