Koenkk/zigbee2mqtt #2881: Philips Hue LWA007 interview failed

nineclicks, 2020-02-03:
Add to devices.js:

```
    {
        zigbeeModel: ['LWA007'],
        model: '929002277501',
        vendor: 'Philips',
        description: 'Hue white A19 bulb E26 bluetooth',
        extend: hue.light_onoff_brightness,
    },
```

```
zigbee2mqtt:info  2020-02-02 17:42:47: Device '0x0017880108681fe1' joined
zigbee2mqtt:info  2020-02-02 17:42:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0017880108681fe1"}}'
zigbee2mqtt:info  2020-02-02 17:42:47: Starting interview of '0x0017880108681fe1'
zigbee2mqtt:info  2020-02-02 17:42:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0017880108681fe1"}}'
zigbee2mqtt:debug 2020-02-02 17:42:47: Device '0x0017880108681fe1' announced itself
zigbee2mqtt:debug 2020-02-02 17:42:47: Device '0x0017880108681fe1' announced itself
zigbee2mqtt:debug 2020-02-02 17:42:48: Received Zigbee message from '0x0017880108681fe1', type 'readResponse', cluster 'genBasic', data '{"modelId":"LWA007","manufacturerName":"Philips","powerSource":1}' from endpoint 11 with groupID 0
zigbee2mqtt:debug 2020-02-02 17:42:48: Received Zigbee message from '0x0017880108681fe1', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2,"appVersion":2,"stackVersion":1}' from endpoint 11 with groupID 0
zigbee2mqtt:error 2020-02-02 17:42:58: Failed to interview '0x0017880108681fe1', device has not successfully been paired
zigbee2mqtt:info  2020-02-02 17:42:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x0017880108681fe1"}}'
```

I am unsure where to add converters on a Hassio docker setup.

Kryzek, 2020-02-03:
There was pr https://github.com/Koenkk/zigbee-herdsman/commit/f3ba8cde0e45778f740c186f6ec6845d8ddd12f3 to address Hue pairing issues. Try to update 1.9.0-dev.

nineclicks, 2020-02-08:
Updated to 1.10.0 and I was able to interview with my devices.js entry. Thanks.

closed 2020-02-08 by nineclicks