Koenkk/zigbee2mqtt #2882: Philips hue Rporting

filoor, 2020-02-03:
Hi 
I have issues with a couple of my hue light that they will not report the state back.
I'am on the Dev. i did try the prod. but have the same issue and I am using "Texas Instruments CC1352P-2"

this are the models of the bulbs that have issus:
LWB004	
LWB006
LCT001
LWB004

Log:
`error 2020-02-02 22:58:26: Failed to setup reporting for '0x0017880100dff179' - Error: AREQ - ZDO - bindRsp after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
error 2020-02-02 22:58:26: Failed to setup reporting for '0x00178801104c549a' - Error: Timeout - 35974 - 11 - 2 - 6 - 7 after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
info  2020-02-02 22:58:34: Zigbee: allowing new devices to join.
info  2020-02-02 22:58:36: Starting network scan (includeRoutes 'false')
error 2020-02-02 22:58:36: Failed to setup reporting for '0x0017880100dd283b' - Error: Timeout - 27601 - 11 - 3 - 6 - 7 after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
info  2020-02-02 22:58:36: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.9.0-dev","commit":"cf1bb386ead98a78b6b3bfed01e8d151cc7e9c6c","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}},"log_level":"info","permit_join":true}'
error 2020-02-02 22:58:36: Failed to setup reporting for '0x0017880100dff8af' - Error: Timeout - 27968 - 11 - 4 - 6 - 7 after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)`

Koenkk, 2020-02-03:
Hue bulbs only support on/off reporting: https://www.zigbee2mqtt.io/information/report.html#support

filoor, 2020-02-03:
@Koenkk thanks for your reply.
I have my hue bulb that is binded to a hue switch.
If I turn it on or off from the switch it won't reflect to my hassio.

Odianosen25, 2020-02-08:
@Koenkk, 

Same issue here, but I have reported it here https://github.com/Koenkk/zigbee2mqtt/issues/1064#issuecomment-583717868

Koenkk, 2020-02-08:
Should be fixed in latest dev branch (#2730, thanks to @docbobo), please confirm.

filoor, 2020-02-10:
I did not receive any update on my hassio edge add-on I will let you know when I have it

Thank you so much

Odianosen25, 2020-02-10:
@Koenkk,

Thanks for the merge. I am now running dev on docker, but I still get the same issue. Also there is this thing of when I use the `TOGGLE` command, sometimes it gets out of sync with the feedback. The bulb will be `on` but its reporting `OFF` and vise versa.

I have the bulbs set to report state on command, and even `report` to `true`, but that doesn't help the issue.

Kind regards

fnfid, 2020-02-10:
Started changing a group of IKEA partwise by Philips Hues, reporting is shurly set to on, and as long as an IKEA bulb is in the group, it reports state and brighthness to the group. Despite Hue reporting read to be not supported, mine do well, but if only Hues and not a single IKEA in the group, each bulb reports its state and brightness, but group does not get it.
As i thought that code writes infos to groups, tried but couldnt find the difference of IKEA and HUE reporting to groups.
Meanwhile node-red reads info from bulbs and simulates this for the group as a workaround for relevant cases, works fine for now, but hope someone has a hint for the corresponding codepiecelet.

Koenkk, 2020-02-12:
Please share your `database.db` and debug logging when changing the state of the bulb using the remote (https://www.zigbee2mqtt.io/information/debug.html#zigbee2mqtt-debug-logging)

Odianosen25, 2020-02-12:
Hello @Koenkk,

How do I share it privately to you? Or is it ok for me to just put it here?

Regards

Odianosen25, 2020-02-12:
@Koenkk,

Here is the log you asked for with `log_level` set to DEBUG. 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4195019/log.txt)

Koenkk, 2020-02-14:
Found the issue, fixed in latest dev branch.

If not, feel free to reopen.

mmaster23, 2020-02-14:
I just tested this with my own setup on the latest dev build. ON command is now reporting fine, even brightness is reporting now with Hue bulbs. However OFF doesn't seem to register.

mmaster23, 2020-02-15:
@Koenkk  Not sure how to reopen this bug, maybe only OP can. Hope this tag mention gets through.

My database.db and logging is found here: https://pastebin.com/Tsss0Fnp

tunip, 2020-02-15:
In my setup with Philips Hue ON/OFF reporting is working when remote is Bond to the device. If i bind the remote to a group ON reporting is working but OFF not.

Koenkk, 2020-02-16:
@mmaster23 did you also bind the remote to a group?

mmaster23, 2020-02-16:
Forgot to mention that.. yes, my remote is binded to a group.

kcarriello, 2020-02-18:
> In my setup with Philips Hue ON/OFF reporting is working when remote is bind to the device. If i bind the remote to a group ON reporting is working but OFF not.

I’m having the same behavior. When group is configured lamps in the group report ON but not report OFF.

Koenkk, 2020-02-22:
Found the issue, fixed in latest dev branch.

closed 2020-02-22 by Koenkk

kcarriello, 2020-02-22:
Thanks! Any possibility to have this updated also for Hassio add-on?

mmaster23, 2020-02-22:
Can confirm that this fixes reporting for off command to groups. (for my setup at least).

kcarriello, 2020-02-22:
What did you do to get this working? I tried uninstalling/reinstalling add-on, but the group behavior is the same. Reports ON but no OFF reports.

mmaster23, 2020-02-22:
Not really related to this thread but the HassIO addon uses a different github repo (https://github.com/danielwelch/hassio-zigbee2mqtt) which isn't in sync (perfectly) with this Koenkk repo. I just ran the docker container as described on the Zigbee2Mqtt docs site.

shlomki, 2023-03-04:
@Koenkk 
Sorry to bump an old bug report, but I figured asking a question here would be better than reopening a new bug.
I seem to be having the same problem with my hue bulb that reporting doesn't work- meaning, if I have optimistic = false and I turn the bulb on or off from z2m, the bulb reacts correctly but the state isn't updated in z2m until I click refresh.
I've tried to configure on/off reporting in the reporting tab for the bulb, but I wasn't able to - it returned errors.
Is this supposed to work like this?

Here is the info for my hue light

```
Zigbee Model
LWB010
Zigbee Manufacturer
Philips
Description
Hue White A60 Single bulb E27/B22
Support status
Supported
IEEE Address
0x00178801026d0508
Network address
0xF3EA
Firmware build date
20221006
Firmware version
1.101.2
Manufacturer
[Philips](https://www.zigbee2mqtt.io/supported-devices/#v=Philips)
Model
[8718696449691](https://www.zigbee2mqtt.io/devices/8718696449691.html#philips-8718696449691)
```