Koenkk/zigbee2mqtt #2884: Xiaomi WXCJKG13LM - double tap / hold not recognized

mcweba, 2020-02-03:
Hi there
I'm running the latest version of zigbee2mqtt (1.9.0) and could successfully pair the aqara opple 6-button switch. The switch works very well, but not all actions are recognized. 

The following table shows what actions are detected on which buttons.

| Button  | Actions  |
| :-------: |-------------|
| 1           | single |
| 2           | single |
| 3           | single, hold, release |
| 4           | single, hold, release |
| 5           | single, hold, release |
| 6           | single, hold, release |

So the double tap is never recognized. Instead, one single click will be recognized when double tapping. Also, for buttons 1 and 2 the hold and release actions are not recognized.

Koenkk, 2020-02-03:
This has been added recently and is only available in the dev branch yet (https://github.com/Koenkk/zigbee-herdsman-converters/pull/945).

Will also be available in zigbee2mqtt 1.10 scheduled for this week.

closed 2020-02-03 by Koenkk

mcweba, 2020-02-03:
Thank you for the quick feedback. Looking forward to version 1.10

stanman71, 2020-02-08:
I have the same problem and the update to version 1.10 did't changed it.

Is there anythink i have to do except the update installation ?

mcweba, 2020-02-08:
I also had to remove the button from the network and add it again

stanman71, 2020-02-08:
Thanks for your fast response.

Everything is working now.

leranp, 2020-11-19:
same here with version 1.16.1 , what can i do?