Koenkk/zigbee2mqtt #2886: Can not interview Paul Neuhaus Q-FLAG LED Panel (100.110.40)

rhuss, 2020-02-03:
# Bug Report

I can't pair with Paul Neuhaus Q-FLAG LED Panel. Its article number is 100.110.40 (not 100.110.39 as listed on the [supported device list](https://www.zigbee2mqtt.io/information/supported_devices.html) but I hope its close enough to work similarly).

The error is, that interview ends with an timeout.

## What happened

<details>
<summary>Debug Log</summary>

```
mqtt zigbee2mqtt-0 zigbee2mqtt zigbee2mqtt:info  2020-02-03 12:51:03: Starting interview of '0x00158d0001d7f109'
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.620Z zigbee-herdsman:controller:device Interview - start device '0x00158d0001d7f109'
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.621Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":55023,"nwkaddrofinterest":55023}
mqtt zigbee2mqtt-0 zigbee2mqtt zigbee2mqtt:info  2020-02-03 12:51:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0001d7f109"}}'
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.621Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,239,214,239,214,35]
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.621Z serialport/stream _write 9 bytes of data
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.621Z serialport/binding-abstract write 9 bytes
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.622Z serialport/bindings/unixWrite Starting write 9 bytes offset 0 bytesToWrite 9
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.622Z serialport/stream _read reading { start: 4525, toRead: 61011 }
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.622Z serialport/binding-abstract read
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.623Z serialport/bindings/unixRead Starting read
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.623Z serialport/bindings/unixWrite write returned: wrote 9 bytes
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.624Z serialport/bindings/unixWrite Finished writing 9 bytes
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.624Z serialport/stream binding.write write finished
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.624Z serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
mqtt zigbee2mqtt-0 zigbee2mqtt   errno: -11,
mqtt zigbee2mqtt-0 zigbee2mqtt   code: 'EAGAIN',
mqtt zigbee2mqtt-0 zigbee2mqtt   syscall: 'read'
mqtt zigbee2mqtt-0 zigbee2mqtt }
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.624Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.624Z serialport/bindings/poller Polling for "readable"
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.634Z serialport/bindings/poller received "readable"
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.634Z serialport/bindings/unixRead Starting read
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.635Z serialport/bindings/unixRead Finished read 6 bytes
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.635Z serialport/stream binding.read finished { bytesRead: 6 }
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.635Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.636Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.636Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.636Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.636Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.637Z serialport/stream _read reading { start: 4531, toRead: 61005 }
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.637Z serialport/binding-abstract read
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.637Z serialport/bindings/unixRead Starting read
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.638Z serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
mqtt zigbee2mqtt-0 zigbee2mqtt   errno: -11,
mqtt zigbee2mqtt-0 zigbee2mqtt   code: 'EAGAIN',
mqtt zigbee2mqtt-0 zigbee2mqtt   syscall: 'read'
mqtt zigbee2mqtt-0 zigbee2mqtt }
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.638Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:03.638Z serialport/bindings/poller Polling for "readable"
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.517Z serialport/bindings/poller received "readable"
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.518Z serialport/bindings/unixRead Starting read
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.518Z serialport/bindings/unixRead Finished read 8 bytes
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.518Z serialport/stream binding.read finished { bytesRead: 8 }
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.518Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,72,224,0,42]
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.518Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,72,224,0,42]
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.518Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [72,224,0] - 42
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.518Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":57416,"relaycount":0,"relaylist":[]}
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.519Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.519Z serialport/stream _read reading { start: 4539, toRead: 60997 }
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.519Z serialport/binding-abstract read
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.519Z serialport/bindings/unixRead Starting read
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.520Z serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
mqtt zigbee2mqtt-0 zigbee2mqtt   errno: -11,
mqtt zigbee2mqtt-0 zigbee2mqtt   code: 'EAGAIN',
mqtt zigbee2mqtt-0 zigbee2mqtt   syscall: 'read'
mqtt zigbee2mqtt-0 zigbee2mqtt }
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:04.520Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
[... 277 lines cut ...]
mqtt zigbee2mqtt-0 zigbee2mqtt }
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:14.268Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:14.268Z serialport/bindings/poller Polling for "readable"
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:24.247Z zigbee-herdsman:controller:device Interview - failed for device '0x00158d0001d7f109' with error 'Error: Timeout - 62939 - 1 - 30 - 0 - 1 after 10000ms
mqtt zigbee2mqtt-0 zigbee2mqtt     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
mqtt zigbee2mqtt-0 zigbee2mqtt     at listOnTimeout (internal/timers.js:531:17)
mqtt zigbee2mqtt-0 zigbee2mqtt     at processTimers (internal/timers.js:475:7)'
mqtt zigbee2mqtt-0 zigbee2mqtt 2020-02-03T11:51:24.249Z zigbee-herdsman:controller:error Interview failed for '0x00158d0001d7f109 with error 'Error: Timeout - 62939 - 1 - 30 - 0 - 1 after 10000ms'
mqtt zigbee2mqtt-0 zigbee2mqtt zigbee2mqtt:error 2020-02-03 12:51:24: Failed to interview '0x00158d0001d7f109', device has not successfully been paired
mqtt zigbee2mqtt-0 zigbee2mqtt zigbee2mqtt:info  2020-02-03 12:51:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00158d0001d7f109"}}'
```
</details>

## What did you expect to happen

I'd expect that the interview succeeds. I tried it several times (5 times), but I always received the same error.

## How to reproduce it (minimal and precise)

Just try to pair the mentioned LED panel.

## Debug Info

zigbee2mqtt version: 1.9.0 (commit #287f843)
CC253X firmware version: CC2531_DEFAULT_20190608.zip

Koenkk, 2020-02-03:
Probably already solved in #2677 

Should work fine with the latest dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

rhuss, 2020-02-03:
Cool, let me try that later ... thx !

rhuss, 2020-02-03:
With the latest developer version (running from docker, tag: "latest-dev") everything runs smoothely.

Thanks  a lot also for that awesome project !

closed 2020-02-03 by rhuss