Koenkk/zigbee2mqtt #2889: Log filename

svh1985, 2020-02-03:
Hi there,

I want to request an enhancement for the log settings. It is to allow to set the name of the log file.

Something like:
```
  log_level: info
  log_directory: /var/log/
  log_filename: zigbee2mqtt.log # also allowes zigbee2mqtt_%TIMESTAMP%.log
  log_output:
    - console
    - file
```

This gives some extra flexibility for the log file. Especially handy for external log rotate scrips.

Koenkk, 2020-02-04:
Sounds good, could you make the pr?

svh1985, 2020-02-04:
Cool! Just wondering, is there a rationale behind the fact that the log is a .txt file and not a .log? Maybe important for the implementation

Koenkk, 2020-02-04:
There is not rationale behind this.

Koenkk, 2020-02-05:
Merged, thanks!

closed 2020-02-05 by Koenkk