Koenkk/zigbee2mqtt #2890: Inaccurate Link Quality Value in Networkmap

myiotcenter, 2020-02-04:
# Bug Report
## What happened
I'm setting up a new Zigbee network at home, my current setup is using CC2538 + CC2592 Coordinator with two DL-22 CC2530 + CC2591 Routers. The Coordinater is connected to a Wemos D1 Mini through UART and I connected the zigbee2mqtt hassio add-on through tcp connection.

I connected few Xiaomi sensors and everything seems to be ok but whenever I generated a network map I can see the linkquality of few sensor is far below the value I can see in the mqtt messages or in HomeAssistant sensor value. 

I connected 1 sensor directly to Coordinator and the value is more accurate compared to sensors connected to router.

This is my networkmap : https://imgur.com/a/pwENLPs
I highlighted the linkquality values which are far below the values shown in HomeAssistant and mqtt messages. There are some value differences compared to the log due to time difference in generating map and logs.

I set `"zigbee_herdsman_debug": true` in add-on configuration and then generate raw networkmap and can see that LQI succeeded for Coordinator and two routers.

These are the values showed in HomeAssistant Sensors:
https://imgur.com/IxXlUVf
https://imgur.com/YIhdmUu
https://imgur.com/gMmZ6R2

Here's the log when requesting raw networkmap and zigbee_herdsman_debug set to true : https://pastebin.com/HQUhDR5F

Here's the generated raw networkmap only: https://jsoneditoronline.org/?id=4bd436cb721e4897b1fdba2091128f01

## What did you expect to happen
Generated Network Map is showing linkquality value closed to what I can see from mqtt message or sensor value in HomeAssistant.

## Debug Info
zigbee2mqtt version: hassio add-on 1.9.0
CC253X firmware version:
I flashed the coordinator with the firmware for CC2538 + CC2592 by @reverieline (MODKAMRU_V3_UART-no-flow-control.hex)

and the router is using firmware by @ptvoinfo 
https://github.com/Koenkk/zigbee2mqtt/issues/1362#issuecomment-499373874

## Questions
1. Is this an issue on zigbee2mqtt side or issue in coordinator / router firmware?
2. Any other things I can set / try to identify the root cause? 

I've been doing fresh install of the add-on, reflash the firmwares of coordinator & routers, etc. but couldn't make it right.

Appreciate your input and advice on this issue. Thanks.

Koenkk, 2020-02-04:
To figure out where the values are coming from, sniff the traffic while executing the network scan: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

nanobra1n, 2020-02-04:
The linkquality in the mqtt message (the ones homeassistant uses) is only the linkquality for the last hop (between router and coordinator).  Which explains that once you connect your sensor directly to the coordinator, you get values that agree with your map.

myiotcenter, 2020-02-04:
> To figure out where the values are coming from, sniff the traffic while executing the network scan: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Let me setup the sniffer and see what I can get...

> The linkquality in the mqtt message (the ones homeassistant uses) is only the linkquality for the last hop (between router and coordinator). Which explains that once you connect your sensor directly to the coordinator, you get values that agree with your map.

I was previously testing my zigbee network setup using CC1352P2 coordinator + 2 CC2530 with antenna routers. I can see low linkquality between coordinator <-> router probably due to location / placement of the routers to coordinator or tx/rx power of CC25350 but end devices which I placed near the router are showing high linkquality values, close to what being showed in HomeAssistant sensors & mqtt messages.

myiotcenter, 2020-02-04:
Hi @Koenkk,
This is the sniffed traffic while executing raw network scan... 
https://pastebin.com/KLxiJC82

Koenkk, 2020-02-05:
@myiotcenter the textual form is a bit hard to read, but if you search for: `ZigBee Device Profile, Link Quality Response, Status: Success`  you see that the showed values of zigbee2mqtt are correct:

```
        Table Entry
            Extended Pan: TexasIns_00:05:a9:90:1d (00:12:4b:00:05:a9:90:1d)
            Extended Address: TexasIns_00:05:a9:90:1d (00:12:4b:00:05:a9:90:1d)
            Addr: 0x0000
            .... ..00 = Type: Coordinator (0)
            .... 10.. = Idle Rx: Unknown (2)
            .000 .... = Relationship: Parent (0)
            .... ..10 = Permit Joining: Unknown (2)
            Depth: 0
            LQI: 1   <------------------------
```

myiotcenter, 2020-02-05:
@Koenkk Thanks for pointing out the important parts of the sniffed traffic... 
I searched specific for that parts and found 4 occurrences of `ZigBee Device Profile, Link Quality Response, Status: Success`. 

Tried to make a summary in form of table for those 4 occurrences 
[image]

Looking at the pattern I would expect that there are records saying that a router is a parent of end device but from the log they are showing that Coordinator is the parent for all end device? I would also expect that relationship between router and coordinator is child but somehow both router showing the `Relationship` as `None`. Are these values all as expected?

My house is 2 storey building and quite wide around +/- 17m wide so I need to make sure that I'm placing Coordinator & Routers in the effective positions and expecting network map can help me to achieve that. I assume having CC2538 + CC2592 (or my CC1352P2) as coordinator and 2 CC2530 + CC2591 would be powerful enough to cover the whole house, but I need some reliable measurement.

This is a non production setup, more to a proof of concept exercise before I deploy sensors, switches, led driver, etc. So far sensors work as expected, Button & Window Sensor are triggering mqtt messages as expected and status in Home Assistant also changing in a timely manner.

1. Any idea what will be the best approach to measure the coverage of my zigbee network other than expecting accurate LQI in network map? 

2. Is there any other thing I can check to narrow down the root cause? not sure if this is an issue in zigbee2mqtt, coordinator or router firmwares?

Appreciate your feedback @Koenkk @ptvoinfo @reverieline

---------------------------------------------------------
I removed other lines and assume these lines are the most interesting parts and move the values to the table above..
```
ZigBee Device Profile, Link Quality Response, Status: Success
    Sequence Number: 22
    Status: Success (0)
    Table Size: 3
    Index: 0
    Table Count: 3
    Neighbor Table
        Table Entry
            Extended Pan: TexasIns_00:05:a9:90:1d (00:12:4b:00:05:a9:90:1d)
            Extended Address: TexasIns_00:05:a9:90:1d (00:12:4b:00:05:a9:90:1d)
            Addr: 0x0000
            .... ..00 = Type: Coordinator (0)
            .... 10.. = Idle Rx: Unknown (2)
            .000 .... = Relationship: Parent (0)
            .... ..10 = Permit Joining: Unknown (2)
            Depth: 0
            LQI: 1
        Table Entry
            Extended Pan: TexasIns_00:05:a9:90:1d (00:12:4b:00:05:a9:90:1d)
            Extended Address: Jennic_00:03:3a:e0:fa (00:15:8d:00:03:3a:e0:fa)
            Addr: 0x4836
            .... ..10 = Type: End Device (2)
            .... 00.. = Idle Rx: False (0)
            .001 .... = Relationship: Child (1)
            .... ..10 = Permit Joining: Unknown (2)
            Depth: 2
            LQI: 26
        Table Entry
            Extended Pan: TexasIns_00:05:a9:90:1d (00:12:4b:00:05:a9:90:1d)
            Extended Address: TexasIns_00:1c:03:a6:f9 (00:12:4b:00:1c:03:a6:f9)
            Addr: 0x599f
            .... ..01 = Type: Router (1)
            .... 10.. = Idle Rx: Unknown (2)
            .011 .... = Relationship: None (3)
            .... ..10 = Permit Joining: Unknown (2)
            Depth: 255
            LQI: 1

ZigBee Device Profile, Link Quality Response, Status: Success
    Sequence Number: 23
    Status: Success (0)
    Table Size: 7
    Index: 0
    Table Count: 3
    Neighbor Table
        Table Entry
            Extended Pan: TexasIns_00:05:a9:90:1d (00:12:4b:00:05:a9:90:1d)
            Extended Address: TexasIns_00:05:a9:90:1d (00:12:4b:00:05:a9:90:1d)
            Addr: 0x0000
[... 59 lines cut ...]
            .... 00.. = Idle Rx: False (0)
            .001 .... = Relationship: Child (1)
            .... ..10 = Permit Joining: Unknown (2)
            Depth: 2
            LQI: 1

ZigBee Device Profile, Link Quality Response, Status: Success
    Sequence Number: 25
    Status: Success (0)
    Table Size: 7
    Index: 6
    Table Count: 1
    Neighbor Table
        Table Entry
            Extended Pan: TexasIns_00:05:a9:90:1d (00:12:4b:00:05:a9:90:1d)
            Extended Address: TexasIns_00:1c:03:48:32 (00:12:4b:00:1c:03:48:32)
            Addr: 0x9d9c
            .... ..01 = Type: Router (1)
            .... 10.. = Idle Rx: Unknown (2)
            .011 .... = Relationship: None (3)
            .... ..10 = Permit Joining: Unknown (2)
            Depth: 255
            LQI: 1

```

Koenkk, 2020-02-06:
> Any idea what will be the best approach to measure the coverage of my zigbee network other than expecting accurate LQI in network map?

Unfortunately not, however I wouldn't worry to much about the LQI of the end devices, just make sure all of them have a router within +- 5-8 meters. For router check below

> Is there any other thing I can check to narrow down the root cause? not sure if this is an issue in zigbee2mqtt, coordinator or router firmwares?

This is the strangest thing about your network map, the routers show a linkquality of 1 to each other. Can you try power cycling the routers and see if this solves the  issue?

> I realize that temp sensors are not providing reports regularly, some sensors are reporting in more frequent than others. Is this normal? Will they only report if detecting changes in Temperature or Humidity?

In order to save battery, they will only report when there is a significant change

The following "rules of thumb" can be applied in order to achieve a reliable and scalable network:
- Use a powerful coordinator, either CC2652R or CC1352-P 
- Have a router on every floor of your house, preferably a strong one like the CC2530 as they have an external antenna, non-strong ones are e.g. TRADFRI devices
- Make sure there is a router within 5-8 meters of every end device
- Make sure there is a router every +- 10 meters

myiotcenter, 2020-02-06:
Hi @Koenkk ,
Thanks for your explanations. I will try to power cycle my routers and see if things are getting better. 
I have few more DL22 stocks I ordered from Taobao so will see if I can make a good mesh with them, 

Will provide further update. Thanks.

myiotcenter, 2020-02-08:
Hi,
I tried to power cycle coordinator and routers but things are remain the same. Since everything works as expected I guess I need to move on and proceed with the proof of concept and eventually roll out my prod setup. Thanks for your help @Koenkk !

closed 2020-02-08 by myiotcenter