Koenkk/zigbee2mqtt #2891: Can't start zigbee herdsman

lancelon, 2020-02-04:
Any ideas what's causing this? I'm running Zigbee2mqtt on hassio, previously using the edge addon. I had to powercycle the Pi yesterday evening and zigbee hasn't worked since. I've tried restoring from a snapshot but same result. Any help much appreciated!!

```
zigbee2mqtt:info  2020-02-04 07:47:46: Starting zigbee-herdsman...
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.9.0-dev start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR!
 Failed at the zigbee2mqtt@1.9.0-dev start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
zigbee2mqtt:error 2020-02-04 07:47:53: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-02-04 07:47:53: Failed to start zigbee
zigbee2mqtt:error 2020-02-04 07:47:53: Exiting...
zigbee2mqtt:error 2020-02-04 07:47:53: SyntaxError: Unexpected token  in JSON at position 191
    at JSON.parse (<anonymous>)
    at Function.open (/app/node_modules/zigbee-herdsman/dist/controller/database.js:17:35)
    at Controller.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:88:48)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:5:58)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)

```

Koenkk, 2020-02-04:
Can you share your `database.db`?

lancelon, 2020-02-04:
**Timeline:**
15:00 yesterday - Pi/hass/hassio froze and never recovered. 
17:00 yesterday - got home, had to powercycle the Pi. Came up again within a few minutes.
17:05 yesterday - everything working apart from zigbee2mqtt, addon wasn't started/wouldn't start.

It's coincidental that I was looking into debugging my database.db yesterday as I hadn't actually made any changes to it, I was _going to_ but was thwarted by the Pi freezing at 15:00...

There was [a discussion yesterday](https://community.home-assistant.io/t/any-zigbee2mqtt-experts-out-there/168638?u=daneboom) about how my database shouldn't be split down the middle like that and shouldn't have duplicates. Thing is, apart from refusing to accept new devices, it's been working like this for months. I think the powercycling I was forced to do may have corrupted the CC2531? 

```
{"id":13,"type":"EndDevice","ieeeAddr":"0x00158d000288e932","nwkAddr":15095,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion","epList":[1],"zclVersion":1,"appVersion":11,"stackVersion":2,"hwVersion":2,"swBuildId":"3000-0001","status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"dateCode":"20160524","_id":"16JBxDgAfeknSuhY"}
{"id":7,"type":"EndDevice","ieeeAddr":"0x00158d00033b1277","nwkAddr":39317,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sens","epList":[1,2,3],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25,65535],"outClusterList":[0,3,4,5,18,25,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[3,4,5,18],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[3,4,5,12],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{}}}}},"_id":"1Mo0rYTEwRUr7OOL"}
{"id":14,"type":"EndDevice","ieeeAddr":"0x00158d0002f08e74","nwkAddr":44100,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"4tzaUStGjxDojE4D"}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d0002275601","nwkAddr":42429,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"Ai17cYZSP42bFcIX"}
{"id":6,"type":"EndDevice","ieeeAddr":"0x00158d000288e9ef","nwkAddr":44838,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"BkIFcTLmZtaILhuw"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001936af5f","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1573921698,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"dev [... cut ...] d":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"UA3oPwYe85qmTfhW"}

```

Koenkk, 2020-02-04:
It indeed shouldn't be split, this is something from zigbee2mqtt 1.6.0 and before

Try replacing the contents of `database.db` with:

```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001936af5f","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1573922747,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"UA3oPwYe85qmTfhW"}
{"id":10,"type":"EndDevice","ieeeAddr":"0x00158d0002f24fb9","nwkAddr":64512,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"trt6jN2ZK6PMl2Gi"}
{"id":11,"type":"EndDevice","ieeeAddr":"0x00158d0002eabfb4","nwkAddr":52269,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"fcFNsq3SrJEIfzDH"}
{"id":12,"type":"EndDevice","ieeeAddr":"0x00158d00031e0be4","nwkAddr":59162,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sens","epList":[1,2,3],"zclVersion":1,"appVersion":2,"stackVersion":2,"hwVersion":30,"swBuildId":"3000-0001","status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25,65535],"outClusterList":[0,3,4,5,18,25,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[3,4,5,18],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[3,4,5,12],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{}}}}},"dateCode":"20160516","_id":"nyKHSaRDu1dpQtbz"}
{"id":13,"type":"EndDevice","ieeeAddr":"0x00158d000288e932","nwkAddr":15095,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion","epList":[1],"zclVersion":1,"appVersion":11,"stackVersion":2,"hwVersion":2,"swBuildId":"3000-0001","status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"dateCode":"20160524","_id":"16JBxDgAfeknSuhY"}
{"id":14,"type":"EndDevice","ieeeAddr":"0x00158d0002f08e74","nwkAddr":44100,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"4tzaUStGjxDojE4D"}
[... cut ...] ":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"dj1YCUfcOTgnflsU"}
```

lancelon, 2020-02-05:
Thanks. I restored from a snapshot and database.db now looks like this:-

It's nothing to do with the fact that the database id reaches 15 is it that zigbee2mqtt currently won't add any new devices? Where is the check made to see whether the coordinator has reached max children? Presumably it's smart enough to see  that there's no <id> 8 and therefore in theory a gap? Cheers!

```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001936af5f","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1573922747,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"UA3oPwYe85qmTfhW"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d00022755ee","nwkAddr":45466,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attributes":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3005},{"elmType":33,"elmVal":17320},{"elmType":36,"elmVal":[0,2]},{"elmType":33,"elmVal":7},{"elmType":32,"elmVal":97}]}},"genIdentify":{"dir":{"value":3},"attributes":{}},"genGroups":{"dir":{"value":2},"attributes":{}},"genScenes":{"dir":{"value":2},"attributes":{}},"genOnOff":{"dir":{"value":2},"attributes":{}},"genLevelCtrl":{"dir":{"value":2},"attributes":{}},"genOta":{"dir":{"value":3},"attributes":{}},"manuSpecificCluster":{"dir":{"value":1},"attributes":{}},"msOccupancySensing":{"attributes":{"occupancy":1}}},"binds":[]}},"interviewCompleted":true,"meta":{},"lastSeen":1580856168321}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d0002091548","nwkAddr":7988,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attributes":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":3035},{"elmType":33,"elmVal":5032},{"elmType":36,"elmVal":[0,5]},{"elmType":33,"elmVal":12},{"elmType":32,"elmVal":102}]}},"genIdentify":{"dir":{"value":3},"attributes":{}},"genGroups":{"dir":{"value":2},"attributes":{}},"genScenes":{"dir":{"value":2},"attributes":{}},"genOnOff":{"dir":{"value":2},"attributes":{}},"genLevelCtrl":{"dir":{"value":2},"attributes":{}},"genOta":{"dir":{"value":3},"attributes":{}},"manuSpecificCluster":{"dir":{"value":1},"attributes":{}},"msOccupancySensing":{"attributes":{"occupancy":1}}}}},"_id":"YSnkIszWI89JvX0E"}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d0002275601","nwkAddr":42429,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attributes":{}},"genIdentify":{"dir":{"value":3},"attributes":{}},"genGroups":{"dir":{"value":2},"attributes":{}},"genScenes":{"dir":{"value":2},"attributes":{}},"genOnOff":{"dir":{"value":2},"attributes":{}},"genLevelCtrl":{"dir":{"value":2},"attributes":{}},"genOta":{"dir":{"value":3},"attributes":{}},"manuSpecificCluster":{"dir":{"value":1},"attributes":{}}},"binds":[]}},"interviewCompleted":true,"meta":{},"lastSeen":1580860475044}
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d00029b6a8f","nwkAddr":13318,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attributes":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":2965},{"elmType":33,"elmVal":17320},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":31},{"elmType":32,"elmVal":92}]}},"genIdentify":{"dir":{"value":3},"attributes":{}},"genGroups":{"dir":{"value":2},"attributes":{}},"genScenes":{"dir":{"value":2},"attributes":{}},"genOnOff":{"dir":{"value":2},"attributes":{}},"genLevelCtrl":{"dir":{"value":2},"attributes":{}},"genOta":{"dir":{"value":3},"attributes":{}},"manuSpecificCluster":{"dir":{"value":1},"attributes":{}}}}},"_id":"ulbpSdmS11nS8xlG"}
{"id":6,"type":"EndDevice","ieeeAddr":"0x00158d000288e9ef","nwkAddr":44838,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attributes":{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":29 [... cut ...] mType":36,"elmVal":[2,16777224]},{"elmType":33,"elmVal":42},{"elmType":32,"elmVal":96}]}},"genIdentify":{"dir":{"value":3},"attributes":{}},"genGroups":{"dir":{"value":2},"attributes":{}},"genScenes":{"dir":{"value":2},"attributes":{}},"genOnOff":{"dir":{"value":2},"attributes":{"onOff":1}},"genLevelCtrl":{"dir":{"value":2},"attributes":{}},"genOta":{"dir":{"value":3},"attributes":{}},"manuSpecificCluster":{"dir":{"value":1},"attributes":{}}}}},"dateCode":"20150424","_id":"fSKDoEbxkeo32eOu"}
```

Koenkk, 2020-02-05:
Looks good

> It's nothing to do with the fact that the database id reaches 15 is it that zigbee2mqtt currently won't add any new devices? 

It cannot easily be checked, but as you only have battery powered devices they have no choice other than the coordinator.

You can find the direct children limit of each firmware here: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

In you case I would definitely recommend adding routers to your network.

lancelon, 2020-02-07:
Well, I tried flashing a second CC2531 as a router and enabling adding devices. Absolutely no attempts at pairing between the coordinator and the router. And, actually, this makes perfect sense. If I'm not able to add any new devices to my existing coordinator, why would it be possible to add a router? So what I've done is installed zigbee2mqtt on another Raspberry Pi and am hoping to run two instances of z2m on my LAN. I don't want to have to go through the hassle of reflashing (and thus repairing) because some of my sensors are buried in the walls! **Quick question, though, should I use the same mqtt topic for both instances of Z2M?**

Koenkk, 2020-02-07:
> Quick question, though, should I use the same mqtt topic for both instances of Z2M?

No, use a different one otherwise both instances will try to execute the commands.

nickman80, 2020-02-17:
Hi Guys,
sorry to bother you, but i have been stuck for couple days with zigbee herdsman, would appreciate for any help, this is my log [https://pastebin.com/i30b2kYH](url)
thank you

closed 2020-04-25 by stale[bot]