Koenkk/zigbee2mqtt #2892: Danalock V3 Interview Failed on v1.9

Stewart86, 2020-02-04:
# Bug Report
## What happened
I was trying to pair Danalock BT/ZB model for many tries now. At one point, I got it show on the device list on home assistant, but I was unable able to control it. 

Many attempts later, I reflash my CC2531 with 20190619 firmware, update danalock firmware to the latest as of today. updated zigbee2mqtt to 1.9 and tried many more re-pairing, and factory reset to no avail.

### Things I've Tried
- Also, I have tried manually config with `zigbee2mqtt/bridge/config`
- remove and re pair device with `zigbee2mqtt/bridge/config/remove`
- and `zigbee2mqtt/bridge/config/force_remove`
- remove battery 
- pair with iOS app and use connect to gateway from the App

### Side note
Danalock is working perfectly with bluetooth!

## What did you expect to happen
able to control danalock with mqtt

## How to reproduce it (minimal and precise)
Pairing with CC2531 

## Debug Info
zigbee2mqtt version: 1.9 
CC253X firmware version: 20190619 

## Log (minimal)
```
  zigbee-herdsman:controller:device Interview - active endpoints request failed for '0x000b57fffe59d658', attempt 2 +10s
  zigbee-herdsman:controller:device Interview - failed for device '0x000b57fffe59d658' with error 'Error: Interview failed because can not get active endpoints ('0x000b57fffe59d658')
  zigbee-herdsman:controller:device     at Device.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:273:23)
  zigbee-herdsman:controller:device     at Generator.throw (<anonymous>)
  zigbee-herdsman:controller:device     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:6:65)' +1ms
  zigbee-herdsman:controller:error Interview failed for '0x000b57fffe59d658 with error 'Error: Interview failed because can not get active endpoints ('0x000b57fffe59d658')' +0ms
zigbee2mqtt:error 2020-02-04 23:02:13: Failed to interview '0x000b57fffe59d658', device has not successfully been paired
zigbee2mqtt:info  2020-02-04 23:02:13: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x000b57fffe59d658"}}'
```
## Log (full-ish) 
Too many devices in my network I don't know if I got everything.

[https://pastebin.com/zfFCfzb2](https://pastebin.com/zfFCfzb2)

Koenkk, 2020-02-04:
- Please try again with the latest zigbee2mqtt dev branch (has pairing fixes)
- If that doesn't fix it, provide the log when pairing the device and having herdsman debug logging enabled

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Stewart86, 2020-02-05:
Same issue after trying with the dev branch.

```
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +8ms
  zigbee-herdsman:controller:device Interview - active endpoints request failed for '0x000b57fffe59d658', attempt 2 +10s
  zigbee-herdsman:controller:device Interview - failed for device '0x000b57fffe59d658' with error 'Error: Interview failed because can not get active endpoints ('0x000b57fffe59d658')
  zigbee-herdsman:controller:device     at Device.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:275:23)
  zigbee-herdsman:controller:device     at Generator.throw (<anonymous>)
  zigbee-herdsman:controller:device     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/device.js:6:65)' +1ms
  zigbee-herdsman:controller:error Interview failed for '0x000b57fffe59d658 with error 'Error: Interview failed because can not get active endpoints ('0x000b57fffe59d658')' +0ms
zigbee2mqtt:error 2020-02-05 22:05:36: Failed to interview '0x000b57fffe59d658', device has not successfully been paired
zigbee2mqtt:info  2020-02-05 22:05:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x000b57fffe59d658"}}'
```

There were 6 attempts at this function if it helps debugging.
```
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x000b57fffe59d658', attempt 6 +10s
```

full log [here](https://pastebin.com/d6ts1jWy)

Koenkk, 2020-02-05:
It seems that zigbee2mqtt cannot communicate with the device at all, I remember that for this type of device people managed to solve it by:
- Pairing it closer to the coordinator
- Connecting the CC2531 via an USB extension cable (to avoid interference)
- Replacing the batteries of the danalock.

Stewart86, 2020-02-05:
Yes you are right! I finally managed to get it paired by using an USB extension cable and putting the device next to it.

If this is a common issue with this device, maybe we can add a note on the device page for supported device.

I have made a [PR](https://github.com/Koenkk/zigbee2mqtt.io/pull/209)

closed 2020-02-05 by Stewart86