Koenkk/zigbee2mqtt #2893: Coordinator failed to start, panID is already in use after re-flashing CC1352P_2

alexsporn, 2020-02-04:
# Bug Report
## What happened
I re-flashed my CC1352P_2 with the latest firmware due to having connection issues in the HASS.io zigbee2mqtt add-on (see further below for more details). I copied over all data files and now when I start zigbee2mqtt from my computer to debug this issue I get the following error:
```zigbee2mqtt:info  2020-02-04 19:12:44: Starting zigbee2mqtt version 1.9.0-dev (commit #a07a28c)
zigbee2mqtt:info  2020-02-04 19:12:44: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-02-04 19:12:44: Using zigbee-herdsman with settings: '{"network":{"panID":6756,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/Users/alexsporn/zigbee2mqtt/data/database.db","databaseBackupPath":"/Users/alexsporn/zigbee2mqtt/data/database.db.backup","backupPath":"/Users/alexsporn/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/tty.usbmodemL43001631"}}'
zigbee2mqtt:error 2020-02-04 19:13:51: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-02-04 19:13:51: Failed to start zigbee
zigbee2mqtt:error 2020-02-04 19:13:51: Exiting...
zigbee2mqtt:error 2020-02-04 19:13:51: Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel
    at /Users/alexsporn/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:164:23
    at Generator.throw (<anonymous>)
    at rejected (/Users/alexsporn/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:6:65)
```

I re-flashed the coordinator, because my homeassistant was showing timeouts when starting. This can also be reproduced from my computer after the error above occurs and I try to start it again:

```
zigbee2mqtt:info  2020-02-04 19:19:52: Starting zigbee2mqtt version 1.9.0-dev (commit #a07a28c)
zigbee2mqtt:info  2020-02-04 19:19:52: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-02-04 19:19:52: Using zigbee-herdsman with settings: '{"network":{"panID":6756,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/Users/alexsporn/zigbee2mqtt/data/database.db","databaseBackupPath":"/Users/alexsporn/zigbee2mqtt/data/database.db.backup","backupPath":"/Users/alexsporn/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/tty.usbmodemL43001631"}}'
zigbee2mqtt:error 2020-02-04 19:20:55: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-02-04 19:20:55: Failed to start zigbee
zigbee2mqtt:error 2020-02-04 19:20:55: Exiting...
zigbee2mqtt:error 2020-02-04 19:20:55: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout._onTimeout (/Users/alexsporn/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
```

This timeout error will keep ocurring until I reset the board by pressing the reset button in the back.

## What did you expect to happen

I expected the coordinator to start with the saved settings and continue working with the paired devices.

I also expect that starting zigbee2mqtt will properly reset the board on initial connection.

Do I need to start from scratch again and repair all devices?

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.9.0 and 1.9.0-dev (commit #a07a28c)
CC1352P_2 firmware version: CC1352P_2_20191106.hex

alexsporn, 2020-02-04:
Ok, after reading https://github.com/Koenkk/zigbee-herdsman/issues/113 I tried starting my CC1352P_2 without the SMA antenna (to reduce the signal strength) and now it started 🙃

CodeFinder2, 2020-02-04:
I was faced with the a similar situation (needed to reflash my CC1352P-2 and than was not able to start, took the 're-pair way').

Isn't this related to security things of ZigBee 3.0 and the fact that some devices (previously paired and still powered) are still sending/broadcasting with the existing/previous PAN ID? And the coordinator picks up those signals and refrains starting with that error msg? So, somewhat 'unfortunately', a normal case / correct behavior? 🙈

imdos, 2020-02-07:
Seems indeed the same as this: 
https://github.com/Koenkk/zigbee2mqtt/issues/2795#issuecomment-577420423

timdonovanuk, 2020-02-21:
I tried wrapping my CC1352P-2 in foil then placing it in a metal box. I guess the antenna is so good, it's still detecting an existing PAN ID :( How is there not a way to force it to rejoin a network?! Is it perhaps possible while writing the firmware to the device to force a pan ID?

CodeFinder2, 2020-02-22:
Simply detach the antenna?

timdonovanuk, 2020-02-22:
Can't, it's a PCB trace antenna, unless you've modded the SMA port to work (I haven't) :)

CodeFinder2, 2020-02-22:
Ah sorry, yes I did that. And disconnecting the antenna worked for me (I was having this issue a 2nd time now while I was modifying zigbee2mqtt hassio addon config).

Another +1 for the antenna mod 😂

CodeFinder2, 2020-02-22:
I think this is a serious issue in either the firmware or in z2m. Adding my log for debugging here:
```text
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-02-13T19:30:30: PM2 log: Launching in no daemon mode
2020-02-13T19:30:30: PM2 log: App [npm:0] starting in -fork mode-
2020-02-13T19:30:30: PM2 log: Aapp [npm:0] online
> zigbee2mqtt@1.10.0 start /zigbee2mqtt-1.10.0
> node index.js
zigbee2mqtt:info  2020-02-13 19:30:33: Logging to console and directory: 'data/log/2020-02-13.19-30-33' filename: log.txt
zigbee2mqtt:info  2020-02-13 19:30:34: Starting zigbee2mqtt version 1.10.0 (commit #unknown)
zigbee2mqtt:info  2020-02-13 19:30:34: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-02-13 19:31:40: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-02-13 19:31:40: Failed to start zigbee
zigbee2mqtt:error 2020-02-13 19:31:40: Exiting...
zigbee2mqtt:error 2020-02-13 19:31:40: Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel
    at /zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:164:23
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:6:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.10.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.10.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-13T18_31_40_150Z-debug.log
2020-02-13T19:31:43: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-02-13T19:31:43: PM2 log: App [npm:0] starting in -fork mode-
2020-02-13T19:31:43: PM2 log: App [npm:0] online
> zigbee2mqtt@1.10.0 start /zigbee2mqtt-1.10.0
> node index.js
zigbee2mqtt:info  2020-02-13 19:31:46: Logging to console and directory: 'data/log/2020-02-13.19-31-46' filename: log.txt
zigbee2mqtt:info  2020-02-13 19:31:47: Starting zigbee2mqtt version 1.10.0 (commit #unknown)
zigbee2mqtt:info  2020-02-13 19:31:47: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-02-13 19:32:48: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-02-13 19:32:48: Failed to start zigbee
zigbee2mqtt:error 2020-02-13 19:32:48: Exiting...
zigbee2mqtt:error 2020-02-13 19:32:48: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.10.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.10.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-13T18_32_48_739Z-debug.log
2020-02-13T19:32:49: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-02-13T19:32:49: PM2 log: App [npm:0] starting in -fork mode-
2020-02-13T19:32:49: PM2 log: App [npm:0] online
> zigbee2mqtt@1.10.0 start /zigbee2mqtt-1.10.0
> node index.js
zigbee2mqtt:info  2020-02-13 19:32:52: Logging to console and directory: 'data/log/2020-02-13.19-32-52' filename: log.txt
zigbee2mqtt:info  2020-02-13 19:32:53: Starting zigbee2mqtt version 1.10.0 (commit #unknown)
zigbee2mqtt:info  2020-02-13 19:32:53: Starting zigbee-herdsman...
```

(As already written, I _somewhat_ got my setup working by disconnecting my external antenna. However, this did not work for my Philips Hue bulbs. That's why I finally decided to repair everything. Now everything is working again. I started to have this issue when changing the network_key in Z2M's hassio addon from the `network_key` to the new `network_key_string` variant to place it in a secret. Nice to have but...(I should) never touch a running system...)

Koenkk, 2020-02-23:
@CodeFinder2 zigbee2mqtt will only reset the CC1352P-2 when the configuration has changed. I think by changing `network_key` to `network_key_string` caused a (non-desired) change in the network key being used, therefore zigbee2mqtt will reset the CC1352P-2.

CodeFinder2, 2020-02-24:
@Koenkk ah, thanks for clarifying! And by 'reset' you mean it requires a new pan ID and repairing, so this is intended behavior? And if reflashing is 'necessary' (e.g., due to pairing and/or connection issues as frequently reported here), it will also always mean repairing all devices (if the 'antenna disconnect trick' does not work)? Would it be possible to be less restrictive of when the coordinator needs to be reset or does it may cause other trouble? 🙈 

(Of course, I indented to use the *same* network key, but just passed via `network_key_string` now.)

Koenkk, 2020-02-24:
Repairing is only necessary when e.g. changing the `network_key` (see https://www.zigbee2mqtt.io/information/FAQ.html#what-does-and-does-not-require-repairing-of-all-devices for more). I think by using `network_key_string` instead of `network_key` you accidentally provided a different network key to zigbee2mqtt (because I assume the format of them is different).

The coordinator will always reset when changing the network key because all devices on the network have to use the same key. A coordinator with a different network key than the other devices is not possible.

CodeFinder2, 2020-02-24:
Yes, thanks for these hints, I know about all that. 🙈 (And yes, I may have accidentally changed my network key due to the different format in network_key_string, my fault.) However, within this issue, it seems that (sometimes?) reflashing a CC1352P-2 requires repairing although the config hasn't changed, and that's weird, isn't it? I mean, why is this
> Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel

showing up?

Koenkk, 2020-02-24:
It's only required after reflashing and when there is no `coordinator_backup.json` present in the zigbee2mqtt data folder.

timdonovanuk, 2020-03-05:
@Koenkk can we not somehow control the panid when flashing the firmware? Why does the firmware need to refuse to become a coordinator it it detects an existing panid? Could the "wrapping it in foil" hack not be written as part of firmware instead (e.g. could the firmware simply ignore broadcasts while booting)?

Koenkk, 2020-03-05:
With zstack 3 its unfortunately not possible, this logic is inside the non open source part of the firmware.

timdonovanuk, 2020-03-07:
Ah fair enough :( Thanks anyway

magpern, 2020-03-10:
@Koenkk If the coordinater cannot start when an existing device sends the panid, then you can never reboot the computer it is connected to?
My CC1352 has run fine for many days, but yesterday I had to reboot. Just sent a `sudo reboot` and now zigbee2mqtt has refused to start.
I changed the panid in the config file, then it started again.

System becomes very vulnerable

Koenkk, 2020-03-11:
@magpern this will only be done after initial flash or reset, a reset can occur due to `pan_id`, `channel` or `network_key` has been changed in the `configuration.yaml`.

magpern, 2020-03-11:
@Koenkk well. I did change the config. I entered MQTT username and MQTT password. Didn't change anything else. Used nano editor.
Everything runs in docker.
Did a sudo reboot.

This was two days ago. Cc1352 still refuses to start.
It has worked for 4-5 weeks, and the computer has restarted a few times during this period.

I flashed the latest firmware yesterday, but no luck.

I managed to get it going while running in development environment on another computer, also ubuntu. It started working after changing the panid.

I moved the device back to the server, tried same panid, but no.. still no reaponse from device (it does not complain about panid, it just says it times out after 6000ms)

Koenkk, 2020-03-11:
@magpern that indicated that either the connection is not correct or the firmware somehow got corrupted (try reflashing: https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html)

magpern, 2020-03-11:
@Koenkk well, I did reflash. And the rest is docker (that worked fine the hour before it didn't).
I'll try and move it to a Pi or something to see if I get it going

runningman84, 2020-03-11:
I have a similar experience after some reboot or crash after a few weeks you get the pan Id error. I have not found any solution but changing the pan Id and repair all 30 devices.

magpern, 2020-03-11:
> I have a similar experience after some reboot or crash after a few weeks you get the pan Id error. I have not found any solution but changing the pan Id and repair all 30 devices.

@runningman84 mmm... it's just that changing the panid makes no differance. It still refuses to open the connection to the device.

I run ubuntu 19.10 latest updates
Latest dev firmware (but problem started on the release version)
Docker to handle all containers.
More or less nothing else is installed on the computer.
It is an intel i7 cpu

runningman84, 2020-03-11:
Pan Id changing allow did not always help. Sometimes I had to reboot or reset the device too.

CodeFinder2, 2020-03-11:
> I have a similar experience after some reboot or crash after a few weeks you get the pan Id error. I have not found any solution but changing the pan Id and repair all 30 devices.

Yeah that's why I was asking a few post above here. I had the same issue, very annoying. 😕 (Although changing the PAN ID solved it with the requirement to repair all devices.)

magpern, 2020-03-12:
@CodeFinder2 @runningman84 @Koenkk 
I solved it in a very unconventional way, that didn't require re-pairing.

On my laptop, I could run the code directly from Visual Code. When I tried to connect to the CC1352, it refused. A few, cannot find device 6000ms timeout, but eventually I got the error saying panId already in use. Don't know exactly how, disconnected cc1352, reconnected, pushed reset etc. No settings on computer was touched.

When the PanId error was confirmed, I got into my car, with laptop and cc1352, drove a few streets, ran the code. And Voila!!! It set the panId.

Back in the house
I never tried to plug it back into my server again, but I configured a docker and moved all config files to a Orange Pi 3 I had laying around. And it worked. All devices are still there and Home Assistant works again.

Files I moved:
- configuration.yaml from original server
- database.db from original server
- coordinator_backuo.json created on the laptop that got cc1352 back online

timdonovanuk, 2020-03-12:
Hah this is the same as wrapping it in foil, although the antenna is so good this didn't work for me. I might have to take a drive with my stick (crazy..).

CodeFinder2, 2020-03-12:
@magpern haha, yeah I was somewhat successful in disconnecting my CC1352's external SMA antenna but the problem was that some devices (IIRC, mainly Philips bulbs) refused to work. Many other devices worked "quite well". However it wasn't perfect and that's why I finally took the repair way nevertheless. Hope your setup works fine now! * finger crossed * :see_no_evil:

magpern, 2020-03-12:
@CodeFinder2 Still using the onboard antenna, but I live in a wooden house so the internal antenna is good enough.
An alternative to taking a drive with the computer, is to cut the mains power of the house/apartment so all mains powered devices gets turned off (like in-wall mounted devices). Requires a laptop or a battery powered mini computer.
But, as all my tinkering is at night time, I didn't want to do that. When the power would go back on, all ikea lights turn on again on full power :)

closed 2020-05-20 by stale[bot]

murjam, 2020-11-01:
So, is there something to do to avoid this happening when re-flashing the firmware? Had today exactly the change issue with `CC1352P_2` after I upgraded the firmware to `20201026`.

Ended up changing the panID, but it was not that easy to re-pair all those ~60+ devices I have in the network. Not sure what I should do different when upgrading the FW next time :man_shrugging:

Koenkk, 2020-11-02:
@murjam is there a `coordinator_backup.json` file in your data directory? This contains the backup of your network and is restored automatically when you start Zigbee2MQTT on a fresh flashed coordinator.

murjam, 2020-11-02:
> @murjam is there a `coordinator_backup.json` file in your data directory? This contains the backup of your network and is restored automatically when you start Zigbee2MQTT on a fresh flashed coordinator.

There was/is. But I was unable to get it started, it was either failing with the "panID" message of this topic or being stuck at `Starting zigbee-herdsman...`.

The question for me is: what is the right procedure to upgrade firmware next time so I did not have the issue? 🙇‍♂️

Koenkk, 2020-11-02:
The right procedure is:

- Stop zigbee2mqtt
- Flash firmware on adapter
- Replug adapter if flashing firmware was done on the same system where Zigbee2MQTT runs on
- Start zigbee2mqtt

If this doesn't work please provide the herdsman debug logging of this so I can check where things go wrong.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

CodeFinder2, 2020-11-02:
> The right procedure is:
>
> * Stop zigbee2mqtt
> * Flash firmware on adapter
> * Replug adapter if flashing firmware was done on the same system where Zigbee2MQTT runs on
> * Start zigbee2mqtt
>
> If this [...]

Worked for me!

murjam, 2020-11-03:
> * Stop zigbee2mqtt
I guess I might have failed on the first step 🤔 

I'll try to be more precise next time. I think what I did was:
1. unplugged the adapter
2. plugged adapter and flashed firmware on different computer
3. replugged the adapter
4. restarted the whole machine that runs zigbee2mqtt

Underknowledge, 2021-01-13:
Thats prittey much the first search machine result when you search for the error, So I ll share my steps. 
other errors I got 
```log
Zigbee2MQTT:error 2021-01-13 10:36:59: Failed to call 'OnEvent' 'stop' (TypeError: Cannot read property 'getEntries' of null
    at Function.loadFromDatabaseIfNecessary (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:214:55)
    at Function.all (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:234:16)
    at Controller.getDevices (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:252:31)
    at Zigbee.getClients (/app/lib/zigbee.js:136:30)
    at OnEvent.stop (/app/lib/extension/onEvent.js:23:42)
    at Controller.callExtensionMethod (/app/lib/controller.js:372:44)
    at async Controller.stop (/app/lib/controller.js:193:9))

 Zigbee2MQTT:error 2021-01-13 10:35:11: Error: AREQ - ZDO - stateChangeInd after 60000ms
     at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
     at listOnTimeout (internal/timers.js:554:17)
     at processTimers (internal/timers.js:497:7)
 npm ERR! code ELIFECYCLE
 npm ERR! errno 1
 npm ERR! zigbee2mqtt@1.15.0-dev start: `node index.js`
 npm ERR! Exit status 1
```

I DID NOT reflased my CC1352P_2 but the container refused to start. 
While nothing was working I just updated the container, didn't helped either, 
Found this thread:
https://forum.iobroker.net/topic/29671/gel%C3%B6st-zigbee-kaputt-panid-already-in-use/2
Slapped the misbehaving router in a Cooking pod and a Biscuit tin to get decent shielding and let the container start.
Running around and re-plugging all the devices that were not reachable and back to a working set-up.
Just had to re-pair 2 out of 15 remotes.

patrislav1, 2021-10-10:
I get the same error after upgrading the zigbee adapter (from CC2531 to CC2652RB):

```
Zigbee2MQTT:info  2021-10-10 14:57:04: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
Zigbee2MQTT:info  2021-10-10 14:57:04: Starting zigbee-herdsman (0.13.71)
Zigbee2MQTT:error 2021-10-10 14:58:20: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-10-10 14:58:20: Failed to start zigbee
Zigbee2MQTT:error 2021-10-10 14:58:20: Exiting...
Zigbee2MQTT:error 2021-10-10 14:58:20: Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel
    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:172:23
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.18.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-10-10T12_58_20_672Z-debug.log
```

JanHACS, 2022-01-28:
> Ok, after reading [Koenkk/zigbee-herdsman#113](https://github.com/Koenkk/zigbee-herdsman/issues/113) I tried starting my CC1352P_2 without the SMA antenna (to reduce the signal strength) and now it started 🙃

That's works pretty well for me, thanks a lot!

ClusterM, 2022-02-26:
Just disassemble your dingle and press the reset button on the board.