Koenkk/zigbee2mqtt #2895: zigbee-herdsman fails to start on CC1352P-2: Timeout on SRSP - SYS - version

magnusnordlander, 2020-02-04:
# Bug Report
## What happened
I try to start zigbee2mqtt in Docker on a HPE Microserver with a CC132P-2 using the following configuration: https://gist.github.com/magnusnordlander/c9f8c979f44a308c5a3b5e3e2e3ce1e3#file-configuration-yaml

When running zigbee2mqtt:latest, I get the following log messages: https://gist.github.com/magnusnordlander/c9f8c979f44a308c5a3b5e3e2e3ce1e3#file-log-log

When running zigbee2mqtt:latest-dev, I get the following log messages:
https://gist.github.com/magnusnordlander/c9f8c979f44a308c5a3b5e3e2e3ce1e3#file-latest-dev-log

## What did you expect to happen
zigbee2mqtt should start

## Debug Info
zigbee2mqtt version: 1.9.0 or 1.9.0-dev (commit #558153e)
Firmware version: CC1352P_2_20191106.hex

I have tried running zigbee2mqtt on directly on the host, power cycling the device, reflashing the device (following the steps on https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html), verifying the flashed image, using a different USB port and changing the PAN id. None of this has yielded any results.

It seeems to me that the device just isn't responding whenever commands are being sent. Is there any simple way (e.g. using minicom) that I can use to coax a response out of the device on another machine? E.g., I tried cat:ing the device and in another shell while running `echo -e "\xFE\x00\x21\x02\x23" > /dev/ttyACM1`, just trying to get any response from the firmware. This didn't work, but I don't know if it was even supposed to.

CodeFinder2, 2020-02-04:
Once connected / powered up, try pressing the button next to the USB connector momentarily.

Then try (re-) starting Z2M.

magnusnordlander, 2020-02-04:
Wow, that seems to be working; at least it's starting up now. Is that a firmware reset or something? Should I expect to have to do that every time the device is reconnected?

CodeFinder2, 2020-02-04:
Unfortunately, yes. But it seems to depend on the underlying system/architecture/USB driver/...(?). See more infos here: https://github.com/Koenkk/zigbee2mqtt/issues/2162#issuecomment-568617134 (also in the other posts there).
TL;DR: I have to push that button on my Intel i7 (x86) as well but *not* on my Odroid XU4 (arm).

I guess, this can be closed now? 🙈

magnusnordlander, 2020-02-04:
Yeah, absolutely. This probably deserves a notice in the FAQ or something though.

closed 2020-02-04 by magnusnordlander

CodeFinder2, 2020-02-04:
Yes I think so, too! According to my knowledge, you just have to create a pull request against https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/information/FAQ.md (@Koenkk  please correct me if I am wrong, I couldn't find any specific information on that in your docs.)

Koenkk, 2020-02-05:
It should definitely be added to the docs, can you create a similar section of https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/information/FAQ.md#press-the-reset-button-on-the-device-only-applicable-for-cc2531 above it?

CodeFinder2, 2020-02-06:
> It should definitely be added to the docs, can you create a similar section of https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/information/FAQ.md#press-the-reset-button-on-the-device-only-applicable-for-cc2531 above it?

Done!

dvbit, 2020-03-04:
Hi I tried and verified all that is said in github and faq to no avail.
device is linked to correct ACM
port is writable 
user is in diallout group
permissions are ok
pushed the reset button

always get the 600 ms error.

Running on a dell x86 6th gen i5 portable

any clue or suggestion?

CodeFinder2, 2020-03-04:
@dvbit did you try to reflash?

dvbit, 2020-03-04:
Reflash with the same image? 
The module was fresh out of the box
I erased  the flash using uniflash on debian
Loaded zstack 3 im age and verified(ok)
Should reflashing with same image hlp?
I will try. Tx

dvbit, 2020-03-04:
No.i will . 
How can this help? Clears some areas?

dvbit, 2020-03-05:
Flashed : no success

dvbit, 2020-03-05:
trying on a raspberrry pi 2. I read somewhere certain revision of the board will not work. Where do I read that info?

dvbit, 2020-03-05:
Ok. No success on the raspberry too.
Qn: do other zigbee network interfere? I have Xiaomi and Hue on and (likely) also the amz echo although without attacched device on

dvbit, 2020-03-05:
Ok. On Raspberry now works. setting the serial in configuration to auto detect