Koenkk/zigbee2mqtt #2896: Ikea Tradfri LED1649C5 state reporting only after 5 minutes

dfsx1, 2020-02-05:
Great project @Koenkk , thanks!
I encountered a small/strange issue with a `LED1649C5` in my setup.

I have configured `report: true`  and `availability_timeout: 60`.
After a powercycle (within the availability timeout) the light goes on but the state stays OFF for exactly 5 minutes.

Other Ikea Tradfri bulbs report fine, only my `LED1649C5` type has this issue.
I compared debug-logs between a working `LED1837R5`.
Both devices succesfully (re-)setup for reporting `genOnOff` and `genLevelCtrl` after announcing, but:
- `LED1649C5` only reports `LevelCtrl`.
After that, it takes 5 minutes (maximumReportInterval?) before it reports it's first `OnOff` state?
- `LED1837R5` immediately reports both `OnOff` state and `LevelCtrl`.

Debug log:
```
LED1649C5 (doesn't work):
----------
zigbee2mqtt:debug 2020-02-05 14:19:41: Device '0xccccccfffe0823df' announced itself
zigbee2mqtt:debug 2020-02-05 14:19:41: Setup reporting for '0xccccccfffe0823df' - 1 - genOnOff
zigbee2mqtt:info  2020-02-05 14:19:42: Successfully setup reporting for '0xccccccfffe0823df' - 1 - genOnOff
zigbee2mqtt:debug 2020-02-05 14:19:42: Setup reporting for '0xccccccfffe0823df' - 1 - genLevelCtrl
zigbee2mqtt:info  2020-02-05 14:19:42: Successfully setup reporting for '0xccccccfffe0823df' - 1 - genLevelCtrl
zigbee2mqtt:debug 2020-02-05 14:19:45: Received Zigbee message from '0xccccccfffe0823df', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":135}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-02-05 14:19:45: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe0823df', payload '{"state":"OFF","brightness":135,"linkquality":76}'
zigbee2mqtt:debug 2020-02-05 14:20:45: Received Zigbee message from '0xccccccfffe0823df', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-05 14:20:45: Successfully pinged '0xccccccfffe0823df'
zigbee2mqtt:debug 2020-02-05 14:21:45: Received Zigbee message from '0xccccccfffe0823df', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-05 14:21:45: Successfully pinged '0xccccccfffe0823df'
zigbee2mqtt:debug 2020-02-05 14:22:46: Received Zigbee message from '0xccccccfffe0823df', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-05 14:22:46: Successfully pinged '0xccccccfffe0823df'
zigbee2mqtt:debug 2020-02-05 14:23:46: Received Zigbee message from '0xccccccfffe0823df', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-05 14:23:46: Successfully pinged '0xccccccfffe0823df'
zigbee2mqtt:debug 2020-02-05 14:24:42: Received Zigbee message from '0xccccccfffe0823df', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-02-05 14:24:42: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe0823df', payload '{"state":"ON","brightness":135,"linkquality":68}'
zigbee2mqtt:debug 2020-02-05 14:24:45: Received Zigbee message from '0xccccccfffe0823df', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":135}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-02-05 14:24:45: MQTT publish: topic 'zigbee2mqtt/0xccccccfffe0823df', payload '{"state":"ON","brightness":135,"linkquality":68}'

LED1837R5 (works):
----------
zigbee2mqtt:debug 2020-02-05 14:19:12: Device '0x14b457fffe4ef754' announced itself
zigbee2mqtt:debug 2020-02-05 14:19:12: Setup reporting for '0x14b457fffe4ef754' - 1 - genOnOff
zigbee2mqtt:debug 2020-02-05 14:19:13: Received Zigbee message from '0x14b457fffe4ef754', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-02-05 14:19:13: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe4ef754', payload '{"state":"ON","brightness":137,"linkquality":81}'
zigbee2mqtt:info  2020-02-05 14:19:14: Successfully setup reporting for '0x14b457fffe4ef754' - 1 - genOnOff
zigbee2mqtt:debug 2020-02-05 14:19:14: Setup reporting for '0x14b457fffe4ef754' - 1 - genLevelCtrl
zigbee2mqtt:info  2020-02-05 14:19:15: Successfully setup reporting for '0x14b457fffe4ef754' - 1 - genLevelCtrl
zigbee2mqtt:debug 2020-02-05 14:19:15: Received Zigbee message from '0x14b457fffe4ef754', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":137}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-02-05 14:19:15: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe4ef754', payload '{"state":"ON","brightness":137,"linkquality":84}'
zigbee2mqtt:debug 2020-02-05 14:20:15: Received Zigbee message from '0x14b457fffe4ef754', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-05 14:20:15: Successfully pinged '0x14b457fffe4ef754'
zigbee2mqtt:debug 2020-02-05 14:21:16: Received Zigbee message from '0x14b457fffe4ef754', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-05 14:21:16: Successfully pinged '0x14b457fffe4ef754'
zigbee2mqtt:debug 2020-02-05 14:22:16: Received Zigbee message from '0x14b457fffe4ef754', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-05 14:22:16: Successfully pinged '0x14b457fffe4ef754'
zigbee2mqtt:debug 2020-02-05 14:23:16: Received Zigbee message from '0x14b457fffe4ef754', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-02-05 14:23:16: Successfully pinged '0x14b457fffe4ef754'
zigbee2mqtt:debug 2020-02-05 14:24:12: Received Zigbee message from '0x14b457fffe4ef754', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-02-05 14:24:12: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe4ef754', payload '{"state":"ON","brightness":137,"linkquality":18}'
zigbee2mqtt:debug 2020-02-05 14:24:15: Received Zigbee message from '0x14b457fffe4ef754', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":137}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-02-05 14:24:15: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe4ef754', payload '{"state":"ON","brightness":137,"linkquality":18}'
```

I found a (dirty) quickfix by commenting out: `!utils.isIkeaTradfriDevice(device)` in \extension\deviceAvailability.js line 194.
Then the correct ON state is immeditely read from the device. #1383

```
const online = this.state.hasOwnProperty(device.ieeeAddr) && this.state[device.ieeeAddr];
if (online && type === 'deviceAnnounce' && !utils.isIkeaTradfriDevice(device)) {
    /**
     * In case the device is powered off AND on within the availability timeout,
     * zigbee2qmtt does not detect the device as offline (device is still marked online).
     * When a device is turned on again the state could be out of sync.
     * https://github.com/Koenkk/zigbee2mqtt/issues/1383#issuecomment-489412168
     * deviceAnnounce is typically send when a device comes online.
     *
     * This isn't needed for TRADFRI devices as they already send the state themself.
     */
    this.onReconnect(device);
}
```
But any idea why the `LED1649C5` bulb is not reporting the `OnOff` state immediately? #966
I noticed zclVersion1 vs zclVersion3, not sure if it has anything to do with firmware?
The device does report at least something (`LevelCtrl`), so I don't think it's unsupported?

sjorge, 2020-02-05:
Maybe the configuration didn't go through for genOnOff... you can try reconfiguring the bulb...
https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

dfsx1, 2020-02-05:
@sjorge thanks, I just tried reconfiguring.
Unfortunatelly that didn't help, the genOnOff report is still only received after 300 seconds :(

connesc, 2020-04-12:
I discovered exactly the same problem with my IKEA LED1624G9 today. Reporting was working perfectly before I updated both Zigbee2mqtt and my bulb firmware. I don't know which one caused the regression.

I also randomly encounter issue #3224 when bringing power back. Both issues may be related somehow.

closed 2020-06-20 by stale[bot]