Koenkk/zigbee2mqtt #2897: Docker install fails with Error: SRSP - SYS - version after 6000ms

GMFalka, 2020-02-05:
# Bug Report
## What happened
Installing zigbee2mqtt in docker, but startup fails.

## log

`lrwxrwxrwx 1 root root 13 febr   5 20:46 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001936940B-if00 -> ../../ttyACM0`

```
sudo docker run \
>    -it \
>    -v $(pwd)/data:/app/data \
>    --device=/dev/ttyACM0 \
>    -e TZ=Europe/Amsterdam \
>    -v /run/udev:/run/udev:ro \
>    --privileged=true \
>    koenkk/zigbee2mqtt
Using '/app/data' as data directory

> zigbee2mqtt@1.9.0 start /app
> node index.js

zigbee2mqtt:info  2020-02-05 20:45:05: Logging to console and directory: '/app/data/log/2020-02-05.20-45-05'
zigbee2mqtt:info  2020-02-05 20:45:05: Starting zigbee2mqtt version 1.9.0 (commit #287f843)
zigbee2mqtt:info  2020-02-05 20:45:05: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-02-05 20:45:12: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-02-05 20:45:12: Failed to start zigbee
zigbee2mqtt:error 2020-02-05 20:45:12: Exiting...
zigbee2mqtt:error 2020-02-05 20:45:12: Error: SRSP - SYS - version after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.9.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.9.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-05T19_45_12_287Z-debug.log
```

dfsx1, 2020-02-05:
You can add:
`-e DEBUG=zigbee-herdsman*` 
or 
`-e DEBUG=*` 
to your docker run command to get more logs.

GMFalka, 2020-02-05:
> You can add:
> `-e DEBUG=zigbee-herdsman*`
> or
> `-e DEBUG=*`
> to your docker run command to get more logs.

thank you for helping out!

heres the extensive log:

```
> zigbee2mqtt@1.9.0 start /app
> node index.js

winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
zigbee2mqtt:info  2020-02-05 21:07:16: Logging to console and directory: '/app/data/log/2020-02-05.21-07-16'
winston:file written true false
zigbee2mqtt:debug 2020-02-05 21:07:16: Removing old log directory '/app/data/log/2020-02-05.19-33-24'
winston:file written true false
zigbee2mqtt:debug 2020-02-05 21:07:16: Can't load state from file /app/data/state.json (doesn't exist)
winston:file written true false
winston:file logged 97 info  2020-02-05 21:07:16: Logging to console and directory: '/app/data/log/2020-02-05.21-07-16'

winston:file logged 187 debug 2020-02-05 21:07:16: Removing old log directory '/app/data/log/2020-02-05.19-33-24'

winston:file logged 278 debug 2020-02-05 21:07:16: Can't load state from file /app/data/state.json (doesn't exist)

winston:file ENOENT ok /app/data/log/2020-02-05.21-07-16/log.txt
winston:file stat done: log.txt { size: 0 }
winston:file create stream start /app/data/log/2020-02-05.21-07-16/log.txt { flags: 'a' }
winston:file create stream ok /app/data/log/2020-02-05.21-07-16/log.txt
zigbee2mqtt:info  2020-02-05 21:07:16: Starting zigbee2mqtt version 1.9.0 (commit #287f843)
winston:file written true false
zigbee2mqtt:info  2020-02-05 21:07:16: Starting zigbee-herdsman...
winston:file written true false
zigbee2mqtt:debug 2020-02-05 21:07:16: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
winston:file written true false
  serialport/bindings loading LinuxBinding +0ms
  serialport/stream .list +0ms
winston:file logged 80 info  2020-02-05 21:07:16: Starting zigbee2mqtt version 1.9.0 (commit #287f843)

winston:file logged 135 info  2020-02-05 21:07:16: Starting zigbee-herdsman...

winston:file logged 528 debug 2020-02-05 21:07:16: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'

winston:file file open ok /app/data/log/2020-02-05.21-07-16/log.txt
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  serialport/stream opening path: /dev/ttyACM0 +60ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +1ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyACM0 +0ms
  serialport/stream _read reading { start: 0, toRead: 65536 } +1ms
  serialport/binding-abstract read +1ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +3ms
[... 25 lines cut ...]
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - version - {} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,2,35] +1s
zigbee2mqtt:error 2020-02-05 21:07:23: Error while starting zigbee-herdsman
winston:file written true false
zigbee2mqtt:error 2020-02-05 21:07:23: Failed to start zigbee
winston:file written true false
zigbee2mqtt:error 2020-02-05 21:07:23: Exiting...
winston:file written true false
zigbee2mqtt:error 2020-02-05 21:07:23: Error: SRSP - SYS - version after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
winston:file written true false
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.9.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.9.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-05T20_07_23_239Z-debug.log

```

GMFalka, 2020-02-06:
Well. I tried installing zigbee2mqtt the regular way. (to the HOST ubuntu, not within docker)
The problem persist.

The stick can be seen and detected on ttyACM0.
The user have write acces to ttyACM0.

I had this stick working for years before on ubuntu 18.10. Now on 19.04 it  doesnt seem to work.

```
npm start

> zigbee2mqtt@1.10.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-02-06 20:39:52: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-02-06.20-39-52' filename: log.txt
zigbee2mqtt:debug 2020-02-06 20:39:52: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
zigbee2mqtt:info  2020-02-06 20:39:52: Starting zigbee2mqtt version 1.10.0 (commit #8df6be8)
zigbee2mqtt:info  2020-02-06 20:39:52: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-02-06 20:39:52: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
zigbee2mqtt:error 2020-02-06 20:40:42: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-02-06 20:40:42: Failed to start zigbee
zigbee2mqtt:error 2020-02-06 20:40:42: Exiting...
zigbee2mqtt:error 2020-02-06 20:40:42: Error: AREQ - SYS - resetInd after 30000ms
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.10.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.10.0 start script.
```

GMFalka, 2020-02-06:
Closing, as its probably a duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/2899

dfsx1, 2020-02-06:
I experienced same error today.
Pulling the USB out and quickly back in, 1 second before starting the docker container seems to work for me.

closed 2020-02-06 by Koenkk