Koenkk/zigbee2mqtt #2899: Error: AREQ - SYS - resetInd after 30000ms

huahe, 2020-02-06:
Hello !

I recently moved everything to a Intel NUC Model # DC3217IYE

I installed Ubuntu 18.04 LTS and set up everything there.

But zigbee2mqtt does not work. 

This is what I am getting:

https://pastebin.com/66a214ME

Everything is ok, the port, the permissions, etc, but it does not work. Same happens in hassio addon.

I connected it directly to one of the NUC USB ports, using a powered hub and through an extension cable. Same result.

It worked once with this setup, I was able to start an interview with an Ikea switch, but never again.

Any ideas?

Thanks.

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

crystallero, 2020-02-06:
I am seeing this in the host too:

```[41651.965433] hassio: port 6(veth459ed60) entered disabled state
[41651.967554] eth0: renamed from vethf989487
[41651.984067] hassio: port 6(veth459ed60) entered blocking state
[41651.984076] hassio: port 6(veth459ed60) entered forwarding state
[42314.441243] cdc_acm 2-1.7:1.0: failed to set dtr/rts
[42314.682897] hassio: port 6(veth459ed60) entered disabled state
[42314.683099] vethf989487: renamed from eth0
[42314.770924] hassio: port 6(veth459ed60) entered disabled state
[42314.775674] device veth459ed60 left promiscuous mode
[42314.775683] hassio: port 6(veth459ed60) entered disabled state
[42315.136450] hassio: port 6(veth44049d9) entered blocking state
[42315.136455] hassio: port 6(veth44049d9) entered disabled state
[42315.136650] device veth44049d9 entered promiscuous mode
[42315.136961] hassio: port 6(veth44049d9) entered blocking state
[42315.136965] hassio: port 6(veth44049d9) entered forwarding state
[42315.151033] IPv6: ADDRCONF(NETDEV_CHANGE): veth44049d9: link becomes ready
[42315.655919] hassio: port 6(veth44049d9) entered disabled state
[42315.660033] eth0: renamed from veth73a873b
[42315.679129] hassio: port 6(veth44049d9) entered blocking state
[42315.679139] hassio: port 6(veth44049d9) entered forwarding state
[42324.681301] cdc_acm 2-1.7:1.0: failed to set dtr/rts
[42410.185871] cdc_acm 2-1.7:1.0: failed to set dtr/rts
[42417.865803] cdc_acm 2-1.7:1.0: failed to set dtr/rts
[42502.346243] cdc_acm 2-1.7:1.0: failed to set dtr/rts
[42511.050343] cdc_acm 2-1.7:1.0: failed to set dtr/rts```

GMFalka, 2020-02-06:
@crystallero 

I have the same problem! So its probably not just on your end.
Mine: https://github.com/Koenkk/zigbee2mqtt/issues/2897

crystallero, 2020-02-06:
The weird thing is that I keep seeing the USB disappearing in the host machine. Sometimes the USB green led even goes off. Is it a hardware issue? Or maybe the NUC USB host? I think I read that @Koenkk uses zigbee2mqtt in a NUC too, so I guess that it should work flawlessly

This is so frustrating :(

GMFalka, 2020-02-06:
> The weird thing is that I keep seeing the USB disappearing in the host machine. Sometimes the USB green led even goes off. Is it a hardware issue? Or maybe the NUC USB host? I think I read that [...]

Can't be a faulty hw. I was using this stick from the same PC, same usb port. Only differenfe is the fresh install of Ubuntu 19.10.
Even tried using the old zigbee install folder. Didn't change the error msg. 

Also modem manager is purged from my system.

oklih, 2020-02-06:
I also have this problem, I cant find what to do to get it to work. Pulling out the USB, sometimes get it back online. Just moved to docker with SSD from an SDcard.

dfsx1, 2020-02-06:
As I reported in https://github.com/Koenkk/zigbee2mqtt/issues/2897#issuecomment-583101035
I got it at least starting up again after pulling the USB out and quickly back in, 1 second before starting the docker container.
But all pairing fails and it it very unstable (the docker container crashes periodically).
I even tried pulling the previous 1.9.0 zigbee2mqtt docker release, but the issue persists?!

oklih, 2020-02-06:
So i had a RFLink also attached to a usb port. I pulled it out and changed the tty port for the Zigbee2mqtt device. Now I have a different error along with the old one! Sorry if I wrongly put in the log. 

<2020-02-07T00:24:47: PM2 log: Launching in no daemon mode
2020-02-07T00:24:47: PM2 log: App [npm:0] starting in -fork mode-
2020-02-07T00:24:47: PM2 log: App [npm:0] online
> zigbee2mqtt@1.10.0 start /zigbee2mqtt-1.10.0
> node index.js
zigbee2mqtt:info  2020-02-07 00:24:50: Logging to console and directory: '/share/zigbee2mqtt/log/2020-02-07.00-24-50' filename: log.txt
zigbee2mqtt:info  2020-02-07 00:24:51: Starting zigbee2mqtt version 1.10.0 (commit #unknown)
zigbee2mqtt:info  2020-02-07 00:24:51: Starting zigbee-herdsman...
(node:58) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - activeEpReq - {"dstaddr":0,"nwkaddrofinterest":0}' failed with status '194' (expected '0')
    at Znp.<anonymous> (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
(node:58) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:58) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
zigbee2mqtt:error 2020-02-07 00:25:02: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-02-07 00:25:02: Failed to start zigbee
zigbee2mqtt:error 2020-02-07 00:25:02: Exiting...
zigbee2mqtt:error 2020-02-07 00:25:02: Error: AREQ - ZDO - activeEpRsp after 10000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)>

Koenkk, 2020-02-07:
Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

GMFalka, 2020-02-07:
> Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

All checked. 
Stick is recognized on the correct port 
User has write access to the same port. 
Tried the reset button.

Koenkk, 2020-02-07:
@GMFalka can you post the log when running with zigbee herdsman debug logging enabled?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

GMFalka, 2020-02-07:
> @GMFalka can you post the log when running with zigbee herdsman debug logging enabled?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Yes sir. Tonight as soon as I get home. 
I have a debug log at hand with the docker version, but probably you need a non-dockerized. :)

dfsx1, 2020-02-07:
With debug I see multiple times:

> serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
>   errno: -11,
>   code: 'EAGAIN',
>   syscall: 'read'
> }

and also:

> zigbee-herdsman:adapter:zStack:znp:error Error while parsing to ZpiObject 'RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 5. Received 6

```
> zigbee2mqtt@1.10.0 start /app
> node index.js

winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
zigbee2mqtt:info  2020-02-07 14:00:47: Logging to console and directory: '/app/data/log/2020-02-07.14-00-46' filename: log.txt
winston:file written true false
zigbee2mqtt:debug 2020-02-07 14:00:47: Can't load state from file /app/data/state.json (doesn't exist)
winston:file written true false
winston:file logged 115 info  2020-02-07 14:00:47: Logging to console and directory: '/app/data/log/2020-02-07.14-00-46' filename: log.txt

winston:file logged 206 debug 2020-02-07 14:00:47: Can't load state from file /app/data/state.json (doesn't exist)

winston:file ENOENT ok /app/data/log/2020-02-07.14-00-46/log.txt
winston:file stat done: log.txt { size: 0 }
winston:file create stream start /app/data/log/2020-02-07.14-00-46/log.txt { flags: 'a' }
winston:file create stream ok /app/data/log/2020-02-07.14-00-46/log.txt
zigbee2mqtt:info  2020-02-07 14:00:47: Starting zigbee2mqtt version 1.10.0 (commit #)
winston:file written true false
zigbee2mqtt:info  2020-02-07 14:00:47: Starting zigbee-herdsman...
winston:file written true false
zigbee2mqtt:debug 2020-02-07 14:00:47: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
winston:file written true false
2020-02-07T13:00:47.613Z serialport/bindings loading LinuxBinding
2020-02-07T13:00:47.743Z serialport/stream .list
winston:file logged 74 info  2020-02-07 14:00:47: Starting zigbee2mqtt version 1.10.0 (commit #)

winston:file logged 129 info  2020-02-07 14:00:47: Starting zigbee-herdsman...

winston:file logged 522 debug 2020-02-07 14:00:47: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'

winston:file file open ok /app/data/log/2020-02-07.14-00-46/log.txt
2020-02-07T13:00:47.879Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}'
2020-02-07T13:00:47.882Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false}
2020-02-07T13:00:47.888Z serialport/stream opening path: /dev/ttyACM0
2020-02-07T13:00:47.889Z serialport/binding-abstract open
2020-02-07T13:00:47.893Z serialport/stream _read queueing _read for after open
2020-02-07T13:00:47.896Z serialport/bindings/poller Creating poller
2020-02-07T13:00:47.897Z serialport/stream opened path: /dev/ttyACM0
2020-02-07T13:00:47.898Z serialport/stream _read reading { start: 0, toRead: 65536 }
2020-02-07T13:00:47.900Z serialport/binding-abstract read
2020-02-07T13:00:47.902Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2020-02-07T13:00:47.903Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2020-02-07T13:00:47.904Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2020-02-07T13:00:47.906Z serialport/stream _write 1 bytes of data
[... 309 lines cut ...]
}
2020-02-07T13:00:49.210Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
2020-02-07T13:00:49.215Z serialport/bindings/poller Polling for "readable"
zigbee2mqtt:error 2020-02-07 14:00:59: Error while starting zigbee-herdsman
winston:file written true false
zigbee2mqtt:error 2020-02-07 14:00:59: Failed to start zigbee
winston:file written true false
zigbee2mqtt:error 2020-02-07 14:00:59: Exiting...
winston:file written true false
[... cut ...]
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
winston:file written true false
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.10.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.10.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-07T13_00_59_274Z-debug.log
```

GMFalka, 2020-02-07:
I have the same error on docker with debug btw 

```
herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-02-07T13:00:49.206Z serialport/stream _read reading { start: 142, toRead: 65394 }
2020-02-07T13:00:49.208Z serialport/binding-abstract read
2020-02-07T13:00:49.208Z serialport/bindings/unixRead Starting read
2020-02-07T13:00:49.209Z serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  
```

oklih, 2020-02-07:
Hi All,
I have temp installed Hass.io on a SD card again and restored my backup. After changing and updating the system again, it still did not work. So I read somewhere that a reflash of the cc2531 did the trick. looks like it is working again. Now I have reconnected everything and is working fine. I don't think i will be going back to docker for now. untill it is better supported and I have better understanding of docker! (lol)

dfsx1, 2020-02-07:
@oklih
Thank you, reflashing the CC2531 also seems to work for me :)

GMFalka, 2020-02-07:
I will try to reflash the stick. 

I saw other new tickets with the same error we have. Maybe an ubuntu update? Strange that a lot of ppl have the same problem out of the blue

papperone, 2020-02-07:
hi, I am trying to have zigbee2mqtt running on a windows 10 pc but it fails with the same error I do see here; I'm using a CC2530 with latest coordinator firmware loaded, what can I try to do to understand the reason of this failure? I can only add I donwload latest jnode (12.15-0 LTS) as the 10 LTS (as suggested in zigbee2mqtt instructions) is not available for download

papperone, 2020-02-07:
> 
> 
> I will try to reflash the stick.
> 
> I saw other new tickets with the same error we have. Maybe an ubuntu update? Strange that a lot of ppl have the same problem out of the blue

I have exactly the same but on Windows10 SO so cannot be linked to Ubuntu....

crystallero, 2020-02-07:
I reflashed the stick and it still doesn't work. 

I tried in an old laptop (same Ubuntu version) and seemed to work in the beginning. But it stopped working after a while :(

Koenkk, 2020-02-08:
To all: in case you have this issue please provide the herdsman debug logging, otherwise I cannot help (https://github.com/Koenkk/zigbee2mqtt/issues/2899#issuecomment-583283159)

papperone, 2020-02-08:
here is my log (the only one I could have found)
https://pastebin.com/6ww6hzYi
if you need other one thanks to let me know where I shoudl look for...

Koenkk, 2020-02-08:
This log will only be logged to the screen when running with the npm start mentioned in https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

murjam, 2020-02-08:
Here's my log: https://pastebin.com/5RRk2H65 :bowing_man:

papperone, 2020-02-08:
> 
> 
> This log will only be logged to the screen when running with the npm start mentioned in https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

how can I run the below under windows CMD window, I do not understand?

To enable debug logging for Zigbee-herdman start zigbee2mqtt with: **DEBUG=zigbee-herdsman* npm start**. Zigbee-herdsman is the Zigbee library where Zigbee2mqtt is based up-on

MarcSN311, 2020-02-08:
https://pastebin.com/fmd9yVsK

Koenkk, 2020-02-08:
@papperone 

```
setx DEBUG "zigbee-herdsman:*"
npm start
--

Koenkk, 2020-02-08:
So for the people having this issue, please fill in the following questionnaire:
- What kind of adapter is it? (CC2531 or CC2530)
- What firmware is flashed on it?
- Did you ever manage to start Zigbee2mqtt with this adapter?
- What is your OS and hardware?
- Did you check every step of the [FAQ](https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start)? Especially verify that you didn't got a CC2540 instead of a CC2531

papperone, 2020-02-08:
here it is:

- CC2530+CC2592
- firmware: CC2530_CC2592_DEFAULT_20190608
- I tried on a friend's PC and it works
- Windows 10 on a i5-6600T / 16Gb DDR4, first time I try on this PC
- all checked, I can confirm my CHIP is CC2530

https://pastebin.com/Zunz9tNC

Koenkk, 2020-02-08:
@papperone for the CC2530 you need to set `rtscts: false` (see docs: https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc2530-optionally-with-a-cc2591-or-cc2592-rf-frontend)

MarcSN311, 2020-02-09:
- What kind of adapter is it? (CC2531 or CC2530)
**2 different CC2531s. After the first one stopped working I flashed a new one.**

- What firmware is flashed on it?
**CC2531_DEFAULT_20190608 and an older firmware on the other stick (don't know which one)**

- Did you ever manage to start Zigbee2mqtt with this adapter?
**yes.**

- What is your OS and hardware? If you every manage to start zigbee2mqtt adapter, did you change anything recently to this?
**debian VM, running on VMware ESXi; Did not change anything recently.**

- Did you check every step of the FAQ? Especially verify that you didn't got a CC2540 instead of a CC2531
**yes.**

GMFalka, 2020-02-09:
> Questionnaire:
> 
> * What kind of adapter is it? (CC2531 or CC2530)
**`CC2531`**
> * What firmware is flashed on it?
**`CC2531_DEFAULT_20190608`**
> * Did you ever manage to start Zigbee2mqtt with this adapter?
**`Yes. Worked VERY reliable for months (maybe a year) without a single problem.`**
> * What is your OS and hardware? If you every manage to start zigbee2mqtt adapter, did you change anything recently to this?
**`A tower PC with intel cpu and ubuntu 18.10. I just did an ssd wipe and installed ubuntu 19.04.`**
> * Did you check every step of the [FAQ](https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start)? Especially verify that you didn't got a CC2540 instead of a CC2531
**`Yes!`**

[Heres my log (DEBUG=zigbee-herdsman* npm start)](https://pastebin.com/vYtaATAc)

dvdbosch, 2020-02-10:
I had the same problem! If fixed it (wonderly) by changes the usb devices to a different port (apperently it was misconfigured in the configuration file).

murjam, 2020-02-10:
I had the same problem and realized I tried to use CC2540 with CC2531 firmware (was thinking I have CC2531) 🙈

Koenkk, 2020-02-10:
@MarcSN311 @GMFalka can you try if reflashing the stick solves the issue?

GMFalka, 2020-02-11:
> @MarcSN311 @GMFalka can you try if reflashing the stick solves the issue?

Hi @Koenkk!

Reflashing the stick worked. Thanks. 
Its strange, it got corrupted because of an ubuntu reinstall (maybe docker is the culprit)

kizhaev, 2020-02-11:
Having the same issue, here is the log: https://pastebin.com/uWxkph1E

What kind of adapter is it? (CC2531 or CC2530)
CC2531

What firmware is flashed on it?
CC2531_DEFAULT_20190608

Did you ever manage to start Zigbee2mqtt with this adapter?
No

What is your OS and hardware? If you every manage to start zigbee2mqtt adapter, did you change anything recently to this?
Hassio + RPi4b

Did you check every step of the FAQ? Especially verify that you didn't got a CC2540 instead of a CC2531
Yes, it is CC2531 for sure.

oklih, 2020-02-13:
So a while ago I was on docker, and had trouble with my CC2531 adapter. I moved to hass.io on SD for testing. Al seemed to work fine, but now in a few days I get the same problems I had while I was on docker. Below I put the questionere:

What kind of adapter is it? (CC2531 or CC2530)
CC2531

What firmware is flashed on it?
CC2531_DEFAULT_20190608, flashed it twice, because I thought that would do the trick to get it working again.

Did you ever manage to start Zigbee2mqtt with this adapter?
Yes, was working before al of the updates recently in Home Assistant. 

What is your OS and hardware? If you every manage to start zigbee2mqtt adapter, did you change anything recently to this?
Hass.io (latest version 105.3) + RPI4B

Did you check every step of the FAQ? Especially verify that you didn't got a CC2540 instead of a CC2531?
Yes, I did. also checked online where I bought it. questioning everything by now. 

Strange thing is that it sometimes does work with the Xiaomi temp sensors. they give data back, but when I set a wall switch to on I get an error in de log. When I get it to work by pulling out the adapter and pluging it in again. at the moment it is not working again and giving the "Error while starting zigbee-herdsman" error again. I hope there will be a sollution soon. :)

Koenkk, 2020-02-13:
@oklih unfortunately this is not something that can be fixed from the zigbee2mqtt side, the issue is that the zigbee2mqtt cannot connect to the adapter, this could e.g. be due to the sniffer firmware being corrupted (try reflashing), wrong port, system issue, etc.

I've improved the error logging in the latest dev branch.

oklih, 2020-02-13:
@Koenkk I have found an old backup of my hassio, I downgraded to this snapshot and al started working again. I don't know why, maybe there is an error in my config after the update to 105.x. currently on 104.2. I have updated the add-on of zigbee2mqtt. No errors there yet. Tomorrow I will try the hass.io update to 105.3. Pray that everything will stay the same.
Or maybe moving to docker and back again, using the config of the snapshots did mess up my config. Thanx for the advice, I will let everyone know how the update will be.

timdonovanuk, 2020-02-20:
Just had this. Was moving my z2m install out of HassOS into a docker running on a host. Brand new z2m install, resigned to the fact I'm going to have to repair everything. Still getting this error:

So for the people having this issue, please fill in the following questionnaire:

What kind of adapter is it? CC1352P-2
What firmware is flashed on it? Latest
Did you ever manage to start Zigbee2mqtt with this adapter? Yes, but not in this environment
What is your OS and hardware? If you every manage to start zigbee2mqtt adapter, did you change anything recently to this?

oppergod, 2020-02-20:
Had the same issue and reflashing the CC2531 fixed the issue for me.

timdonovanuk, 2020-02-20:
Have now tried a reflash, no luck. If I could figure out how to turn on zigbee-herdsman debug logging in a docker container I'd provide logs.

cg-n, 2020-02-20:
I tried reflashing multiple times on a CC2531 I purchased in the last couple weeks, but had no success on getting it to connect and appear as `/tty/ACM0`.  The flash would by all accounts appear successful (success messages, proper green lights on debugger and CC2531), and then when I would plug in the CC2531 by itself while running `dmesg -wH` I wouldn't see it recognized. The green light on the CC2531 would stay on for a few seconds and then turn off until I plugged it in again.

I purchased a second CC2531, followed the same instructions for flashing it using the same equipment and environment, and it works perfectly. It shows up in `dmesg -wH` on connection and I can see it as `/tty/ACM0`. I feel like a bad batch of CC2531s went out, or at least a batch that isn't compatible with the latest CC2531_DEFAULT_20190608 firmware.

timdonovanuk, 2020-02-20:
Still no luck here, although now I can't even get a trusty CC2531 working in a VM. Not sure where it's all gone wrong, as I had it working fine the other day! Will try on bare metal Pi tomorrow.

timdonovanuk, 2020-02-20:
Argh! Somehow my docker install had shat the bed. It was persisting z2m invocations even through reboots, and restarting the containers again, even though I didn't background them when I ran them. I don't know how I managed to get it doing that..not experienced it before. I started to wonder why my VM running z2m needed 4GB of ram and was using 100% CPU..and then found this:
[image]

I had z2m running over 50 times somehow! No wonder none of my zigbee USB sticks would work! I killed them all off and then z2m started first time. Dooooh.

Tamadite, 2020-02-28:
Just for the record: same error code same solution as given above that is **reflashing**. I reflashed it with the same version I had (ver DEFAULT_20190608).
Mine died without no apparent reason in the middle of the day.

closed 2020-05-06 by stale[bot]

joaofl, 2020-05-19:
I had the exact same error, which manifested after I replaced my CC2531 with another with external antenna. Tried to re-flash, reinstall the container, followed the FAQ. Nothing worked.
What solved my problem was to simply remove the zigbee2mqtt folder, and restart the container.

`rm -rf /shared/zigbee2mqtt`

joaofl, 2020-05-20:
@Koenkk There might be something with the state machine at start. Maybe worth checking that.

Koenkk, 2020-05-20:
@joaofl not sure where this could come from. Can you provide the herdsman debug logging when the error occurs?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

joaofl, 2020-05-31:
Thing is that I can no longer reproduce the issue, but my logs/errors were not any different from the ones mentioned in this thread.
Sorry that I'm not able to help much further. If it happens again, I'll make sure to dump the logs here.

delphir-com, 2020-06-08:
here is my log, the dongle was working fine for a long time and then Xiaomi button stopped working without any reason. After investigation I found that for some reason it "quit" the network, so I went into config to allow joining of new devices. Restart, managed to "join" the network and... zigbee stopped working after this with `Error: SRSP - SYS - version after 6000ms`.

 I had a "working" SD card backup, rolled back, but still same issue.

"Pulling out"/whatever is not helping. Didn't try to reflash yet.

CC2531, I can't find the firmware version I was flashing 6 months ago, but as the latest Z-Stack_Home update was 12months ago - I assume it was the "latest" one.

```
$ DEBUG=* npm start

> zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt
> node index.js

winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
zigbee2mqtt:info  2020-06-07T23:53:50: Logging to directory: '/opt/zigbee2mqtt/data/log/2020-06-08.00-53-49'
winston:file written true false
zigbee2mqtt:debug 2020-06-07T23:53:50: Removing old log directory '/opt/zigbee2mqtt/data/log/2019-11-29.19-53-05'
winston:file written true false
zigbee2mqtt:debug 2020-06-07T23:53:50: Loaded state from file /opt/zigbee2mqtt/data/state.json
winston:file written true false
winston:file logged 97 info  2020-06-07T23:53:50: Logging to directory: '/opt/zigbee2mqtt/data/log/2020-06-08.00-53-49'

winston:file logged 199 debug 2020-06-07T23:53:50: Removing old log directory '/opt/zigbee2mqtt/data/log/2019-11-29.19-53-05'

winston:file logged 282 debug 2020-06-07T23:53:50: Loaded state from file /opt/zigbee2mqtt/data/state.json

winston:file ENOENT ok /opt/zigbee2mqtt/data/log/2020-06-08.00-53-49/log.txt
winston:file stat done: log.txt { size: 0 }
winston:file create stream start /opt/zigbee2mqtt/data/log/2020-06-08.00-53-49/log.txt { flags: 'a' }
winston:file create stream ok /opt/zigbee2mqtt/data/log/2020-06-08.00-53-49/log.txt
winston:file file open ok /opt/zigbee2mqtt/data/log/2020-06-08.00-53-49/log.txt
zigbee2mqtt:info  2020-06-07T23:53:50: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
winston:file written true false
zigbee2mqtt:info  2020-06-07T23:53:50: Starting zigbee-herdsman...
winston:file written true false
zigbee2mqtt:debug 2020-06-07T23:53:50: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
winston:file written true false
  serialport/bindings loading LinuxBinding +0ms
  serialport/stream .list +0ms
winston:file logged 80 info  2020-06-07T23:53:50: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)

winston:file logged 135 info  2020-06-07T23:53:50: Starting zigbee-herdsman...

winston:file logged 564 debug 2020-06-07T23:53:50: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'

  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
  serialport/stream opening path: /dev/ttyACM0 +361ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +9ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyACM0 +5s
  serialport/stream _read reading { start: 0, toRead: 65536 } +2ms
  serialport/binding-abstract read +5s
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +5s
[... 18 lines cut ...]
  serialport/stream binding.write write finished +6ms
zigbee2mqtt:error 2020-06-07T23:54:03: Error while starting zigbee-herdsman
winston:file written true false
[... cut ...]
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
winston:file written true false
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-06-07T23_54_08_551Z-debug.log
```

GMFalka, 2020-06-08:
Hi @Koenkk 
Sorry for poking in here again. 
Today there was a split second blackout in my home. The server (with the CC2351) is on UPS, protected, but all the lights got turned on and the stick got stuck again in `Error: AREQ - SYS - resetInd after 30000ms
`
I dont know if this is any help to You, but i wanted to share my experience.
Only a full stick reflash got the CC2351 out of that error message. 
Isn't that intresting? The stick got "corrupted" because the lights lost power for a split second. 
There was no command issued during this time, but device riporting is turned on.

Thank you for this great firmware and support.

delphir-com, 2020-06-08:
re-flashing helped. I made a flash-dump before flashing. maybe it will help to understand the core of the issue

[read.zip](https://github.com/Koenkk/zigbee2mqtt/files/4746460/read.zip)

empatel96, 2020-07-03:
My stick has just stopped working after upgrading 112.1 and is giving me the following errors: 
[https://pastebin.com/eJ3eL5dL](url)

I read the FAQ and got this when checking for the correct port

```
~ $ ls -l /dev/serial/by-id
total 0
crw-rw----    1 root     audio     166,   0 Jul  3 13:08 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0019368D4C-if00
```

This is what I get when checking for write permissions
```
crw-rw----    1 root     audio     166,   0 Jul  3 13:08 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0019368D4C-if00
~ $ test -w /dev/ttyACM0 && echo success || echo failure
success
```

scoat76, 2020-10-03:
hi, 
this issue happend to me ... reflashing didnt helped to me.

What kind of adapter is it? (CC2531 or CC2530)
CC2531

What firmware is flashed on it?
CC2531_DEFAULT_20190608

Did you ever manage to start Zigbee2mqtt with this adapter?
No

What is your OS and hardware? If you every manage to start zigbee2mqtt adapter, did you change anything recently to this?
Hassio + RPi4b

Did you check every step of the FAQ? Especially verify that you didn't got a CC2540 instead of a CC2531
Yes, it is CC2531 for sure.

below the log with debug mode:
2020-10-03T23:43:20: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-10-03T23:43:20: PM2 log: App [npm:0] starting in -fork mode-
2020-10-03T23:43:20: PM2 log: App [npm:0] online
> zigbee2mqtt@1.15.0 start /zigbee2mqtt-1.15.0
> node index.js
Zigbee2MQTT:info  2020-10-03 23:43:24: Logging to console and directory: '/share/zigbee2mqtt/log/2020-10-03.23-43-23' filename: log.txt
Zigbee2MQTT:info  2020-10-03 23:43:25: Starting Zigbee2MQTT version 1.15.0 (commit #1.15.0)
Zigbee2MQTT:info  2020-10-03 23:43:25: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-10-03 23:43:44: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-03 23:43:44: Failed to start zigbee
Zigbee2MQTT:error 2020-10-03 23:43:44: Exiting...
Zigbee2MQTT:error 2020-10-03 23:43:44: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:92:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm 
ERR! zigbee2mqtt@1.15.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-10-03T21_43_44_913Z-debug.log
2020-10-03T23:43:45: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-10-03T23:43:45: PM2 log: App [npm:0] starting in -fork mode-
2020-10-03T23:43:45: PM2 log: App [npm:0] online

RoellT, 2020-11-30:
Hi,

today I was facing problems similar to above.

Using a CC2531, installed everything according to the zigbee2mqtt site. Nothing extra.

The way I worked;
I have multiple lights some remotes a motion sensor and a window sensor.
I have managed to put 10 devices in the zigbee2mqtt. But all of the sudden it stopped working. 
Yesterday I had the same problem after 15 devices. I have reflashed Home assistant and debugged the CC2531

2020-11-30T18:38:06: PM2 log: Launching in no daemon mode
2020-11-30T18:38:06: PM2 log: App [npm:0] starting in -fork mode-
2020-11-30T18:38:06: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.1 start /zigbee2mqtt-1.16.1
> node index.js
Zigbee2MQTT:info  2020-11-30 18:38:11: Logging to console and directory: '/share/zigbee2mqtt/log/2020-11-30.18-38-10' filename: log.txt
Zigbee2MQTT:info  2020-11-30 18:38:11: Starting Zigbee2MQTT version 1.16.1 (commit #1.16.1)
Zigbee2MQTT:info  2020-11-30 18:38:11: Starting zigbee-herdsman...
2020-11-30T17:38:11.694Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2020-11-30T17:38:11.700Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","adapter":{"concurrent":null}}'
2020-11-30T17:38:11.706Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2020-11-30T17:38:16.910Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2020-11-30T17:38:32.928Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2020-11-30T17:38:32.930Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2020-11-30T17:38:33.943Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-11-30T17:38:33.949Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-11-30T17:38:39.959Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-11-30T17:38:39.961Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2020-11-30T17:38:45.968Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2020-11-30T17:38:45.970Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
Zigbee2MQTT:error 2020-11-30 18:38:51: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-11-30 18:38:51: Failed to start zigbee
Zigbee2MQTT:error 2020-11-30 18:38:51: Exiting...
Zigbee2MQTT:error 2020-11-30 18:38:51: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 
ERR! code ELIFECYCLE
npm ERR! errno 1
npm 
ERR! zigbee2mqtt@1.16.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-30T17_39_27_600Z-debug.log

Hope you can help me! pretty new to this, sorry for my newbie description. If you need extra info i will try my best to retrieve it.

skrobul, 2021-01-08:
Just hit the same problem on CC2531 that was working correctly for weeks and stopped all of a sudden. I have tried reinserting the stick, rebooting raspberry pi and it did not help.
 
the logs are here: https://pastebin.com/dWJKkdxh

skrobul, 2021-01-09:
Reflashed the stick today and it's working correctly again. I will leave it running with debug logs enabled, so hopefully it will show some valuable information if it breaks again.

eldarmusin, 2022-01-06:
Hi @Koenkk 

**What kind of adapter is it?**
CC2652R
**What firmware is flashed on it?**
CC2652R_coordinator_20211217
**Did you ever manage to start Zigbee2mqtt with this adapter?**
first time, somehow started after several errors (see below)
**What is your OS and hardware?**
RPi4 + HA 2021.12.08 + OS 7.1
**If you every manage to start zigbee2mqtt adapter, did you change anything recently to this?**
first setup
**Did you check every step of the FAQ? Especially verify that you didn't got a CC2540 instead of a CC2531**
yes, port checked, R/W checked.

Using 2m USB cable for the connection.
I will try without and get you informed

https://pastebin.com/tJY1MfEC

Mainly:
> Error: AREQ - SYS - resetInd after 30000ms
> Error: SRSP - APP_CNF - bdbStartCommissioning after 40000ms
> Error: SRSP - SYS - getExtAddr after 6000ms

Obad94, 2023-12-07:
I was facing the same error. Resolved the error just by opening power shell as administrator in Windows OS.