Koenkk/zigbee2mqtt #2901: Repeating extRouteDisc errors after updating to 1.10.0

mLupine, 2020-02-06:
# Bug Report
## What happened

Last night I updated Zigbee2Mqtt to v1.10.0 and started getting lots of errors in the log since. The devices connected to Z2M work fine, but I think it shouldn't be happening anyway.

I'm using Z-Stack 3.0.x on my CC2530/CC2591 coordinator because for some reason Herdsman wasn't able to connect to it when I flashed it with either the source routing or the base 1.2 firmware.

```
zigbee2mqtt:debug 2020-02-06 10:35:13: Received Zigbee message from 'zigbee_router_1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
zigbee2mqtt:info  2020-02-06 10:35:13: MQTT publish: topic 'zigbee/zigbee_router_1', payload '{"description":"99/0x00158D0004658DF9","type":"CHILD_RFD","rssi":83,"linkquality":85,"state":true,"led_state":false}'
zigbee2mqtt:debug 2020-02-06 10:35:13: Received Zigbee message from 'zigbee_router_1', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"0/0x00124B001F32083F","inactiveText":"COORD","presentValue":0,"relinquishDefault":0,"minimumOffTime":0}' from endpoint 8 with groupID 0
zigbee2mqtt:info  2020-02-06 10:35:13: MQTT publish: topic 'zigbee/zigbee_router_1', payload '{"description":"0/0x00124B001F32083F","type":"COORD","rssi":0,"linkquality":85,"state":true,"led_state":false}'
zigbee2mqtt:debug 2020-02-06 10:35:13: Received Zigbee message from 'zigbee_router_1', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"99/0x00158D0004658DF9","inactiveText":"CHILD_RFD","presentValue":83,"relinquishDefault":2,"minimumOffTime":1}' from endpoint 8 with groupID 0
zigbee2mqtt:info  2020-02-06 10:35:13: MQTT publish: topic 'zigbee/zigbee_router_1', payload '{"description":"99/0x00158D0004658DF9","type":"CHILD_RFD","rssi":83,"linkquality":85,"state":true,"led_state":false}'
(node:18) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - extRouteDisc - {"dstAddr":53966,"options":2,"radius":30}' failed with status '199' (expected '0')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:94:5)
(node:18) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 456)
(node:18) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - extRouteDisc - {"dstAddr":53966,"options":2,"radius":30}' failed with status '199' (expected '0')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:94:5)
(node:18) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 457)
```

## How to reproduce it (minimal and precise)
- run a 3.0.x coordinator
- launch Zigbee2MQTT 1.10.0
- pair a CC2531 router to it
- observe the logs

## Debug Info
- zigbee2mqtt version: 1.10.0 (commit #8df6be8)
- Coordinator: Z-Stack_3.0.x 20190523 on CC2530 + CC2591
- Router: CC2531 running [this](https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/CC2531/bin/CC2531_router_2019_02.zip)
- Z2M connected to the coordinator via remote serial using an ESP8266 with ESPEasy

Koenkk, 2020-02-07:
Fixed in latest dev, thanks!

closed 2020-02-07 by Koenkk

timdonovanuk, 2020-02-19:
I'm still seeing this (same error I think?):

```
zigbee2mqtt:info  2020-02-19 17:34:27: MQTT publish: topic 'zigbee2mqtt/bathroom_1_light_2', payload '{"state":"OFF","brightness":254,"linkquality":81}'
(node:57) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - extRouteDisc - {"dstAddr":54724,"options":2,"radius":30}' failed with status '199' (expected '0')
    at Znp.<anonymous> (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
(node:57) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:57) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:57) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - extRouteDisc - {"dstAddr":36998,"options":2,"radius":30}' failed with status '199' (expected '0')
    at Znp.<anonymous> (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
(node:57) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
```

and I think the dev fix for this is merged into master?

Koenkk, 2020-02-20:
Pushed some more improvements yesterday tp the dev brancg

timdonovanuk, 2020-02-20:
Thanks @Koenkk ! Just sponsored you. Really appreciate the time/effort you put into this project. Cheers.