Koenkk/zigbee2mqtt #2902: Konke motion sensor 2AJZ4KPBS not working

mLupine, 2020-02-06:
# Bug Report
## What happened

Let me start with a bit of a background story. I purchased a Konke motion sensor (2AJZ4KPBS) to try and check if it's any better than the Aqara one. A few days ago I received it and added to my Zigbee network (based on Zigbee2Mqtt v1.9.0). It worked relatively fine. A few hours later I switched to a new coordinator, so I had to repair all the devices. Since the coordinator didn't want to run Z-Stack firmware v1.2, I uploaded v3.0.x to it and repaired the devices - including the Konke motion sensor.

And it worked fine once again. For a few hours.

When I woke up next morning, I noticed that it doesn't react to movement anymore. When I pressed the button on it, Z2M was receiving messages from the sensor but it wasn't able to use them. What's even more surprising was that it started mentioning "aqaraOpple", which are Xiaomi Aqara wall switches that I don't even own. 

Here's what Z2M knew about the device:
```
info  2020-02-06 00:10:13: MQTT publish: topic 'homeassistant/sensor/0x14b457fffe0771cb/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee/0x14b457fffe0771cb","json_attributes_topic":"zigbee/0x14b457fffe0771cb","name":"0x14b457fffe0771cb_battery","unique_id":"0x14b457fffe0771cb_battery_zigbee","device":{"identifiers":["zigbee2mqtt_0x14b457fffe0771cb"],"name":"0x14b457fffe0771cb","sw_version":"Zigbee2mqtt 1.10.0","model":"Motion sensor (2AJZ4KPBS)","manufacturer":"Konke"},"availability_topic":"zigbee/bridge/state"}'
info  2020-02-06 00:10:13: MQTT publish: topic 'homeassistant/binary_sensor/0x14b457fffe0771cb/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee/0x14b457fffe0771cb","json_attributes_topic":"zigbee/0x14b457fffe0771cb","name":"0x14b457fffe0771cb_occupancy","unique_id":"0x14b457fffe0771cb_occupancy_zigbee","device":{"identifiers":["zigbee2mqtt_0x14b457fffe0771cb"],"name":"0x14b457fffe0771cb","sw_version":"Zigbee2mqtt 1.10.0","model":"Motion sensor (2AJZ4KPBS)","manufacturer":"Konke"},"availability_topic":"zigbee/bridge/state"}'
info  2020-02-06 00:10:13: MQTT publish: topic 'homeassistant/binary_sensor/0x14b457fffe0771cb/battery_low/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.battery_low}}","device_class":"battery","state_topic":"zigbee/0x14b457fffe0771cb","json_attributes_topic":"zigbee/0x14b457fffe0771cb","name":"0x14b457fffe0771cb_battery_low","unique_id":"0x14b457fffe0771cb_battery_low_zigbee","device":{"identifiers":["zigbee2mqtt_0x14b457fffe0771cb"],"name":"0x14b457fffe0771cb","sw_version":"Zigbee2mqtt 1.10.0","model":"Motion sensor (2AJZ4KPBS)","manufacturer":"Konke"},"availability_topic":"zigbee/bridge/state"}'
info  2020-02-06 00:10:13: MQTT publish: topic 'homeassistant/sensor/0x14b457fffe0771cb/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee/0x14b457fffe0771cb","json_attributes_topic":"zigbee/0x14b457fffe0771cb","name":"0x14b457fffe0771cb_linkquality","unique_id":"0x14b457fffe0771cb_linkquality_zigbee","device":{"identifiers":["zigbee2mqtt_0x14b457fffe0771cb"],"name":"0x14b457fffe0771cb","sw_version":"Zigbee2mqtt 1.10.0","model":"Motion sensor (2AJZ4KPBS)","manufacturer":"Konke"},"availability_topic":"zigbee/bridge/state"}'
debug 2020-02-06 00:10:22: Received Zigbee message from 'office_multi_sensor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3015,"4":5032,"5":8,"6":[3,33554435],"10":10040,"100":1964,"101":4243,"102":101735}}' from endpoint 1 with groupID 0
```

And here's what the log looked like when I pressed a button on the sensor:
```
debug 2020-02-06 00:10:31: Device '0x14b457fffe0771cb' announced itself
debug 2020-02-06 00:10:38: Received Zigbee message from '0x14b457fffe0771cb', type 'attributeReport', cluster 'aqaraOpple', data '{"16":"\u0013\u0004\u0002\u0000\r111011720002087"}' from endpoint 1 with groupID 0
debug 2020-02-06 00:10:38: No converter available for '2AJZ4KPBS' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"16":"\u0013\u0004\u0002\u0000\r111011720002087"}'
debug 2020-02-06 00:10:38: Received Zigbee message from '0x14b457fffe0771cb', type 'attributeReport', cluster 'aqaraOpple', data '{"17":"\u0018\u0004\u0002\u0000\r332Q1P000209BS11002Q"}' from endpoint 1 with groupID 0
debug 2020-02-06 00:10:38: No converter available for '2AJZ4KPBS' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"17":"\u0018\u0004\u0002\u0000\r332Q1P000209BS11002Q"}'
debug 2020-02-06 00:10:40: Received Zigbee message from '0x14b457fffe0771cb', type 'attributeReport', cluster 'aqaraOpple', data '{"17":"\u0018\u0004\u0002\u0000\r332Q1P000209BS11002Q"}' from endpoint 1 with groupID 0
debug 2020-02-06 00:10:40: No converter available for '2AJZ4KPBS' with cluster 'aqaraOpple' and type 'attributeReport' and data '{"17":"\u0018\u0004\u0002\u0000\r332Q1P000209BS11002Q"}'
```

So, I tried re-pairing the sensor. I removed all the mentions of it in the configuration, state file, and the database, and then started from the beginning. And that's when the real problem started to surface.

I'm unable to pair the device anymore. The LED light on the sensor keeps blinking when I long-press the button but there's completely no sign of Z2M knowing that anything is happening. There's no log, no config entries.

## What did you expect to happen

I expected it to work 😢 

## Debug Info
zigbee2mqtt version: Started out on 1.9.0, through 1.9.0-dev, now on 1.10.0
CC2530+CC2591 firmware version: Z-Stack v3.0.x 20190523

Koenkk, 2020-02-07:
- I assume you are already on channel 15, 20 or 25 (as the devices requires this)
- Can you try disconnecting the battery for a while (or replacing it)?

djchen, 2020-02-26:
I just ran into a similar problem on this exact motion sensor. I am using one of those 3 channels and I have disconnected the battery for over a day. It reports in to Z2M  when I put the battery back in but it never reports occupancy true.

```
info  2020-02-25 20:20:42: MQTT publish: topic 'zigbee2mqtt/Dining Room Motion Sensor/availability', payload 'online'

info  2020-02-25 20:21:26: MQTT publish: topic 'zigbee2mqtt/Dining Room Motion Sensor', payload '{"battery":"60.00","voltage":2800,"linkquality":168,"last_seen":1582690886839,"occupancy":false,"tamper":false,"battery_low":false}'

info  2020-02-25 20:21:27: MQTT publish: topic 'zigbee2mqtt/Dining Room Motion Sensor', payload '{"battery":"60.00","voltage":2800,"linkquality":144,"last_seen":1582690887681,"occupancy":false,"tamper":false,"battery_low":false}'
```

With debug
```
debug 2020-02-25 20:25:57: Device 'Dining Room Motion Sensor' announced itself

debug 2020-02-25 20:25:58: Received Zigbee message from 'Dining Room Motion Sensor', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltage":28}' from endpoint 1 with groupID 0

info  2020-02-25 20:25:58: MQTT publish: topic 'zigbee2mqtt/Dining Room Motion Sensor', payload '{"battery":"60.00","voltage":2800,"linkquality":132,"last_seen":1582691158735,"occupancy":false,"tamper":false,"battery_low":false}'

```

Koenkk, 2020-02-26:
Could you sniff the traffic when the device stopped working? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

evanvruggink, 2020-03-22:
I cannot seem to add the Konke 2AJZ4KPBS motion sensor, i get the following error:
  zigbee2mqtt:warn 3/18/2020, 6:37:39 PM Device with modelID '3AFE28010402000D' is not supported.
  zigbee2mqtt:warn 3/18/2020, 6:37:39 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

I have the latest version of Hassio and zigbee2mqqt running

Koenkk, 2020-03-22:
Please try the edge verison, it should be added there.

evanvruggink, 2020-04-04:
> 
> 
> Please try the edge verison, it should be added there.

Great that worked! After switching the channel to 15 (and unfortunately re-pairing all the devices) it is now up and running. I already like it more than the xiaomi motion sensor with the limited field of view!

ramakersr, 2020-05-30:
I can't get it to work with this sensor:
tried on channels 15,20 and 25.
nothing in the log about this sensor
cc2531 with CC2531_20190425 firmware.
zigbee2mqtt version 1.13.1.1 and also tried with the actual edge version.

closed 2020-07-26 by stale[bot]