Koenkk/zigbee2mqtt #2907: Error starting Zigbee Herdsman

Davepw4, 2020-02-06:
#Bug Report
## What happened
Couldn't start zigbee herdsman

Apologies as I'm a complete novice, picked up home assistant about 6 months ago and been adding to it as I learn. I've spent the last 2 weeks trying to get zigbee2mqtt working but just getting the same result. I've double and triple checked all the spacings in the config, reflashed the CC2531 a few times but still the same issue.

Just get the below from home assistant and firing it up on home assistant

`./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-02-06T22:10:57: PM2 log: Launching in no daemon mode
2020-02-06T22:10:57: PM2 log: App [npm:0] starting in -fork mode-
2020-02-06T22:10:57: PM2 log: App [npm:0] online
> zigbee2mqtt@1.10.0 start /zigbee2mqtt-1.10.0
> node index.js
zigbee2mqtt:info  2020-02-06 22:11:05: Logging to console and directory: '/share/zigbee2mqtt/log/2020-02-06.22-11-05' filename: log.txt
zigbee2mqtt:info  2020-02-06 22:11:06: Starting zigbee2mqtt version 1.10.0 (commit #unknown)
zigbee2mqtt:info  2020-02-06 22:11:06: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-02-06 22:11:56: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-02-06 22:11:56: Failed to start zigbee
zigbee2mqtt:error 2020-02-06 22:11:56: Exiting...
zigbee2mqtt:error 2020-02-06 22:11:56: Error: AREQ - SYS - resetInd after 30000ms
    at Timeout._onTimeout (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.10.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.10.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-06T22_11_56_148Z-debug.log
2020-02-06T22:11:56: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-02-06T22:11:56: PM2 log: App [npm:0] starting in -fork mode-
2020-02-06T22:11:56: PM2 log: App [npm:0] online
> zigbee2mqtt@1.10.0 start /zigbee2mqtt-1.10.0
> node index.js`

This is the log from starting on raspberry pi

`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.13.4
3 info using node@v12.15.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.9.0~prestart: zigbee2mqtt@1.9.0
6 info lifecycle zigbee2mqtt@1.9.0~start: zigbee2mqtt@1.9.0
7 verbose lifecycle zigbee2mqtt@1.9.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.9.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@1.9.0~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.9.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.9.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.9.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.9.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:223:5)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:223:5)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
14 verbose pkgid zigbee2mqtt@1.9.0
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.19.93-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v12.15.0
19 verbose npm  v6.13.4
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.9.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.9.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
`

GMFalka, 2020-02-06:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/2899

Koenkk, 2020-02-07:
Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

Davepw4, 2020-02-09:
Hi, yes I've done everything in that link and it's all coming back fine. Is there anyway to isolate whether its the CC2531 flashing or the MQTT login which is the problem?

Koenkk, 2020-02-11:
The CC2531 is the problem, but lets continue in #2899

closed 2020-02-11 by Koenkk