Koenkk/zigbee2mqtt #2908: Devices lost after hassio zigbee2mqtt restart

autopower, 2020-02-07:
# Bug Report
Environment:
Hassio on Synology, docker
Z2M add on, 1.10.0
CC2530 connected to ESP8266, EspEasy as serial server
Looks like similar issue: https://github.com/Koenkk/zigbee2mqtt/issues/1657 but there is no interference 

## What happened
All devices are lost after z2m addon restart. If device is successfully joined (see logs attached) and Z2M addon is restarted, connection to device is lost and device not reporting correct values to hassio. 

## What did you expect to happen
Device joined after restart automatically without manual rejoin.

## How to reproduce it (minimal and precise)
1) Start Z2M addon
1) join device
1) use device, everything is reported ok
1) restart Z2M addon
1) communication with device lost

## Debug Info
zigbee2mqtt version: 1.10.0
CC2530 firmware version: Z-Stack_Home_1.2 with source routing
ESPEasy: ESP_Easy_mega-20191208_normal_ESP8266_4M1M
Device: Xiaomi, as stated in log
[after-restart.txt](https://github.com/Koenkk/zigbee2mqtt/files/4169673/after-restart.txt)
[configuration.txt](https://github.com/Koenkk/zigbee2mqtt/files/4169674/configuration.txt)
[succ_join.txt](https://github.com/Koenkk/zigbee2mqtt/files/4169675/succ_join.txt)

Koenkk, 2020-02-07:
It looks like your `database.db` file is cleared on every restart, can you verify this?

autopower, 2020-02-07:
Yes, `database.db` is cleared on every start. Only coordinator entry (`id: 1`) is present after restart

Koenkk, 2020-02-07:
Is the database.db populated with devices after pairing them? This doesnt look like a direct zigbee2mqtt issue but rather with the addon (https://github.com/danielwelch/hassio-zigbee2mqtt/issues) or hassio

Koenkk, 2020-02-07:
Can you switch the the edge addon and set `zigbee_herdsman_debug` to `true` after that start zigbee2mqtt and provide the log

autopower, 2020-02-07:
Edge addon working ok, logs attached (docker log export, in csv format):
[after-addon_7ad98f9c_zigbee2mqtt_edge.txt](https://github.com/Koenkk/zigbee2mqtt/files/4172183/after-addon_7ad98f9c_zigbee2mqtt_edge.txt)
[before-addon_7ad98f9c_zigbee2mqtt_edge.txt](https://github.com/Koenkk/zigbee2mqtt/files/4172184/before-addon_7ad98f9c_zigbee2mqtt_edge.txt)
What is the cause of the problem?

Koenkk, 2020-02-07:
Something goes wrong with the network key, could it be that the `network_key` you specified is too long? Should be 16 values (https://www.zigbee2mqtt.io/how_tos/how_to_secure_network.html#change-zigbee-network-encryption-key)

autopower, 2020-02-07:
Oh, my fault! not 16 but 20!!! bytes! Let me check if its working in edge and/or "normal" release.

closed 2020-02-07 by autopower

Koenkk, 2020-02-08:
Fixed now, zigbee2mqtt will refuse to start when an incorrect length is set. (and print an error message about it).