Koenkk/zigbee2mqtt #2909: Support for Hue bulb model 9290018217

rogers33github, 2020-02-07:
Hello,
I have tried to manually amend the devices file myself, but failed at the first hurdle, I can ssh to my pi but when i try and run docker it doesn't like it.

I have a Philips Hue Warm White, A60 bulb, model 9290018217 which isn't listed in the supported devices.  It fails the interview, suspecting its because its not in the supported listed.  Looking in zigbee2mqtt devices.yaml i ahve the following entry.

{"id":9,"type":"Router","ieeeAddr":"0x001788010602a353","nwkAddr":37355,"manufId":4107,"epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64514],"outClusterList":[25],"clusters":{},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{}}

I dont't have a hue bridge, so i can't reset the bulb either.
Many thznks
Mark

Koenkk, 2020-02-07:
Can you try repairing it and provide the databse.db after that? Interview should succeed (even though its not supported yet).

You can reset the bulb via touchlink:  https://www.zigbee2mqtt.io/information/touchlink.html

rogers33github, 2020-02-08:
I've reset via touchlink, but the bulb isn't re-pairing.  Upgraded from 1.09 to 1.10 and its made no difference. Turned the bulb off and on, and restarted Zigbee2Mqtt and still not being seen.

turboproc, 2020-02-08:
Hi @rogers33github ,
what if you would trigger z2m to reconfigure and provide a logging of the complete process whilst having z2m in debug mode? Use
````
zigbee2mqtt/bridge/configure
````
 for this as documentend in [https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html](url)

rogers33github, 2020-02-08:
I left it a few hours and tried again, and it successfully paired.  here is the line from database.db
{"id":9,"type":"Router","ieeeAddr":"0x001788010602a353","nwkAddr":17233,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LWA001","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64514],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LWA001","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190122","swBuildId":"1.53.3_r27175"}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20190122","swBuildId":"1.53.3_r27175","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1581176799017}

If you need anything else, just ask.
Thanks for your support

Koenkk, 2020-02-08:
This device is already supported in zigbee2mqtt 1.10.0

rogers33github, 2020-02-08:
Cheers, Yes I upgraded to 1.10 then once it finally paired it worked.

closed 2020-02-08 by rogers33github