Koenkk/zigbee2mqtt #2910: Unable to start zigbee2mqtt - Error: AREQ - SYS

dvdbosch, 2020-02-07:
# Bug Report
## What happened
Unable to start zigbee2mqtt

## What did you expect to happen
Starting the service :-)

## How to reproduce it (minimal and precise)
Not none!
## Debug Info
zigbee2mqtt version: 1.10
CC253X firmware version: Most recent (not flashed by myself)
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@raspberrypi:~ $ cat /sys/firmware/devicetree/base/model
Raspberry Pi 2 Model B Rev 1.1pi@raspberrypi:~ $

##logging available:
https://hastebin.com/veqocerilu.sql

ginkel, 2020-02-07:
Have you tried power-cycling your Zigbee USB stick? Does the stick show up using lsusb? Is your configuration.yaml pointing to the right device?

Koenkk, 2020-02-08:
Duplicate of #2899

closed 2020-02-08 by Koenkk