Koenkk/zigbee2mqtt #2911: IKEA LED1623G12 ruins network

ChrisSantiago82, 2020-02-07:
Hello,
I have a network of zigbeedevices for a while. Everything is stable.
I have a 4 Securifi Peanut Smart Plugs, and 6 Motion Sensors (Ikea and others).
Everything runs stable, some of the sensor are at the edge of the network.

I'm trying to add now 4 IKEA LED1623G12 Bulbs. They are in good positions and should not only bring lights, they should also improve the network by routing it.
 Pairing and everything works. But approximately after 1 day sometimes less, everything breaks down.

- If I try to send a message to the lights, I'm getting a No network route' (205). But the lights are in range because they were working well all day.

- And the Plugs crush now as well. Here I'm getting a 'MAC no ack' (233)'. And it's the same with all of them (4 Plugs). The sensors that are still in rage work. The once that are depending on the Plugs or Lights to relay the message, of course stop working.

- Just to rejoin the devices is not possible. Even if they are just next to the Raspberry/USB. If I delete them out of the Database, sometimes they can be rejoined but mostly not and when this happens, I have to delete everything and pretty much ending up Repairing the whole network.
Sometimes this also happens when in the process of repairing the network. As I said, if I do it without the IKEA bulbs, I don't have those problems.

After 2 weeks of doing this, repairing everything daily, I have removed the ikea bulbs yesterday, and now everything runs stable.
I don't think it's a range/network problem, because it should be much worse without the bulbs than with them. They should lead the path to all of the devices.

What can I do to improve this situation?

tricoos, 2020-02-08:
With the IKEA devices (10) I also have a few issues here:
- I have two FLOALT panels in the hallway around 1.5 meters alway from each other. The second one which is more far away from the Zigbee stick often needs a second MQTT message to turn on.
- the IKEA remotes delivered with the FLOALT panels also drop out off the network every now and then.
- I also have many 205 errors every now and then

With the single Aqara bulb I have I do not see this issue.

I also have the feeling that IKEA devices and Zigbee2Mqtt do not work well together. I am now actually importing Zigbee error messages into InfluxDB and visualize them in Grafana to get an overview.
[image]

Doesn't look good  most often these are IKEA lamps reporting a 205.

ChrisSantiago82, 2020-02-08:
Well, my problem is that I'm getting to the point where I'm only receiving 205 from the bulbs, and the other devices 233. Like it's not once in a while and that brings down the whole network.
Now I have my network up without the bulbs for a couple of days and it's works perfectly...

tricoos, 2020-02-08:
It most often helps for me to restart zigbee2mqtt, but then again it's not all bulbs at the same time reporting 205 but as you can see from the spikes in the graph there might be some kind of correlation.
I cannot take out the FLOALT panels because these are the main lighting in this apartment, unfortunately.

ChrisSantiago82, 2020-02-08:
When I'm at this stage, restarting zigbee2mqtt doesn't help anymore. Even restoring the Database doesn't help. At this moment the only thing that seems to work is delete everything and start from zero again

tricoos, 2020-02-08:
If nothing at all works anymore for me I stop zigbee2mqtt, unplug the stick, wait a few seconds, repg the stick and then start zigbee2mqtt again. Maybe the stick is locking up?

ChrisSantiago82, 2020-02-08:
No, that doesn't help either. And just to repair the devices doesn't help neither. I really have to delete them out of the database.

tricoos, 2020-02-08:
Yes, I have to remove them as well.

ChrisSantiago82, 2020-02-08:
Now the big question, how can we improve it? How can we improve the ikea bulbs support?

sjorge, 2020-02-08:
Just out of interest, are you on ikea's newer zigbee 3.0 firmware or still on the older one?

ChrisSantiago82, 2020-02-08:
I'm sorry I'm not so experienced with this. Where can I find the ikea zigbee firmware version? And how could I update it?

sjorge, 2020-02-08:
[zigbee2mqtt/bridge/config/devices/get](
https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigdevicesget) should have the firmware info in the return payload.

For now, you need the official ikea gateway to update the firmware... but this will soon be possible using zigbee2mqtt too... but that is still in development.

Release notes are here: https://ww8.ikea.com/ikeahomesmart/releasenotes/releasenotes.html

e.g.

```
WS 1.0 (E14, E27, GU10) V-2.3.007.
Devices are upgraded with ZB3.0.
```

Means that firmware 2.3.007 or newer will be running zigbee 3.0 instead of the older 1.2 variant.

ChrisSantiago82, 2020-02-08:
Here is the Log File. I have joined one of the bulbs before I made this log. There are also some ikea motion detectors, but with those I never had an issue. The Light bulb has the address 0x000b57fffebb51a5

[{"ieeeAddr":"0x00124b0018e32224","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190619","lastSeen":1581180833427},{"ieeeAddr":"0x000d6f0004a87e3a","type":"Router","networkAddress":37390,"model":"PP-WHT-US","friendly_name":"Plug1","manufacturerID":4098,"manufacturerName":"Securifi Ltd.\u0000\u0000\u0000\u0000\u0000\u0000�W\u0001\b�#\u0000 �Y\u0001\b�","powerSource":"Unknown","modelID":"PP-WHT-US","lastSeen":1581180735922},{"ieeeAddr":"0x000d6f0004f75240","type":"Router","networkAddress":57446,"model":"PP-WHT-US","friendly_name":"Plug2","manufacturerID":4098,"manufacturerName":"Securifi Ltd.\u0000\u0000\u0000\u0000\u0000\u0000�W\u0001\b�#\u0000 �Y\u0001\b�","powerSource":"Unknown","modelID":"PP-WHT-US","lastSeen":1581180735908},{"ieeeAddr":"0x00158d0002f3c10c","type":"EndDevice","networkAddress":20534,"model":"RTCGQ11LM","friendly_name":"Sensor1","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_motion.aq2","lastSeen":1581114058933},{"ieeeAddr":"0x000b57fffe956f3d","type":"EndDevice","networkAddress":47301,"model":"E1525/E1745","friendly_name":"Sensor5","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI motion sensor","hardwareVersion":1,"softwareBuildID":"1.1.1.1-5.7.2.0","dateCode":"20160927","lastSeen":1581078780260},{"ieeeAddr":"0x000d6f000a784aa0","type":"Router","networkAddress":23927,"model":"PP-WHT-US","friendly_name":"Plug5","manufacturerID":4098,"manufacturerName":"Securifi Ltd.\u0000\u0000\u0000\u0000\u0000\u0000�W\u0001\b�#\u0000 �Y\u0001\b�","powerSource":"Unknown","modelID":"PP-WHT-US","lastSeen":1581180737449},{"ieeeAddr":"0x000d6f000a77ed09","type":"Router","networkAddress":13096,"model":"PP-WHT-US","friendly_name":"Plug99","manufacturerID":4098,"manufacturerName":"Securifi Ltd.\u0000\u0000\u0000\u0000\u0000\u0000�W\u0001\b�#\u0000 �Y\u0001\b�","powerSource":"Unknown","modelID":"PP-WHT-US","lastSeen":1581180735997},{"ieeeAddr":"0x000b57fffe9bc1ea","type":"EndDevice","networkAddress":23759,"model":"E1525/E1745","friendly_name":"Sensor2","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI motion sensor","hardwareVersion":1,"softwareBuildID":"1.1.1.1-5.7.2.0","dateCode":"20160927","lastSeen":1581077533452},{"ieeeAddr":"0x000b57fffe92cc8f","type":"EndDevice","networkAddress":10078,"model":"E1525/E1745","friendly_name":"Sensor3","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI motion sensor","hardwareVersion":1,"softwareBuildID":"1.1.1.1-5.7.2.0","dateCode":"20160927","lastSeen":1581026684768},{"ieeeAddr":"0x000d6f000a76de47","type":"Router","networkAddress":46009,"model":"PP-WHT-US","friendly_name":"Plug4","manufacturerID":4098,"manufacturerName":"Securifi Ltd.\u0000\u0000\u0000\u0000\u0000\u0000�W\u0001\b�#\u0000 �Y\u0001\b�","powerSource":"Unknown","modelID":"PP-WHT-US","lastSeen":1581180784711},{"ieeeAddr":"0x000b57fffe9653f8","type":"EndDevice","networkAddress":3847,"model":"E1525/E1745","friendly_name":"Sensor4","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI motion sensor","hardwareVersion":1,"softwareBuildID":"1.1.1.1-5.7.2.0","dateCode":"20160927","lastSeen":1581026190584},{"ieeeAddr":"0x000b57fffe920d2a","type":"EndDevice","networkAddress":34440,"model":"E1525/E1745","friendly_name":"Sensor7","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI motion sensor","hardwareVersion":1,"softwareBuildID":"1.1.1.1-5.7.2.0","dateCode":"20160927","lastSeen":1581026559995},{"ieeeAddr":"0x000b57fffe9b47ac","type":"EndDevice","networkAddress":31983,"model":"E1525/E1745","friendly_name":"Sensor6","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Battery","modelID":"TRADFRI motion sensor","hardwareVersion":1,"softwareBuildID":"1.1.1.1-5.7.2.0","dateCode":"20160927","lastSeen":1581026557483},{"ieeeAddr":"0x000b57fffebb51a5","type":"Router","networkAddress":3080,"model":"LED1622G12","friendly_name":"0x000b57fffebb51a5","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelID":"TRADFRI bulb E26 W opal 1000lm","hardwareVersion":1,"softwareBuildID":"1.2.214","dateCode":"20170302","lastSeen":1581180739380}]

ChrisSantiago82, 2020-02-08:
If now the sodwareBuildID is 1.2.214, that means it's the old version?

ChrisSantiago82, 2020-02-08:
But maybe I would have to do this with all the bulbs? Maybe there are different versions?

sjorge, 2020-02-08:
Yep, looks like they are still on the ZB1.2 firmware, I have had less issues with those than with my one on ZB3.0 that keeps dropping out of the network.

ChrisSantiago82, 2020-02-08:
Well, my problem isn't that they drop out of the network. My problem is, that they drop out and take down the whole network as well.
But if you say you have more problem with the newer version, than my hope isn't wait for the firmware upgrade;)

sjorge, 2020-02-08:
Mine isn't acting as a router, as all device in that room use a hue bulb. If your bulb is, that would explain the entire network drop.

tricoos, 2020-02-14:
It just happened from one minute to the next: my IKEA lamp in the toilet dropped out of the network ("Error: Data request failed with error: 'No network route' (205)").
I noticed something interesting:
- as mentioned before you can obviously only rejoin IKEA lamps to the network by removing them from the config and restarting zigbee2mqtt
- my Aqara bulb however was off for many hours when I found that the switch had been turned off accidentally. I turned it on, the lamp turned on, I closed the door (it's in a closet) which has an Aqara door sensor and when I closed it the light turned off immediately.

It looks like there is an issue with the IKEA lamps - we just need to find out if we can do anything about it or if it is an issue IKEA would have to fix (and probably wouldn't as long as it works with their gateway).
I was really astonished how qucikly the Aqara bulb rejoined the network.

ChrisSantiago82, 2020-02-14:
Yes, I can understand that sometimes a device can drop out or loose connection. But it shouldn't be that it can rejoin it or even bring down other devices that have a good connection without the bulb as well.
Yesterday I had the same situation again, the closest bulb lost connection and gave at 205. Usually I would have to remove it from the config and database and rejoin it again.
This time I only remove the USB stick. Without stopping or starting the zigbee2mqtt. Just removed it and put it back in. Of course in other times I restarted the process, or Raspberry. And nothing helped.
Now just to unplug the USB and put it back solved the issue with that one bulb. I didn't have to rejoin it.

ChrisSantiago82, 2020-02-14:
The bulb that is the farthest away, I didn't put it back again. And since then the whole network is much more stable.
I can understand if it's at the limit that this one bulb has problems. But why does it bring down the whole network?

tricoos, 2020-02-14:
Weird, I just tried this right now as now a second lamp has gone offline with a 205. It worked - I just unplugged the stick and replugged it.
However, I had to start Zigbee2Mqtt manually because....:
```
zigbee2mqtt:error 2020-02-14 20:00:24: Adapter disconnected, stopping
zigbee2mqtt:info  2020-02-14 20:00:24: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
zigbee2mqtt:info  2020-02-14 20:00:24: Disconnecting from MQTT server
zigbee2mqtt:error 2020-02-14 20:00:24: Failed to stop zigbee
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.10.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.10.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

```
But this actually solved the problem!

ChrisSantiago82, 2020-02-14:
Maybe this is the problem isn't just on the lamp side?

Tom85649, 2020-02-16:
Hi !  I had the same problems but with Osram. Seems to occur when communication goes via more than one router. Anyway. I don't have a solution but found this work around. Create for each device its own group and communicate only with the group. The difference I believe is that group communication is via broadcast messages. In that config I don't have no 205 or 233 error messages no more and it is 99.99 % reliable. Also very remote devices in the garden which produced many 205 or 233 problems I can reach now via group communication without any problems.

ChrisSantiago82, 2020-02-16:
Hi,
This is a good idea. I will try it out. Thanks a lot!

tricoos, 2020-03-02:
@Tom85649  Adding a device to its own group sounds great. Does this also support passing additional parameters to the group? For example the color temperature? I need to set this for the lamps depending on the time of day. This is not mentioned explicitly in the documentation. But the documentation mentions that these use broadcast :)

Tom85649, 2020-03-02:
Hi tricoos,
for my OSRAM garden poles I am using this
        {"color_temp":370,"color":{"x": 0.458,"y":0.41}}
to get the color I want (the warm white color they also show on start-up). However, with
        {"color":{"x":0.458,"y":0.41},"color_temp":370}, 
(= different order of "color" and "color temp") it does not work; but it did work in in this other order in non-group communication. So it seems there is some difference in communicating with groups compared to communicating with single devices.  I am using the group communication to control my 4 OSRAM garden pole devices for about 6 months now. And there wasn't a single day with problems. No 205 or 233 problems. Lights go on and off like perfect.

tricoos, 2020-03-03:
@Tom85649 Thanks! I added groups for the IKEA lamps and - most of the time - they switch on a whole lot faster! Interestingly I do not see the problems you mention here, but that's maybe because I already did it the suggested way :)
Sometimes though they seem "stuck" but as I'm mostly using IKEA buttons to turn on the lamps I cannot find out whether it depends on the lamps or the buttons.
I will check if I will see anymore 205 errors. Thanks!

jk-5, 2020-03-24:
I have exactly the same issues since about a week. I did not update any device firmwares. It seemed to start after the update to zigbee2mqtt 0.11.0, but that might be unrelated.
I have tried moving to the source routing coordinator firmware since i have ~30 devices. I have not added any devices in the last few months, so even that is not the issue as far as i can see.

What i'm experiencing is that all my tradfri plug switches become unresponsive with the 205 and 233 errors. All other devices still work. I am able to get them to work by resetting the plug, but after one or two hours, the setup breaks itself again.

I really don't have a clue where to look for the issues. I assume there is a device in the network that triggers this. My network consists of mostly ikea devices, with 5 hue bulbs. No other vendors.

tricoos, 2020-03-24:
I suggest adding the devices to one group per device as suggested by @Tom85649  as it works very well for me since three weeks.

jk-5, 2020-03-27:
You can ignore my problems here. I fixed it by reverting to the normal firmware from source-routing. Thanks for the suggestion anyways

ChrisSantiago82, 2020-03-27:
Thank you, I have tried booth. With and without source-routing. It doesn't make any difference here.
I haven't tried out the Grouping yet. But just removing and reenter the USB Sniffer already resolves the network without the need of pairing everything again. With that it's less urgent..

closed 2020-06-06 by stale[bot]