Koenkk/zigbee2mqtt #2912: Using CC1352-P2 external antenna: increased "Failure to ping"

nanobra1n, 2020-02-08:
# Bug Report
## What happened
I have 2 CC1352-P2: one stock, one with the hardware modification to use the external antenna. WIth the pcb antenna I have one device that is becoming unavailable on a regular basis. With the external one I have this happening all over my network. and the whole network became unstable. Looking at the networkmap I see more direct connections to the coordinator using the external antenna as I would expect, but I am now wondering if thats really a good thing.
 
I only waited a bit over an hour for the network to settle, and then switched back to the coordinator  with the pcb antenna.

## What did you expect to happen
Making my network more stable, not less 
## How to reproduce it (minimal and precise)
Get 2 CC1352-P2s, find someone with the microscope, the steady hands and 2 small tipped soldering irons to move the capacitor between the pcb and the sma antenna trace. Take network maps for both configurations. 
## Debug Info
zigbee2mqtt version: 1.10
CC1352-P2 firmware version: {"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}} 

[External antenna log](https://pastebin.com/tbDEfyGc)
[External antenna network map raw](https://pastebin.com/YHDvg4K1)

[PCB antenna log](https://pastebin.com/K46dLj0Z)
[PCB antenna network map raw](https://pastebin.com/tGVqJSvQ)

CodeFinder2, 2020-02-08:
I am not having such issues, see also https://github.com/Koenkk/zigbee2mqtt/issues/2162#issuecomment-570286663 but I also haven't used the CC1352P-2 with the PCB antenna for long (and only in a '1 or 2 device test setup'). 

Do both boards have the exact same revisions? Are you sure that your resoldering mod was successful? 🙈 

Out of curiosity, how many devices are in your network? Anyway, I think more direct connections are good iff their signal strengths is 'good enough'.

Just ideas coming to my mind...the way this has to be reproduced is... "not easy". 😅🙈

nanobra1n, 2020-02-09:
Both boards are HW Rev B, can't find the chip Rev anywhere 😞. I am confident that the mod worked because:
- I still can see all my 18 devices connected in the network map.
- The link qualities both incoming and outgoing for the external antenna are better (incoming greater improvement than outgoing) 

I was wrong with my earlier statement: with the external antenna I have 6 direct outgoing connections, with the pcb antenna I have 12. I have to investigate more if the failure to ping affects devices that have to go through routers, but right now my wife would not appreciate if I started mesing around with her ability to turn lights on/off. Also maybe I should have waited longer for the mesh to settle. 

I have 15 routers (8 Cree bulbs, 2 OSRAM LED strips, 2 Peanut, 1 centralite and 2 innr smart plugs and 3 end devices - 2 Xiaomi sensors and 1 Xiaomi button). 

Thanks for the link to the issue, very informative. I have a second case in the works, I might insulate the inside with copper foil as well. One question: how did you set transmit power? Is that currently configurable?

CodeFinder2, 2020-02-09:
> One question: how did you set transmit power? Is that currently configurable?

See here https://github.com/Koenkk/zigbee2mqtt/issues/2253

nanobra1n, 2020-02-11:
I am going to close this because I think that the external antenna was a red herring. I now see increased occurrences of "Failure to ping" on my  pcb-antenna coordinator as well. It might be a 1.10 issue, since I upgraded just before I started the comparison, though I fail to see anything in the release notes that could have impacted this.

closed 2020-02-11 by nanobra1n