Koenkk/zigbee2mqtt #2913: Unable to pair Letv Zigbee remote control

myiotcenter, 2020-02-08:
# Bug Report
Hi @Koenkk 
I bought this very cheap ($2) Letv Zigbee remote controller with 8 buttons I thought I could use but I couldn't pair at all. These things were sold as an as is condition in Taobao so I'm not sure if they are defective or this failure was caused by something else.

I've tried following steps but still failed:
1. Take off the battery for some times and try again
2. Pair the device right next to the coordinator
3. Replace the battery
4. Switch to Dev branch, using edge in hassio add-on

This is the log while trying to pair:
https://pastebin.com/Uvg5m1A5

These are the pictures of the device :
[image]
[image]
[image]

## What did you expect to happen
Device can be paired so I can create custom entry for the device.

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: Hassio Add-on 1.10.0
CC253X firmware version:
CC2538 + CC2592 by @reverieline (MODKAMRU_V3_UART-no-flow-control.hex)

closed 2020-04-17 by stale[bot]

tercioglu, 2021-12-01:
How do you pair the original firmware to Zigbee2mqqt?

tercioglu, 2021-12-06:
what buton to press for pairing?

vasil-github, 2022-01-11:
I also have issues paring this device.
tried deferent combos and nothing.....
latest ZB2MQTTT version to date and CC2538 coordinator.

tercioglu, 2022-03-06:
this is strange 
this item is listed as supported.
however NO one knows how to pair.

what are the steps to pair?
Letv
letv_8
letv 8

vasil-github, 2022-03-08:
I hope some will share but...
Interesting is that i did research the info for 7 days and did not find anything.
In some cases it states that to pair you have to hold button 1 (left arrow symbol) for 10 seconds, but it does not work in my case.
Also did try to pair after removing the battery and putting it again and button 1 for 10 seconds or more but NADA...

tercioglu, 2022-03-08:
"have to hold button 1 (left arrow symbol) for 10 seconds" 
 this action is for "https://github.com/diyruz/freepad" firmware

vasil-github, 2022-03-08:
Yes, maybe . I did buy the device 2 months ago, and after one week of struggling it is e-waste now unfortunately .

tercioglu, 2022-03-11:
Oh No it is NOT.
https://diyruz.github.io/posts/letv/
https://modkam.ru/2020/11/12/pult-zigbee-letv/

just update the firmware to
https://github.com/diyruz/freepad

lglw, 2022-10-10:
This still does not work on the original remote firmware.

booroondook-eng, 2023-03-10:
With original firmware try to unplug/re-plug battery for 5 times.

vasil-github, 2023-03-10:
> With original firmware try to unplug/re-plug battery for 5 times.

And then what?
What button combination or witch button for how long to hold down for device pairing?

booroondook-eng, 2023-03-10:
> And then what? What button combination or witch button for how long to hold down for device pairing?

After the fifth time, just finally insert the battery into the socket and fix it. Then wait for pairing. You can also short press the OK button if you want the LED to turn on.

However, I will disappoint you. Zigbee2MQTT cannot work with this device with native firmware. Zigbee2MQTT recognize it as "unknown device" and cannot interview it.