Koenkk/zigbee2mqtt #2914: Can't pair anything

Jeoffreybauvin, 2020-02-08:
Hi there,

I have a major issue with Zigbee2mqtt. 

This was my Zigbee map last night :

[image]

And my configuration file  : 

```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://broker:8883'
  ca: /opt/zigbee2mqtt_ssl/ca.crt
  key: /opt/zigbee2mqtt_ssl/zigbee2mqtt.key
  cert: /opt/zigbee2mqtt_ssl/zigbee2mqtt.crt
advanced:
  log_level: debug
  network_key:
...
serial:
  port: /dev/ttyACM0
```

All my sensors worked fine. I tried to plug new devices : 

- Ikea Bulb (LED1623G12)
- Ikea plug wall (E1603/E1702)
- Xiaomi Motion Sensor (RTCGQ11LM)

I was not able to pair any of these devices. Nothing in z2mqtt's logs :(. 

The coordinator is a ready2use CC2531 with antenna from h4nc. 

So I took an old CC2531, and flashed it as a router (CC2531_router_2019_02).  Again, nothing in log file.
 
I decided to start from scratch : 

- Stop z2mqtt
- Delete state.json, database.db,  all devices in configuration file
- Start zigbee2mqtt

Now,  I only have my coordinator. I'm trying to pair my CC2531 router : nothing in log file, again. 

My router is blinking red, but I don't see it in z2mqtt...

I'm not sure what to do now...  I though I'll pair all my routers before pairing any  end device...

Any ideas ?

Regards,

lancelon, 2020-02-08:
Have had very similar experiences myself, see my various comments

CodeFinder2, 2020-02-08:
What version of Z2M are you using?

tdn131, 2020-02-12:
I have the same issue. Nothing pairs. Nothing is shown in logs. I am using Z2M 1.10 and the dev of today. I try to re-pair a IKEA Signal amplifier and on/off switch.

lancelon, 2020-02-12:
@tdn131 so many people seem to report that reflashing the firmware fixes problems like this. I wonder what happens to the CC2531...

talondnb, 2020-02-17:
I've been having trouble re-pairing IKEA bulbs only and on both 1.10 and dev. Xiaomi devices seem to re-pair fine on 1.10. The only way I could get an IKEA bulb back on the network was to rollback to a previous snapshot in Hassio.

closed 2020-04-25 by stale[bot]