Koenkk/zigbee2mqtt #2918: Trying to install zigbee2mqtt on a Pi 2 and getting "npm ERR! Maximum call stack size exceeded"

lancelon, 2020-02-08:
I'm following [the official guide](https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html) and I'm just getting 

`npm ERR! Maximum call stack size exceeded
`

```
root@raspbx:/media/usb/zigbee2mqtt# rm -rf node_modules
root@raspbx:/media/usb/zigbee2mqtt# npm i
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.11 (node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.11: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman-converters/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.2 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

npm ERR! Maximum call stack size exceeded

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-08T13_52_38_715Z-debug.log
```

I've been Googling for solutions to the problem for the last 24 hours and despite trying all manner or force deletes, cache flushes, attempts to increase my maximum call stack - nothing works or helps.

Any ideas?!

Creo2005, 2020-02-09:
The same problem:
many: npm WARN
and in end of install:
npm ERR! Maximum call stack size exceeded
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-09T11_21_21_731Z-debug.log

HedgU4, 2020-02-12:
Have the same problem with Raspberry Pi Zero and node v11.x
with node v10.9.0 all working fine

Koenkk, 2020-02-12:
https://github.com/Koenkk/zigbee2mqtt/issues/2934#issue-563273044 suggests to use `npm ci` instead of `npm install`, can you try?

9SMTM6, 2020-02-12:
Yeah for me installation succeeded with `npm ci`, and `ci` probably reduces stack depth, though I was also using node v10.x (I believe it was 10.4, but not sure, I just took what was in the default repository and it ended up working). Was using an Raspberry Pi B+, unsure on the OS installed, didnt mess with it and not an expert on Raspberry Pi.

But mostly `npm ci` cuts down significantly on the installation time, with `npm install` it went on for like 2 hours and still wasnt finished, with `npm ci` it took somewhere around the 20 min mark.

BTW, I also had to change the path to the npm executable in the zigbee2mqtt.service file:

```
ExecStart=/usr/local/bin/npm start
```

Not sure if thats a result of me installing by the default package manager (what is the reason for the special nodejs variant anyways?) or an mistake in the documentation.

Anyways, the main purpose of the issue you closed was to change to/or add at least the option for `npm ci` in the official documentation though it ended up not being clear about that in the text of the post (was tired I guess). I think considering the time saved at least in my case thats a significant optimization. So even if this doesn't help here I would like to either have you do it or look into adding it myself with a pull request.

Koenkk, 2020-02-12:
@9SMTM6 it looks like a good idea to completely drop `npm install` in favor of `npm ci` (as we will always have a `npm-shrinkwrap.json`)

Also in e.g.
- https://github.com/Koenkk/zigbee2mqtt/blob/master/docker/Dockerfile#L14
- https://github.com/Koenkk/zigbee2mqtt/blob/master/update.sh#L15
- All docs

Would you mind making a pr for this?

9SMTM6, 2020-02-12:
@Koenkk Yeah, can do. One thing though: Last I checked 'npm ci' doesn't support the --production flag and has no replacement for it, so the Docker-container will be bloated with developer dependencies if we switch in the dockerfile too.

Thats at least my understanding, the docs I saw for `npm ci` were not the clearest ones.

So assuming thats how it just is, either we keep the `install --production` in the dockerfile or we take the probably small hit in increased size and get slightly faster build speed (on regular PCs the performance impact of `ci` isn't nearly so large as on the Raspberry Pis with their SD-Card or USB memory.)

9SMTM6, 2020-02-12:
And since all node-.modules will be deleted regardless I dont do any fancy things in my upodate script, it looks essentially like:

```sh
#!/bin/sh -e

# To be executed only on raspberry pi (uses debians apt-get and defaults to the user pi etc)

INSTALLOCATION=/opt/zigbee2mqtt
BACKUPLOCATION=/tmp/z2m-data-backup

# Preparation: Stop the service if already running, backup current data, clear any old install location
sudo systemctl stop zigbee2mqtt || true
cp -R "$INSTALLOCATION/data" "$BACKUPLOCATION"
sudo rm -R "$INSTALLOCATION"

# Install dependencies for node and installation
sudo apt-get install -y nodejs git make g++ gcc mosquitto

# Get repo and own the directory (as its owned by SU by default)
sudo git clone https://github.com/Koenkk/zigbee2mqtt.git "$INSTALLOCATION"
sudo chown -R pi:pi "$INSTALLOCATION"

# Install node packages
cd "$INSTALLOCATION"
# basically npm install but faster (especially on the raspberry pi with its slow memory)
npm ci

# Add our config back
cp -r "$BACKUPLOCATION" "$INSTALLOCATION/data"

sudo systemctl start zigbee2mqtt

echo "Installation finished. Check status with systemctl status zigbee2mqtt.service"
```

Koenkk, 2020-02-12:
`--production` works (75 vs 924 packages)

```
zigbee-herdsman on  master [$] is 📦 v0.12.47 via ⬢ v10.17.0 
➜ npm ci --production
npm WARN prepare removing existing node_modules/ before installation

> @serialport/bindings@8.0.7 install /Users/Koenkk/Git/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> serialport@8.0.6 postinstall /Users/Koenkk/Git/zigbee-herdsman/node_modules/serialport
> node thank-you.js

Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

added 75 packages in 4.021s

zigbee-herdsman on  master [$] is 📦 v0.12.47 via ⬢ v10.17.0 took 5s 
```
```
➜ npm ci             
npm WARN prepare removing existing node_modules/ before installation

> fsevents@1.2.11 install /Users/Koenkk/Git/zigbee-herdsman/node_modules/fsevents
> node-gyp rebuild

  SOLINK_MODULE(target) Release/.node
  CXX(target) Release/obj.target/fse/fsevents.o
  SOLINK_MODULE(target) Release/fse.node

> @serialport/bindings@8.0.7 install /Users/Koenkk/Git/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> serialport@8.0.6 postinstall /Users/Koenkk/Git/zigbee-herdsman/node_modules/serialport
> node thank-you.js

Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

added 924 packages in 14.614s
```

Git commands in de update script were needed because npm install modifies it, so can also be removed.

Creo2005, 2020-02-12:
(Symology DS218+) npm ci:

npm ERR! code Z_DATA_ERROR
npm ERR! errno -3
npm ERR! zlib: invalid distance too far back

Creo2005, 2020-02-13:
**npm ci:**
npm ERR! code Z_DATA_ERROR
npm ERR! errno -3
**PS:**
root@SynoDS218:~# node --version
v12.14.0
root@SynoDS218:~# npm --version
6.13.4
root@SynoDS218:~#

lancelon, 2020-02-25:
`root@raspbx:/media/usb/zigbee2mqtt# npm ci
npm ERR! code EPERM
npm ERR! syscall symlink
npm ERR! path ../semver/bin/semver.js
npm ERR! dest /media/usb/zigbee2mqtt/node_modules/eslint/node_modules/.bin/semver
npm ERR! errno -1
npm ERR! Error: EPERM: operation not permitted, symlink '../semver/bin/semver.js' -> '/media/usb/zigbee2mqtt/node_modules/eslint/node_modules/.bin/semver'
npm ERR!  [OperationalError: EPERM: operation not permitted, symlink '../semver/bin/semver.js' -> '/media/usb/zigbee2mqtt/node_modules/eslint/node_modules/.bin/semver'] {
npm ERR!   cause: [Error: EPERM: operation not permitted, symlink '../semver/bin/semver.js' -> '/media/usb/zigbee2mqtt/node_modules/eslint/node_modules/.bin/semver'] {
npm ERR!     errno: -1,
npm ERR!     code: 'EPERM',
npm ERR!     syscall: 'symlink',
npm ERR!     path: '../semver/bin/semver.js',
npm ERR!     dest: '/media/usb/zigbee2mqtt/node_modules/eslint/node_modules/.bin/semver'
npm ERR!   },
npm ERR!   stack: "Error: EPERM: operation not permitted, symlink '../semver/bin/semver.js' -> '/media/usb/zigbee2mqtt/node_modules/eslint/node_modules/.bin/semver'",
npm ERR!   errno: -1,
npm ERR!   code: 'EPERM',
npm ERR!   syscall: 'symlink',
npm ERR!   path: '../semver/bin/semver.js',
npm ERR!   dest: '/media/usb/zigbee2mqtt/node_modules/eslint/node_modules/.bin/semver'
npm ERR! }
npm ERR! 
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR! 
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-25T20_45_30_143Z-debug.log`

Koenkk, 2020-02-26:
You are having some permission issues, fix the permission or try `sudo npm ci`

lancelon, 2020-02-28:
odd. raspbx is the root user. hmmm. OK. Maybe I'd better create a new non-root user and try to install under that user.

closed 2020-05-06 by stale[bot]