Koenkk/zigbee2mqtt #2921: Experimental feature: OTA updates

Koenkk, 2020-02-08:
This feature allows to OTA update your devices, documentation is available here:

- Currently it's only available for IKEA TRADFRI devices, don't ask if device X can also be supported but rather search for the firmware files (if available). If they are available it could be implemented.
- No guarantees, but I don't think it's possible to brick your devices with this. The device will only start upgrading once the image is valid. So don't worry if an update fails or is interrupted.

sjorge, 2020-02-08:
Are results only logged or is there a mqtt message also? Will update once I get home.

timstanley1985, 2020-02-08:
Successfully updated a tradfri bulb! 🎉🎉

Having to watch the logs on info level is a pain though.  Can you add an mqtt message that confirms if an update is available or not and a couple of messages that say updated started and then update successful/unsuccessful please.

atxbyea, 2020-02-08:
So I just upgraded to the latest dev branch to check on the tradfri ota feature, however I don't see anything in the logs when I try, running `Starting zigbee2mqtt version 1.10.0-dev (commit #5e152f2) `and trying to push the following to mosquitto `mosquitto_pub -h 192.168.1.40 -u user -P password -t "zigbee2mqtt/bridge/ota_update/check" -m "Stue Stuebord"`

Koenkk, 2020-02-08:
There is currently a problem with the docker builds, so they don't support OTA yet.

Koenkk, 2020-02-08:
Fixed, OTA is available on latest-dev docker now.

copystring, 2020-02-08:
I'm currently doing an OTA. Yeah. It is really slow but ok. But waiting for it to finish I think it does make sense to log a message with the current state in % of the update lets say every 60 seconds regardless of progress.
That way I know for sure if the OTA is still running or already finished. Helps a lot when there is a lot going on like in my network with 60+ devices sending a lot of messages.

PedroLamas, 2020-02-08:
> Fixed, OTA is available on latest-dev docker now.

Fantastic!! WIll update my local docker instance and report back in a bit once it updates the 2 smart plugs I have! 😄 

> Salus https://eu.salusconnect.io/demo/default/status/firmware?timestamp=0

If no one does the Salus integration before, I can take a look... I have a Salus SP600 that I'm not using and will be willing to "sacrifice" it!

way2many, 2020-02-08:
I'm trying to update a Ikea dimmer ICTC-G-1 but it fails:

```
info  2020-02-08 21:00:26: Checking if update available for 'Dimmer: Salon'
info  2020-02-08 21:00:27: Update available for 'Dimmer: Salon'
info  2020-02-08 21:00:27: Starting update of 'Dimmer: Salon'
info  2020-02-08 21:00:29: Update of 'Dimmer: Salon' at 0%
error 2020-02-08 21:00:38: Failed to call 'OTAUpdate' 'onMQTTMessage' (Error: Update failed with code '149'
    at fulfilled (/app/node_modules/zigbee-herdsman-converters/ota/common.js:126:24))
```

Device is awake. What is error code 149?

PedroLamas, 2020-02-08:
I'm having the same issue as @way2many while trying to update my IKEA Tradfri E1603/E1702 smart plugs

Koenkk, 2020-02-08:
[image]

`149` = `0x95`, not sure why it happens try removing power from the device for a minute and try again.

sjorge, 2020-02-08:
I tried both my currently paired trådfri bulbs and repeated, some have no update (but have old firmware AFAIK) and some throw

```
zigbee2mqtt:error 2020-02-08 21:07:52: Failed to call 'OTAUpdate' 'onMQTTMessage' (Error: Timeout - 47816 - 1 - null - 25 - 1 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:425:11)
    at tryOnTimeout (timers.js:289:5)
    at listOnTimeout (timers.js:252:5)
    at Timer.processTimers (timers.js:212:10))
```

That error I get for my repeater and for my remotes.

PedroLamas, 2020-02-08:
I've corrected my comment above as my error actually matches what @sjorge is getting (timeout).

I'm using pihole and a bunch of other stuff, I wonder if this is related... here's my docker configuration just in case: https://github.com/PedroLamas/nanopineo2-docker-config/blob/master/docker-compose.yml

sjorge, 2020-02-08:
@PedroLamas I'm not sure, I can download the json file from the url in ota/tradfri.js just fine.
And for my bulbs it reports no update.

Koenkk, 2020-02-08:
It seems it doesn't work well for the battery powered devices yet, only for bulbs ATM.

sjorge, 2020-02-08:
> It seems it doesn't work well for the battery powered devices yet, only for bulbs ATM.

Repeater is mains power though, and I am 100% sure that one should have an update.

PedroLamas, 2020-02-08:
Like @sjorge said, same would apply to Tradfri smart plugs!

Koenkk, 2020-02-08:
Could indeed also reproduce it for my plug, will investigate

sjorge, 2020-02-08:
Did some more investigating, all my bulbs (that I have paired) are already up to date. And other things (remotes and repeater) are failing.

Both repeat and remotes do have updates, I just verified in the json file.

kiall, 2020-02-08:
Looking forward to seeing this work - I know loads of my stuff is out of date! I tried to update an IKEA bulb, and got this.. happy to help debug tomorrow if there's new code or some pointers. 

`zigbee2mqtt    | zigbee2mqtt:info  2020-02-08 21:27:36: Checking if update available for 'master_bedroom_ceiling_light'
zigbee2mqtt    | zigbee2mqtt:info  2020-02-08 21:27:36: Update available for 'master_bedroom_ceiling_light'
zigbee2mqtt    | zigbee2mqtt:info  2020-02-08 21:27:36: Starting update of 'master_bedroom_ceiling_light'
zigbee2mqtt    | zigbee2mqtt:info  2020-02-08 21:27:37: Update of 'master_bedroom_ceiling_light' at 0%
zigbee2mqtt    | zigbee2mqtt:info  2020-02-08 21:27:41: MQTT publish: topic 'zigbee2mqtt/living_room_motion_sensor', payload '{"occupancy":true,"tamper":true,"battery_low":false,"linkquality":33,"last_seen":1581197261979,"temperature":18.9,"device":{"friendlyName":"living_room_motion_sensor","model":"AC01353010G","ieeeAddr":"0x000d6f0011087982","networkAddress":57565,"type":"EndDevice","manufacturerID":4174,"manufacturerName":"CentraLite","powerSource":"Battery","stackVersion":2,"zclVersion":1,"hardwareVersion":1,"dateCode":"����������������"}}'
zigbee2mqtt    | (node:16) UnhandledPromiseRejectionWarning: Error: CommandResponse 0x000b57fffef588ae/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4476,"imageType":8706,"fileVersion":587232817,"fileOffset":0,"dataSize":63,"data":{"type":"Buffer","data":[30,241,238,11,0,1,56,0,0,0,124,17,2,34,49,118,0,35,2,0,69,66,76,32,116,114,97,100,102,114,105,95,108,105,103,104,116,95,49,48,48,48,109,108,0,0,0,0,0,0,0,0,62,47,3,0,0,0,0,47,3,0,0]}}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":12554,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":38,"options":0,"radius":30,"len":80,"data":{"type":"Buffer","data":[25,26,5,0,124,17,2,34,49,118,0,35,0,0,0,0,63,30,241,238,11,0,1,56,0,0,0,124,17,2,34,49,118,0,35,2,0,69,66,76,32,116,114,97,100,102,114,105,95,108,105,103,104,116,95,49,48,48,48,109,108,0,0,0,0,0,0,0,0,62,47,3,0,0,0,0,47,3,0,0]}}' failed with status '16' (expected '0'))
zigbee2mqtt    |     at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:372:23)
zigbee2mqtt    |     at Generator.throw (<anonymous>)
zigbee2mqtt    |     at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
zigbee2mqtt    | (node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
zigbee2mqtt    | (node:16) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
zigbee2mqtt    | (node:16) UnhandledPromiseRejectionWarning: Error: CommandResponse 0x000b57fffef588ae/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4476,"imageType":8706,"fileVersion":587232817,"fileOffset":0,"dataSize":63,"data":{"type":"Buffer","data":[30,241,238,11,0,1,56,0,0,0,124,17,2,34,49,118,0,35,2,0,69,66,76,32,116,114,97,100,102,114,105,95,108,105,103,104,116,95,49,48,48,48,109,108,0,0,0,0,0,0,0,0,62,47,3,0,0,0,0,47,3,0,0]}}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":12554,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":40,"options":0,"radius":30,"len":80,"data":{"type":"Buffer","data":[25,27,5,0,124,17,2,34,49,118,0,35,0,0,0,0,63,30,241,238,11,0,1,56,0,0,0,124,17,2,34,49,118,0,35,2,0,69,66,76,32,116,114,97,100,102,114,105,95,108,105,103,104,116,95,49,48,48,48,109,108,0,0,0,0,0,0,0,0,62,47,3,0,0,0,0,47,3,0,0]}}' failed with status '16' (expected '0'))
zigbee2mqtt    |     at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:372:23)
zigbee2mqtt    |     at Generator.throw (<anonymous>)
zigbee2mqtt    |     at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
zigbee2mqtt    | (node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
zigbee2mqtt    | (node:16) UnhandledPromiseRejectionWarning: Error: CommandResponse 0x000b57fffef588ae/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4476,"imageType":8706,"fileVersion":587232817,"fileOffset":0,"dataSize":63,"data":{"type":"Buffer","data":[30,241,238,11,0,1,56,0,0,0,124,17,2,34,49,118,0,35,2,0,69,66,76,32,116,114,97,100,102,114,105,95,108,105,103,104,116,95,49,48,48,48,109,108,0,0,0,0,0,0,0,0,62,47,3,0,0,0,0,47,3,0,0]}}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":12554,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":42,"options":0,"radius":30,"len":80,"data":{"type":"Buffer","data":[25,28,5,0,124,17,2,34,49,118,0,35,0,0,0,0,63,30,241,238,11,0,1,56,0,0,0,124,17,2,34,49,118,0,35,2,0,69,66,76,32,116,114,97,100,102,114,105,95,108,105,103,104,116,95,49,48,48,48,109,108,0,0,0,0,0,0,0,0,62,47,3,0,0,0,0,47,3,0,0]} [... cut ...]
zigbee2mqtt    | zigbee2mqtt:info  2020-02-08 21:27:58: MQTT publish: topic 'zigbee2mqtt/signal_repeater_1', payload '{"linkquality":33,"last_seen":1581197278974,"device":{"friendlyName":"signal_repeater_1","model":"E1746","ieeeAddr":"0xd0cf5efffed82436","networkAddress":55232,"type":"Router","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","powerSource":"Mains (single phase)","applicationVersion":32,"stackVersion":98,"zclVersion":3,"hardwareVersion":1,"dateCode":"20180828","softwareBuildID":"2.0.019"}}'`

Koenkk, 2020-02-08:
So for the battery powered devices, you need to wake it up right before sending the MQTT command, updating a TRADFRI 5 button remote control now, fingers crossed.

sjorge, 2020-02-08:
I had mine awake (On/Off) as I was turning on and of the lamp before doing the request.
I can't pair my other bulbs again for some reason, I have the I can't pair anything bug again :(

Koenkk, 2020-02-08:
I've pushed some small improvements to the latest dev branch

@sjorge my on/off remote also refused to update, I then replaced the battery with a new one, executed update request and it worked!

sjorge, 2020-02-08:
Mine are at 74% and 93% respectively... I wonder if it needs 100%?

Koenkk, 2020-02-08:
@sjorge great that it started, you only have to press it once (right before), once you see `Update of 'bedroom/night_light/remote' at 0%` the device stays awake by itself. Progress will be logged at the following percentages: `[1, 5, 10, 20, 30, 40, 50, 60, 70, 80, 90, 95, 100]`

PedroLamas, 2020-02-08:
Here's what I get with the latest dev version from my Tradfri smart plug:

```
zigbee2mqtt:info  2020-02-08 23:14:37: Checking if update available for 'SmartPlug1'
zigbee2mqtt:error 2020-02-08 23:14:47: Failed to call 'OTAUpdate' 'onMQTTMessage' (Error: Device didn't respond to OTA request
    at getDeviceInfo (/app/node_modules/zigbee-herdsman-converters/ota/tradfri.js:18:15))
```

sjorge, 2020-02-08:
@Koenkk great... looks like it is at 10% now, the repeater still gets the timeout though.

Koenkk, 2020-02-08:
will look into control outlet/repeater later, will also update to code to provide updates every 30 seconds instead of the fixed levels (https://github.com/Koenkk/zigbee2mqtt/issues/2921#issuecomment-583785567)

sjorge, 2020-02-08:
Seems the rate is now ~10min for 5%, not to bad. Is there a limit on how many of devices we can update concurrently?

Koenkk, 2020-02-08:
@sjorge theoretically not, but note that there will be a lot of traffic on the network.

sjorge, 2020-02-08:
Cool, I was already looking at a more generic check and update that would do all eligable devices... but if I or someone else every does that, there probably needs to be a 2 or so concurrent limit.

Dennis14e, 2020-02-08:
I have tried to update my E1743 remote control. I get similar error messages as @kiall.
Log: https://pastebin.com/B6EWmUA0

dfsx1, 2020-02-08:
Great feature, thank you!

I already updated my Hue bulbs with my phone over bluetooth earlier this week.
The recently published Philips Hue exploit attended me how important the lastest firmware is: 
https://blog.checkpoint.com/2020/02/05/the-dark-side-of-smart-lighting-check-point-research-shows-how-business-and-home-networks-can-be-hacked-from-a-lightbulb/

As I don't own a Ikea hub, this OTA feature comes very handy.
I'm going to update all my Ikea devices tonight ;)

Koenkk, 2020-02-08:
@kiall @Dennis14e what coordinator/firmware are you using?

Dennis14e, 2020-02-08:
@Koenkk 
CC2531
Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

sjorge, 2020-02-08:
@Koenkk the remote failed after 40%

```
zigbee2mqtt:error 2020-02-08 23:38:46: Failed to call 'OTAUpdate' 'onMQTTMessage' (Error: Update failed with code '149'
```

PedroLamas, 2020-02-08:
@sjorge 149 = 0x95 ABORT, from the table above: https://github.com/Koenkk/zigbee2mqtt/issues/2921#issuecomment-583776981

sjorge, 2020-02-09:
2nd remote also failed:
```
zigbee2mqtt:info  2020-02-08 23:54:16: Update of 'bedroom/desk_lamp/remote' at 1%
zigbee2mqtt:info  2020-02-09 00:00:11: Update of 'bedroom/desk_lamp/remote' at 5%
zigbee2mqtt:info  2020-02-09 00:07:36: Update of 'bedroom/desk_lamp/remote' at 10%
zigbee2mqtt:error 2020-02-09 00:10:49: Failed to call 'OTAUpdate' 'onMQTTMessage' (Error: Update failed with code '149'
```

It was no longer responding to clicks either, removed the battery and it works again, but like the first remote it refuses to take another update. Battery was fresh, I tried with both IKEA Platbilja and a vara cell.

PedroLamas, 2020-02-09:
So I have some interesting news: I unplugged one of my Tradfri smart plugs counted to 10 and plugged it in again, ran update check and got a positive!

I'm now running the update process, as of writing it is at 30%.

sjorge, 2020-02-09:
Both remotes are now dead, I wanted to turn of the lights before going to bed and they no longer seem to respond. I swapped batteries with no luck. Will try to reset the near the coordinator tomorrow.

dfsx1, 2020-02-09:
I can confirm that a powercycle (manual unplugging and back in) is required for the E1603/E1702 - IKEA TRADFRI control outlet (Router).

Before powercycle:

zigbee2mqtt:info  2020-02-09 01:24:13: Checking if update available for '0x000d6ffffecdd459'
zigbee2mqtt:error 2020-02-09 01:24:23: Failed to call 'OTAUpdate' 'onMQTTMessage' (Error: Device didn't respond to OTA request
    at getDeviceInfo (/app/node_modules/zigbee-herdsman-converters/ota/tradfri.js:18:15))

After powercycle:

zigbee2mqtt:info  2020-02-09 01:25:07: Checking if update available for '0x000d6ffffecdd459'
zigbee2mqtt:info  2020-02-09 01:25:09: Update available for '0x000d6ffffecdd459'

atxbyea, 2020-02-09:
Even with my bulb I had to power it off for 10 seconds before doing a upgrade, before that it would just fail (see log)
This is with the Ikea Tradfri 1000lm CT bulb and CC1352
https://hastebin.com/emalaxecen.sql

Update took 8ish minutes for one bulb
zigbee2mqtt:info  2020-02-09 08:05:47: Update of 'Stue Stuebord' at 0%
zigbee2mqtt:info  2020-02-09 08:14:02: Finished update of 'Stue Stuebord', from '{"softwareBuildID":"2.0.022","dateCode":"20190308"}' to '{"softwareBuildID":"2.0.023","dateCode":"20190407"}'

It also turns off for 10 seconds going from 95% to 100% fyi

Koenkk, 2020-02-09:
The 10 seconds turn of is the bulb rebooting after the OTA update :smile:

kiall, 2020-02-09:
> @kiall @Dennis14e what coordinator/firmware are you using?

Currently on Stack3x0 / 20191106 with a CC26X2R1.

Despite lots of `UnhandledPromiseRejectionWarning` exceptions, things do seem to be working: 

> 2020-02-09 08:41:04: Update of 'kitchen_ceiling_bank_4_light_3' at 50%

atxbyea, 2020-02-09:
So far I have tried on 6 Tradfri bulbs, only one has succeeded so far, most of them fail with 149 after 10-90% and even after powering them off for 20 seconds.

kiall, 2020-02-09:
I've now successfully updated 2 bulbs, and failed with 1 - I'll stop here, and wait for any more updates from Koenkk so I still have something to test with ;)

1) I needed to power off for 10 sec before an update would be accepted. I powered them all off at once, and that seems to have do it - i.e. I didn't need to power off immediately prior to updating.

2) I noticed the state in HASS is incorrect after the update. The bulb was off, at 95% during the reboot, it powered on - but the state was never updated to reflect that. Maybe we need to explicitly query the device state after an update has completed?

3) The 3rd bulb failed with error 149 after 80%.

4) Throughout all of this, lots and lots of `UnhandledPromiseRejectionWarning`'s are logged.

GrumpyMeow, 2020-02-09:
I tried to update my Ikea Tradfiri Leptiter spots. In the log I see that it logs information of a firmware-url, but still says “No update available”. Are the Leptiter spots a special case?

sjorge, 2020-02-09:
Good news, after leaving the batteries out both on/off remotes were able to pair again. They still refuse to any attempt to start an update now though. Even with another set of fresh batteries.

copystring, 2020-02-09:
> Good news, after leaving the batteries out both on/off remotes were able to pair again. They still refuse to any attempt to start an update now though. Even with another set of fresh batteries.

I had this too. Pair the force remove the device and pair it again. Then it works. It seems the device ist stuck in OTA mode.

sjorge, 2020-02-09:
On a more positive note, the repeater update after unplugging it and replogging it:
```
{"softwareBuildID":"2.0.019","dateCode":"20180828"}' to '{"softwareBuildID":"2.2.005","dateCode":"20190318"}
```

copystring, 2020-02-09:
@sjorge what is the link quality of the device? Maybe it is related to that? I also think that traffic of other devices may interfere during an OTA?

sjorge, 2020-02-09:
> @sjorge what is the link quality of the device? Maybe it is related to that? I also think that traffic of other devices may interfere during an OTA?

It was around 60, it was right next to the coordinator.

After unplugging both the bulb and coordinator, the next update worked fine:
```
{"softwareBuildID":"1.3.002","dateCode":"20170315"}' to '{"softwareBuildID":"1.3.009","dateCode":"20180410"}
```

copystring, 2020-02-09:
Which coordinator do you have? I had a link quality of 54 right next to the coordinator.
Turning the coordinator 90° and that changed it to 194.

[image]

ophilips, 2020-02-09:
I also just updated all my Tradfri devices. For 2 bulbs (LED1624G9) everything was fine without errors:
```
zigbee2mqtt:info  2020-02-09 10:33:59: Finished update of '0x000b57fffea7eab1', from '{"softwareBuildID":"1.3.002","dateCode":"20170315"}' to '{"softwareBuildID":"1.3.009","dateCode":"20180410"}'

zigbee2mqtt:info  2020-02-09 11:04:39: Finished update of '0x000b57fffeae4a78', from '{"softwareBuildID":"1.3.002","dateCode":"20170315"}' to '{"softwareBuildID":"1.3.009","dateCode":"20180410“}‘
```

For one plug from the very first batch (E1603/E1702) I did get a strange status update with no previous firmware version found:

```
zigbee2mqtt:info  2020-02-09 11:24:49: Finished update of '0x000d6ffffef32609', from '{}' to '{"softwareBuildID":"2.0.022","dateCode":"20190308"}'
```

And for the last device, also a plug from the very first batch, I got a warning but it was already beyond 95% of the update. I had to reset the plug some minutes after the warning and it seems to work now with the newest version:
```
zigbee2mqtt:error 2020-02-09 10:44:36: Failed to call 'OTAUpdate' 'onMQTTMessage' (Error: Read 0xd0cf5efffe43baaa/1 genBasic(["dateCode","swBuildId"], {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 59470 - 1 - 69 - 0 - 1 after 10000ms)

zigbee2mqtt:info  2020-02-09 11:40:38: No update available for '0xd0cf5efffe43baaa'
```

sjorge, 2020-02-09:
> Which coordinator do you have? I had a link quality of 54 right next to the coordinator.
> Turning the coordinator 90° and that changed it to 194.

It's a simple CC2531, so far only the on/off remotes are left.

orrpan, 2020-02-09:
```log

```

<details><summary>Full log</summary>
<p>

#### yes, even hidden code blocks!

```log
Feb 09 11:45:03 hem bash[22563]: zigbee2mqtt:info  2020-02-09 11:45:03: Checking if update available for '0xd0cf5efffec0156d'
Feb 09 11:45:06 hem bash[22563]: zigbee2mqtt:info  2020-02-09 11:45:06: Update available for '0xd0cf5efffec0156d'
Feb 09 11:45:06 hem bash[22563]: zigbee2mqtt:info  2020-02-09 11:45:06: Starting update of '0xd0cf5efffec0156d'
Feb 09 11:45:07 hem bash[22563]: zigbee2mqtt:info  2020-02-09 11:45:07: Update of '0xd0cf5efffec0156d' at 0%
Feb 09 11:45:21 hem bash[22563]: zigbee2mqtt:info  2020-02-09 11:45:21: Update of '0xd0cf5efffec0156d' at 1%
Feb 09 11:45:40 hem bash[22563]: zigbee2mqtt:info  2020-02-09 11:45:40: Update of '0xd0cf5efffec0156d' at 5%
Feb 09 11:46:03 hem bash[22563]: zigbee2mqtt:info  2020-02-09 11:46:03: Update of '0xd0cf5efffec0156d' at 10%
Feb 09 11:46:51 hem bash[22563]: zigbee2mqtt:info  2020-02-09 11:46:51: Update of '0xd0cf5efffec0156d' at 20%
Feb 09 11:46:58 hem bash[22563]: (node:22586) UnhandledPromiseRejectionWarning: Error: CommandResponse 0xd0cf5efffec0156d/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817,"fileOffset":44982,"dataSize":63,"data":{"type":"Buffer","data":[232,255,0,44,24,191,0,40,1,208,32,70,50,189,0,32,50,189,248,181,7,70,192,178,255,247,219,255,6,70,56,4,0,14,255,247,214,255,5,70,56,2,0,14,255,247,209,255,4,70,56,14,255,247,205,255,0,46,24,191,0,45,5]}}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":40662,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":70,"options":0,"radius":30,"len":80,"data":{"type":"Buffer","data":[25,76,5,0,124,17,1,34,49,118,0,35,182,175,0,0,63,232,255,0,44,24,191,0,40,1,208,32,70,50,189,0,32,50,189,248,181,7,70,192,178,255,247,219,255,6,70,56,4,0,14,255,247,214,255,5,70,56,2,0,14,255,247,209,255,4,70,56,14,255,247,205,255,0,46,24,191,0,45,5]}}' failed with status '16' (expected '0'))
Feb 09 11:46:58 hem bash[22563]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:372:23)
Feb 09 11:46:58 hem bash[22563]:     at Generator.throw (<anonymous>)
Feb 09 11:46:58 hem bash[22563]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
Feb 09 11:46:58 hem bash[22563]: (node:22586) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 21)
Feb 09 11:47:43 hem bash[22563]: zigbee2mqtt:info  2020-02-09 11:47:43: Update of '0xd0cf5efffec0156d' at 30%
Feb 09 11:48:31 hem bash[22563]: zigbee2mqtt:info  2020-02-09 11:48:31: Update of '0xd0cf5efffec0156d' at 40%
Feb 09 11:49:16 hem bash[22563]: (node:22586) UnhandledPromiseRejectionWarning: Error: CommandResponse 0xd0cf5efffec0156d/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817,"fileOffset":103131,"dataSize":63,"data":{"type":"Buffer","data":[231,80,234,65,7,48,70,97,70,28,191,65,234,5,81,240,189,82,234,67,12,8,191,193,67,240,189,0,32,79,244,0,22,22,235,65,15,152,191,22,235,67,15,8,191,22,235,65,15,36,191,193,67,5,224,22,235,67,15,12,191,97,70]}}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":40662,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":137,"options":0,"radius":30,"len":80,"data":{"type":"Buffer","data":[25,104,5,0,124,17,1,34,49,118,0,35,219,146,1,0,63,231,80,234,65,7,48,70,97,70,28,191,65,234,5,81,240,189,82,234,67,12,8,191,193,67,240,189,0,32,79,244,0,22,22,235,65,15,152,191,22,235,67,15,8,191,22,235,65,15,36,191,193,67,5,224,22,235,67,15,12,191,97,70]}}' failed with status '16' (expected '0'))
Feb 09 11:49:16 hem bash[22563]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:372:23)
Feb 09 11:49:16 hem bash[22563]:     at Generator.throw (<anonymous>)
Feb 09 11:49:16 hem bash[22563]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
Feb 09 11:49:16 hem bash[22563]: (node:22586) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 22)
Feb 09 11:49:21 hem bash[22563]: (node:22586) UnhandledPromiseRejectionWarning: Error: CommandResponse 0xd0cf5efffec0156d/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817,"fileOffset":103131,"dataSize":63,"data":{"type":"Buffer","data":[231,80,234,65,7,48,70,97,70,28,191,65,234,5,81,240,189,82,234,67,12,8,191,193,67,240,189,0,32,79,244,0,22,22,235,65,15,152,191,22,235,67,15,8,191,22,235,65,15,36,191,193,67,5,224,22,235,67,15,12,191,97,70]}}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":40662,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":139,"options":0,"radius":30,"len":80,"data":{"type":"Buffer","data":[25,105,5,0,124,17,1,34,49,118,0,35,2 [... cut ...]
Feb 09 11:58:29 hem bash[22563]: zigbee2mqtt:debug 2020-02-09 11:58:29: Got device info '{"fieldControl":1,"manufacturerCode":4476,"imageType":8705,"fileVersion":587232817}'
Feb 09 11:58:30 hem bash[22563]: zigbee2mqtt:debug 2020-02-09 11:58:30: Got image '{"fileVersion":587232817,"fileSize":208126,"url":"http://fw.ota.homesmart.ikea.net/Tradfri_OTA_release_signed_2020_01_17_124857/bin/10035514-2.1-TRADFRI-bulb-ws-2.3.007.ota.ota.signed"}'
Feb 09 11:58:30 hem bash[22563]: zigbee2mqtt:info  2020-02-09 11:58:30: No update available for '0x90fd9ffffeeaa0ba'
```

</p>
</details>

nldroid, 2020-02-09:
No luck for me updating a bulb:
```
Feb 09 11:17:37 nuc npm[25072]:     at Generator.throw (<anonymous>)
Feb 09 11:17:37 nuc npm[25072]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
Feb 09 11:17:37 nuc npm[25072]: (node:25103) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 10)
Feb 09 11:17:42 nuc npm[25072]: (node:25103) UnhandledPromiseRejectionWarning: Error: CommandResponse 0xd0cf5efffe2268bb/1 genOta.imageBlockResponse({"status":0,"manufacturerCode":4476,"imageType":10241,"fileVersion":318805362,"fileOffset":6741,"dataSize":63,"data":{"type":"Buffer","data":[244,32,105,93,77,46,29,93,72,1,104,113,101,1,106,177,101,91,73,10,104,242,101,73,104,49,102,65,106,113,102,129,106,177,102,193,106,241,102,1,107,49,103,128,107,112,103,150,248,52,0,64,7,76,191,247,107,183,107,184,70,2,240]}}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: SREQ '--> AF - dataRequest - {"dstaddr":52517,"destendpoint":1,"srcendpoint":1,"clusterid":25,"transid":4,"options":0,"radius":30,"len":80,"data":{"type":"Buffer","data":[25,125,5,0,124,17,1,40,114,149,0,19,85,26,0,0,63,244,32,105,93,77,46,29,93,72,1,104,113,101,1,106,177,101,91,73,10,104,242,101,73,104,49,102,65,106,113,102,129,106,177,102,193,106,241,102,1,107,49,103,128,107,112,103,150,248,52,0,64,7,76,191,247,107,183,107,184,70,2,240]}}' failed with status '16' (expected '0'))
Feb 09 11:17:42 nuc npm[25072]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:372:23)
Feb 09 11:17:42 nuc npm[25072]:     at Generator.throw (<anonymous>)
Feb 09 11:17:42 nuc npm[25072]:     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
Feb 09 11:17:42 nuc npm[25072]: (node:25103) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 11)
Feb 09 11:17:47 nuc npm[25072]: zigbee2mqtt:error 2020-02-09 11:17:47: Failed to call 'OTAUpdate' 'onMQTTMessage' (Error: Update failed with code '149'
Feb 09 11:17:47 nuc npm[25072]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/ota/common.js:126:24))
```

lolorc, 2020-02-09:
it would be very nice if the firmware could be loaded from a local dir, and the firmware list maintained by the converter lib. I don't plan on giving my z2m box any external network access.

nldroid, 2020-02-09:
Moved the bulb to 1 meter from coordinator and it succeeded. What a great new feature!

```
Feb 09 11:57:14 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 11:57:14: Checking if update available for 'slaapkamer_light'
Feb 09 11:57:14 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 11:57:14: Update available for 'slaapkamer_light'
Feb 09 11:57:14 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 11:57:14: Starting update of 'slaapkamer_light'
Feb 09 11:57:15 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 11:57:15: Update of 'slaapkamer_light' at 0%
Feb 09 11:57:27 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 11:57:27: Update of 'slaapkamer_light' at 1%
Feb 09 11:57:40 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 11:57:40: Update of 'slaapkamer_light' at 5%
Feb 09 11:57:56 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 11:57:56: Update of 'slaapkamer_light' at 10%
Feb 09 11:58:33 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 11:58:33: Update of 'slaapkamer_light' at 20%
Feb 09 11:59:11 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 11:59:11: Update of 'slaapkamer_light' at 30%
Feb 09 11:59:49 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 11:59:49: Update of 'slaapkamer_light' at 40%
Feb 09 12:00:27 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 12:00:27: Update of 'slaapkamer_light' at 50%
Feb 09 12:01:05 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 12:01:05: Update of 'slaapkamer_light' at 60%
Feb 09 12:01:43 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 12:01:43: Update of 'slaapkamer_light' at 70%
Feb 09 12:02:21 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 12:02:21: Update of 'slaapkamer_light' at 80%
Feb 09 12:02:59 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 12:02:59: Update of 'slaapkamer_light' at 90%
Feb 09 12:03:18 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 12:03:18: Update of 'slaapkamer_light' at 95%
Feb 09 12:04:09 nuc npm[25072]: zigbee2mqtt:info  2020-02-09 12:04:09: Finished update of 'slaapkamer_light', from '{"softwareBuildID":"1.3.002","dateCode":"20170315"}' to '{"softwareBuildID":"1.3.009","dateCode":"20180410"}'
```

Koenkk, 2020-02-09:
Thanks for all the feedback guys, based on the input above I'm currently improving this feature.

I kindly ask NOT the update all your devices now, otherwise we wont have devices to test anymore.

Kryzek, 2020-02-09:
I was about to check wheter I have something to update...
But since I also got this:
```
zigbee2mqtt:info  2020-02-09 14:15:20: Checking if update available for '0x000b3cfffef6e0d1'
zigbee2mqtt:debug 2020-02-09 14:15:20: Check if update available for '0x000b3cfffef6e0d1' (TRADFRI control outlet)
zigbee2mqtt:debug 2020-02-09 14:15:20: Using endpoint '1'
zigbee2mqtt:error 2020-02-09 14:15:30: Failed to call 'OTAUpdate' 'onMQTTMessage' (Error: Device didn't respond to OTA request
    at getDeviceInfo (/opt/zigbee2mqtt110-dev/node_modules/zigbee-herdsman-converters/ota/tradfri.js:18:15))
```
I think I'll just wait a bit then for improvements.

[396 more replies]

closed 2020-12-11 by Koenkk