Koenkk/zigbee2mqtt #2923: 'No network route' (205) in some devices and without lines in network map

kitopopo, 2020-02-09:
Dear @Koenkk ,

Sometimes after make several resets to the raspberry some devices no appears drawn in the network map, the icon appear but without lines.
Now i have 2 devices without lines, in this moment are 2 smoke detectors. 

this is the network map file, the devices. 
devices that are drawn without lines appear only once in this file (smoke detectors):

```
...
"0x00158d00026xxxxx" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Switch alarma|0x00158d00026xxxxx (49158)|Xiaomi MiJia wireless switch (WXKG01LM)|2020-02-09T02:40:04+01:00}"];
  "0x00158d00026aac7c" -> "0x00158d00026xxxxx" [penwidth=1, weight=0, color="#994444", label="59"]
  "0x00158d00027xxxxx" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Sensor caldera|0x00158d00027xxxxx (52032)|Xiaomi Aqara water leak sensor (SJCGQ11LM)|2020-02-09T01:51:17+01:00}"];
  "0x00158d00027xxxxx" -> "0x00158d00030xxxxx" [penwidth=1, weight=0, color="#994444", label="90"]
  "0x00158d00041xxxxx" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Smoke detector armario|0x00158d00041xxxxx (55138)|Xiaomi MiJia Honeywell smoke detector (JTYJ-GD-01LM/BW)|2020-02-09T01:03:52+01:0$
  "0x00158d000411xxxxx" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Smoke detector sai|0x00158d00041xxxxx (49850)|Xiaomi MiJia Honeywell smoke detector (JTYJ-GD-01LM/BW)|2020-02-09T03:15:49+01:00}"];
  "0x00158d00027xxxxx" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Sensor1 salon|0x00158d00027xxxxx (62386)|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|2020-02-09T02:27:05+01:00}"];
  "0x00158d00027xxxxx" -> "0x00158d00026xxxxx" [penwidth=1, weight=0, color="#994444", label="90"]
  "0x00158d0002axxxxx" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Sensor vibracion|0x00158d0002xxxxx (43403)|Xiaomi Aqara vibration sensor (DJT11LM)|2020-02-09T01:35:44+01:00}"];
  "0x00158d0002axxxxx" -> "0x00158d00026xxxxx" [penwidth=1, weight=0, color="#994444", label="113"]
...

```

I have tried to send a self-test to the smoke sensor but I get the message "no network route 205".
I have also tried to send the following repair configure comand but a message appears in the log: this device not require this.

`sudo mosquitto_pub -h 127.0.0.1 -t zigbee2mqtt/bridge/configure -m '0x00158d00041xxxxx'
`
Is there any command or topic or some way to repair or force the routes without having to remove the devices and add them back to the network again?

some days ago there were 13 devices with this problem, i solve this erasing device by device and pairing all again but this requires a lot of work and time

More people have complained about this problem. 

Thanks in advanced. 

Best regards

# Bug Report
## What happened

Sometimes some devices are drawn on the map without routing lines

## What did you expect to happen

be able to repair device paths without having to unlink them and link them again, some days ago 

## How to reproduce it (minimal and precise)

Randomly, maybe making several resets to the raspberry or changing the sd card. 

## Debug Info
zigbee2mqtt version: version 1.10.0 Lastet dev brach
CC2531 firmware version: 20190608.zip

Koenkk, 2020-02-09:
Does it only happen for battery powered devices? If yes, this is can indeed happen because they are sleeping. Repairing might solve it for some time but once they sleep for too long they will disappear.

You dont have to worry about this, its just a cosmetic.

kitopopo, 2020-02-09:
Dear friend,

Yes, it only happens with sensors that have a battery. Today 2 more sensors have been added to the list, now there are 4 devices with no route on the zigbee map. This has worked well for months because I made a map of the network every day. and they have never disappeared, as I will not give importance if the devices cannot receive orders such as smoke sensors, I can not do a selftest. This is the error message I receive when i send a selftest:

feb 09 12:53:07 raspberrypi npm[3935]: zigbee2mqtt:info  2020-02-09 12:53:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'selftest' to 'Smoke detector sai' failed: 'Error: Write 0x00158d000411c0b5/1 ssIasZone({\"65521\":{\"value\":50397184,\"type\":35}}, {\"timeout\":35000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":4447,\"disableDefaultResponse\":true}) failed (Error: Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"Smoke detector sai"}}'

if the devices were sleeping when I sent a selftest they should wake up, and for example the temperature sensors are not sleeping because they keep recording the temperature every 5 minutes, I think the devices lose the network path. What do you mean by repair? unlink and relink to solve the problem?

kitopopo, 2020-02-09:
Dear @Koenkk ,

I have found another problem related to the same instability after several resets.
 
I have had to do many reboots to try to adjust the resolution of the raspberry to a monitor, after these reboots, apart from having many devices that do not appear the lines on the zigbee map, the problem of no lqi appears on two xiaomi smart routers plug, it can simply be corrected by resetting the smart plug, unplugging and plugging it in after a few seconds. This solves the problem of 13 devices without drawing. Now 11 devices are drawn correctly

Also, Now the zigbe2mqtt service does not start at the first, makes 3 attempts to boot and the third attempt starts, this causes several problems for example that domoticz starts before the zigbe2mqtt service and no device works anymore. i tried update again npn (same process for update the new brand ) but the service no startup at the first attempts . Needs 3 attempts
The message of the zigbe2mqtt log is as follows:

```
-- Logs begin at Sun 2020-02-09 18:56:41 CET. --
feb 09 18:56:45 raspberrypi systemd[1]: Started zigbee2mqtt.
feb 09 18:56:50 raspberrypi npm[446]: > zigbee2mqtt@1.10.0-dev start /opt/zigbee2mqtt
feb 09 18:56:50 raspberrypi npm[446]: > node index.js
feb 09 18:56:53 raspberrypi npm[446]: zigbee2mqtt:info  2020-02-09 18:56:53: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-02-09.18-56-53' filename: log.txt
feb 09 18:56:54 raspberrypi npm[446]: zigbee2mqtt:info  2020-02-09 18:56:54: Starting zigbee2mqtt version 1.10.0-dev (commit #13b996e)
feb 09 18:56:54 raspberrypi npm[446]: zigbee2mqtt:info  2020-02-09 18:56:54: Starting zigbee-herdsman...
feb 09 18:57:56 raspberrypi npm[446]: zigbee2mqtt:error 2020-02-09 18:57:56: Error while starting zigbee-herdsman
feb 09 18:57:56 raspberrypi npm[446]: zigbee2mqtt:error 2020-02-09 18:57:56: Failed to start zigbee
feb 09 18:57:56 raspberrypi npm[446]: zigbee2mqtt:error 2020-02-09 18:57:56: Exiting...
feb 09 18:57:56 raspberrypi npm[446]: zigbee2mqtt:error 2020-02-09 18:57:56: Error: AREQ - SYS - resetInd after 30000ms
feb 09 18:57:56 raspberrypi npm[446]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
feb 09 18:57:56 raspberrypi npm[446]:     at ontimeout (timers.js:436:11)
feb 09 18:57:56 raspberrypi npm[446]:     at tryOnTimeout (timers.js:300:5)
feb 09 18:57:56 raspberrypi npm[446]:     at listOnTimeout (timers.js:263:5)
feb 09 18:57:56 raspberrypi npm[446]:     at Timer.processTimers (timers.js:223:10)
feb 09 18:57:56 raspberrypi npm[446]: npm ERR! code ELIFECYCLE
feb 09 18:57:56 raspberrypi npm[446]: npm ERR! errno 1
feb 09 18:57:56 raspberrypi npm[446]: npm ERR! zigbee2mqtt@1.10.0-dev start: `node index.js`
feb 09 18:57:56 raspberrypi npm[446]: npm ERR! Exit status 1
feb 09 18:57:56 raspberrypi npm[446]: npm ERR!
feb 09 18:57:56 raspberrypi npm[446]: npm ERR! Failed at the zigbee2mqtt@1.10.0-dev start script.
feb 09 18:57:56 raspberrypi npm[446]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
feb 09 18:57:56 raspberrypi npm[446]: npm ERR! A complete log of this run can be found in:
feb 09 18:57:56 raspberrypi npm[446]: npm ERR!     /home/pi/.npm/_logs/2020-02-09T17_57_56_740Z-debug.log
feb 09 18:57:56 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
feb 09 18:57:56 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
feb 09 18:57:56 raspberrypi systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
feb 09 18:57:56 raspberrypi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 1.
feb 09 18:57:56 raspberrypi systemd[1]: Stopped zigbee2mqtt.
feb 09 18:57:56 raspberrypi systemd[1]: Started zigbee2mqtt.
feb 09 18:57:57 raspberrypi npm[1234]: > zigbee2mqtt@1.10.0-dev start /opt/zigbee2mqtt
feb 09 18:57:57 raspberrypi npm[1234]: > node index.js
feb 09 18:57:58 raspberrypi npm[1234]: zigbee2mqtt:info  2020-02-09 18:57:58: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-02-09.18-57-58' filename: log.txt
feb 09 18:57:59 raspberrypi npm[1234]: zigbee2mqtt:info  2020-02-09 18:57:59: Starting zigbee2mqtt version 1.10.0-dev (commit #13b996e)
feb 09 18:57:59 raspberrypi npm[1234]: zigbee2mqtt:info  2020-02-09 18:57:59: Starting zigbee-herdsman...
feb 09 18:58:48 raspberrypi npm[1234]: zigbee2mqtt:error 2020-02-09 18:58:48: Error while starting zigbee-herdsman
feb 09 18:58:48 raspberrypi npm[1234]: zigbee2mqtt:error 2020-02-09 18:58:48: Failed to start zigbee
feb 09 18:58:48 raspberrypi npm[1234]: zigbee2mqtt:error 2020-02-09 18:58:48: Exiting...
feb 09 18:58:48 raspberrypi npm[1234]: zigbee2mqtt:error 2020-02-09 18:58:48: Error: AREQ - SYS - resetInd after 30000ms
feb 09 18:58:48 raspberrypi npm[1234]:     at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
feb 09 18:58:48 raspberrypi npm[1234]:     at ontimeout (timers.js:436:11)
feb 09 18:58:48 raspberrypi npm[1234]:     at tryOnTimeout (timers.js:300:5)
feb 09 18:58:48 raspberrypi npm[1234]:     at listOnTimeout (timers.js:263:5)
feb 09 18:58:48 raspberrypi npm[1234]:     at Timer.processTimers (timers.js:223:10)
feb 09 18:58:48 raspberrypi npm[1234]: npm ERR! code ELIFECYCLE
feb 09 18:58:48 raspberrypi npm[1234]: npm ERR! errno 1
[... cut ...]
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.19.0
19 verbose npm  v6.13.4
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.10.0-dev start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.10.0-dev start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

Any idea for solve the service restart to the first attempt? 
Thanks in advanced, best regards.

kitopopo, 2020-02-10:
I found the problem, after doing a lot of resets to the raspberry devices that are routers, in my case (smart zigbe socket) they stay locked and showing the error message lqi. Simply unplugging these plugs for 10 seconds the problem is solved, but there is a new router plug that does not unplug for 10 seconds, I think that for this reason the new devices that I added were not routed well. After unplugging the 3 smart plug reapeters, devices with lines are drawn on the zigbee network map.

closed 2020-02-10 by kitopopo