Koenkk/zigbee2mqtt #2924: IKEA E1743 and default_bind_group

sjorge, 2020-02-09:
# Bug Report
## What happened
After some issues with OTA I had to join and rebind both of my [E1743](https://www.zigbee2mqtt.io/devices/E1743.html) On/Off remotes.

As per the docs:
```markdown
# Binding

The E1743 can be bound to groups using binding. It can only be bound to 1 group at a time and cannot be bound to a device.

By default this remote is bound to the default bind group which you first have to unbind it from. This can be done by sending to zigbee2mqtt/bridge/unbind/[DEVICE_FRIENLDY_NAME]] payload default_bind_group.
```

`default_bind_group` is 901 according to the source, however I was unable to unbind that group, neither was I able to unbind the old group that was bound before having to re-join them (that last bit makes sense).

Originally i had to unbind them from group 99, so I created a new group with id 99... now the unbind works.

## What did you expect to happen
- Docs to mention the correct group to unbind (99)
or 
- Bind the group to `default_bind_group` by default (it should be according to devices.js, but maybe the bind fails because the hardware is bound to group 99 instead?)

## How to reproduce it (minimal and precise)
- remove an E1743 from the network
- force_remove to clean up all config
- enable join
- join E1743
- try and unbind `default_bind_group`, it fails
- try and unbind group 99, it succeeds

## Debug Info
zigbee2mqtt version: 1.10.0+dev
CC253X firmware version:

Koenkk, 2020-02-11:
Bind group 99 was previously used I think (1.6.0 or earlier)

sjorge, 2020-02-11:
Yeah, but I complete removed, reset and rejoined a remote.
And it did not have default_bind_group attached after the configure step.

I was/am using 1.10+dev

Koenkk, 2020-02-11:
Maybe you still have this group in your `configuration.yaml`?

sjorge, 2020-02-11:
Nope, I had to add one again with id 99 to unbind.
So the remote did indeed get reset given it had a different group bound before.

~ sjorge

BudBundi, 2020-02-12:
I have also rejoin a E1743 after OTA,
it was bind to group 0,

> zigbee2mqtt/bridge/unbind/[DEVICE_FRIENLDY_NAME]] payload default_bind_group

work after waking the remote, and bind to wanted group was okay
no problem for me

sjorge, 2020-02-12:
Weird because default_bind_group's ID 901 (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L17)

I did a remove again and the remote left the network, took out the battery, waited a few hours and put it back in and rejoined it. It was bound to group 99 again. So not 901 (default_bind_group) and not to the group I had it bound before removing it.

BudBundi, 2020-02-12:
I had the same behavior with a second device, also sending to group 0, after a second rebind sending to group 901, this device reacts to the default_bind_group just on group 901

ginkel, 2020-03-10:
Any hints how to resolve this? I have been able to re-bind a couple of E1743 remotes, but some fail to perform the required unbind. BTW, this has only become a problem after re-pairing with a 3.x.0 coordinator (the 1.2 HA firmware for CC2531 did not have this problem).

sjorge, 2020-03-10:
Try unbinding from both group 0 and 901... regardsless of the result, try binding your new group.

ginkel, 2020-03-10:
What do I need to put into the payload to unbind from a numeric group id? If I just use the group number I get:

```
Failed to call 'DeviceBind' 'onMQTTMessage' (AssertionError [ERR_ASSERTION]: Target is unknown
    at DeviceBind.onMQTTMessage (/app/lib/extension/deviceBind.js:35:9)
    at Controller.callExtensionMethod (/app/lib/controller.js:329:44)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5))
```

sjorge, 2020-03-10:
I think you just need to add the group with that ID to configuration.yaml

ginkel, 2020-03-10:
Hm, done, but all unbinds from 0 or 901 fail and binds to the new group also fail. Just noticed, that the device in question is a E1524 with the ZB 3 firmware...

Reset, re-pair, replaced the battery, nothing seemed to help... Until I restarted z2m...

Koenkk, 2020-03-10:
@ginkel did you wake-up the device right before sending the commands (by pressing a button on it)?

ginkel, 2020-03-10:
Of course ;-)

Koenkk, 2020-03-11:
I've updated my E1524 to the latest firmware (`2.3.014`). One thing I noticed that after this the original binding was lost, it bound it self to a random group number. I decided to repair the device and after this it was bound to the `default_bind_group` again. Now I followed https://www.zigbee2mqtt.io/devices/E1524_E1810.html#binding to successfully bind it to a group of choice.

sjorge, 2020-03-11:
Is there a way to restore the bindings after an update? (automatically i mean)

Koenkk, 2020-03-11:
@sjorge this is hard as the device is sleeping after this.

sjorge, 2020-03-11:
Ah that circles back to the delayed command queue for battery devices doesn't it.
I think someone opened a feature request for it, also in relation to the 'null' version after update of battery devices... were we could then deferrer the version update until we see the device send a frame... as we know it is awake imediatly after that.

Koenkk, 2020-03-11:
Yes, that indeed would be a solution

sjorge, 2020-03-25:
I'll close this, as a fresh out of package device seems to work as expected... just not an updated one.

But updating it before swapping the bindings works.

closed 2020-03-25 by sjorge

Nauttor, 2020-07-15:
Hi, 

I have experienced the same unreliable behave from the unbinding/binding procedure. I have 4 devices that does not work as expected when trying to unbind/bind them all of them on the last Firmware version. The cc2531 is loaded with the last 1.2 and the z2m version is the last from GitHub. 

E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
E1525/E1745 - IKEA TRADFRI motion sensor (EndDevice)
ZYCT-202 - Trust Remote control (EndDevice)

The first two initially appears to be in the group 901. No matter what I tried they didn't leave the group. Only after I create the group by hand, bind them both and then unbind they start working on the new group. However the process wasn't that easy was more a hit and miss, I try different orders and sometimes it works as expected and sometimes not. 

About the other two devices, the Motion sensor is on group 7787(randomly changed every time I remove and add it) and the Trust on 150, no matter the procedure I follow them stay in their group and I can't unbind them. BTW, yes I wake them every time. Funny thing, it always says successful unbinding / binding, but it continues in the same group. 

Unbinding 

Jul 15 14:29:43 atom npm[4866]: #033[34mzigbee2mqtt:debug#033[39m 2020-07-15 14:29:43: Received MQTT message on 'zigbee2mqtt/bridge/unbind/0x000d6ffffe2672bd' with data '7787'
Jul 15 14:29:43 atom npm[4866]: #033[34mzigbee2mqtt:debug#033[39m 2020-07-15 14:29:43: unbinding cluster 'genOnOff' from '0x000d6ffffe2672bd' to 'tele'
Jul 15 14:29:46 atom npm[4866]: #033[32mzigbee2mqtt:info #033[39m 2020-07-15 14:29:46: Successfully unbound cluster 'genOnOff' from '0x000d6ffffe2672bd' to 'tele'
Jul 15 14:29:46 atom npm[4866]: #033[32mzigbee2mqtt:info #033[39m 2020-07-15 14:29:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_unbind","message":{"from":"0x000d6ffffe2672bd","to":"tele","cluster":"genOnOff"}}'

Binding

[34mzigbee2mqtt:debug#033[39m` 2020-07-15 14:25:54: Received MQTT message on 'zigbee2mqtt/bridge/bind/0x000d6ffffe2672bd' with data 'g_2'
[39m 2020-07-15 14:25:54: binding cluster 'genOnOff' from '0x000d6ffffe2672bd' to 'g_2'
[34mzigbee2mqtt:debug#033[39m 2020-07-15 14:25:56: Received Zigbee message from '0x000d6ffffe2672bd', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":0,"ontime":600,"offwaittime":0}' from endpoint 1 with groupID 7787
[39m 2020-07-15 14:25:56: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2672bd', payload '{"occupancy":true,"linkquality":65,"update_available":false}'
[32mzigbee2mqtt:info #033[39m 2020-07-15 14:25:57: Successfully bound cluster 'genOnOff' from '0x000d6ffffe2672bd' to 'g_2'
[32mzigbee2mqtt:info #033[39m 2020-07-15 14:25:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_bind","message":{"from":"0x000d6ffffe2672bd","to":"g_2","cluster":"genOnOff"}}'
[34mzigbee2mqtt:debug#033[39m 2020-07-15 14:26:00: Received Zigbee message from '0x000d6ffffe2672bd', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":0,"ontime":600,"offwaittime":0}' from endpoint 1 with groupID 7787
[32mzigbee2mqtt:info #033[39m 2020-07-15 14:26:00: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe2672bd', payload '{"occupancy":true,"linkquality":63,"update_available":false}'

hesspoint, 2024-04-13:
I had the same issue but eventually managed to unbind from the default_bind_group by clicking on the remote while unbinding. I think it had to do with the remote being in sleep. Had the same issues also when OTA Updating.

But I still have the issue that in my groups I have the default_bind_group and I cannot delete it. Always gives me the error that the group does not exist.

Has anyone managed to remove the group?