Koenkk/zigbee2mqtt #2925: Xiaomi curtain motor: status unknown

BarBaar44, 2020-02-09:
# Bug Report
## What happened
Home Assistant no longer reports the current status of my Xiaomi Curtain motors (ZNCLDJ11LM). I have 2, both have the same problem. As a result, I am no longer able to toggle the curtain motor.

Nothing is reported (status, link quality, etc)

## What did you expect to happen
Have the current status of the motor reported in HA

## How to reproduce it (minimal and precise)
Use Xiaomi ZNCLDJ11LM curtain motors.

Please note: The ZNCLDJ12LM (Aqara B1) is working correctly
## Debug Info
zigbee2mqtt version: 1.10. The issue first occurred in 1.9
CC253X firmware version: latest

ruslanruslanruslan, 2020-02-11:
ZNCLDJ11LM does not work on my side too

Koenkk, 2020-02-11:
Can you debug provide the logs when controlling the curtain? https://www.zigbee2mqtt.io/information/debug.html

BarBaar44, 2020-02-11:
Have not yet found time to put zigbee2mqtt into debug. But here's something I just picked up from the logs:

`zigbee2mqtt:error 2020-02-11 23:06:39: Publish 'set' 'position' to '0x00158d000394c113' failed: 'Error: Write 0x00158d000394c113/1 genAnalogOutput({"85":{"value":99,"type":57}}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 55311 - 1 - 6 - 13 - 4 after 10000ms)'`

ercsey, 2020-02-12:
Kind of same problem here with unknown state of Aqara curtain motor (ZNCLDJ11LM). After a successful pairing the state and position remains unknown.  I'm using zigbee2mqtt-edge under Hass.io.

```zigbee2mqtt:info  2020-02-12 11:41:38: MQTT publish: topic 'homeassistant/cover/0x00158d0003fa4f53/cover/config', payload '{"command_topic":"zigbee2mqtt/Kinga&Szabi szoba függöny/set","position_topic":"zigbee2mqtt/Kinga&Szabi szoba függöny","set_position_topic":"zigbee2mqtt/Kinga&Szabi szoba függöny/set","set_position_template":"{ \"position\": {{ position }} }","value_template":"{{ value_json.position }}","json_attributes_topic":"zigbee2mqtt/Kinga&Szabi szoba függöny","name":"Kinga&Szabi szoba függöny_cover","unique_id":"0x00158d0003fa4f53_cover_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0003fa4f53"],"name":"Kinga&Szabi szoba függöny","sw_version":"Zigbee2mqtt 1.10.0-dev","model":"Aqara curtain motor (ZNCLDJ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'```
```zigbee2mqtt:debug 2020-02-12 11:45:52: Received Zigbee message from 'Kinga&Szabi szoba függöny', type 'attributeReport', cluster 'genBasic', data '{"65281":{"3":30,"5":36,"7":"0x0000000000000000","8":4384,"9":512,"100":100}}' from endpoint 1 with groupID 0``` 
If I trigger the the open service I'm receiving these messages and the state and position becomes available with correct value.:
```zigbee2mqtt:debug 2020-02-12 11:51:55: Received MQTT message on 'zigbee2mqtt/Kinga&Szabi szoba függöny/set' with data 'OPEN'
zigbee2mqtt:debug 2020-02-12 11:51:55: Publishing 'set' 'state' to 'Kinga&Szabi szoba függöny'
zigbee2mqtt:debug 2020-02-12 11:51:55: Received Zigbee message from 'Kinga&Szabi szoba függöny', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440514,"presentValue":100}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-02-12 11:51:55: MQTT publish: topic 'zigbee2mqtt/Kinga&Szabi szoba függöny', payload '{"position":100,"running":true,"linkquality":86,"device":{"friendlyName":"Kinga&Szabi szoba függöny","model":"ZNCLDJ11LM","ieeeAddr":"0x00158d0003fa4f53","networkAddress":19178,"type":"Router","manufacturerID":4447,"manufacturerName":"LUMI","powerSource":"Mains (single phase)","applicationVersion":32,"stackVersion":2,"zclVersion":1,"hardwareVersion":17,"dateCode":"08-13-2018"}}'```

Hope it helps to identify the problem. Thank you.

Koenkk, 2020-02-12:
@BarBaar44 looks that there is a connection problem, can you try repairing it?

BarBaar44, 2020-02-14:
Yes, repairing seems to fixed it for me. However, kind of annoying that I need to repair Xiaomi devices every once and a while. See also https://github.com/Koenkk/zigbee2mqtt/issues/2798

ercsey, 2020-02-14:
At me thevrepairing did not solve the problem. After HA restart the status remains unknown until the cover it's triggered.

Koenkk, 2020-02-15:
@BarBaar44 what coordinator and firmware version are you using (default or source routing)?

@ercsey please provide the complete logging when setting the state of the curtain.

ercsey, 2020-02-16:
This is the part of the log right after HA start. The state of the cover is ```unknown```:
Complete log: [https://pastebin.com/FxCFLfhu](url)

If I do a restart to the zigbee2mqtt addon the state of the cover changes to ```unavailable``` >>> ```unknown``` >>> ```open``` (the actual state of the cover)

When calling service cover.open_cover (or any cover service) the cover changes from ```unknown``` >>> ```open``` (the actual state of the cover)
```
zigbee2mqtt:debug 2020-02-16 09:59:29: Received MQTT message on 'zigbee2mqtt/Szoba 2 függöny/set' with data 'OPEN'
zigbee2mqtt:debug 2020-02-16 09:59:29: Publishing 'set' 'state' to 'Szoba 2 függöny'
zigbee2mqtt:debug 2020-02-16 09:59:29: Received Zigbee message from 'Szoba 2 függöny', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":117440514,"presentValue":100}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-02-16 09:59:29: MQTT publish: topic 'zigbee2mqtt/Szoba 2 függöny', payload '{"position":100,"running":true,"linkquality":107}'
zigbee2mqtt:debug 2020-02-16 09:59:30: Received Zigbee message from 'Szoba 2 függöny', type 'attributeReport', cluster 'genBasic', data '{"65281":{"3":32,"5":37,"7":"0x0000000000000000","8":4384,"9":1026,"100":100}}' from endpoint 1 with groupID 0
```
Without the addon restart or triggering a service call to the cover the state remains unchanged (unknown).

Koenkk, 2020-02-16:
@ercsey so the problem is that HA doesn't have the correct state after a HA restart? In that case make sure you use the Home Assistant configuration from https://www.zigbee2mqtt.io/integration/home_assistant.html#mqtt-discovery  , with the configured zigbee2mqtt will resend all states max 20 seconds after HA restarts.

BarBaar44, 2020-02-19:
> @BarBaar44 what coordinator and firmware version are you using (default or source routing)?
> 
> @ercsey please provide the complete logging when setting the state of the curtain.

Hmm, have to say: The curtain status is now always open. Which obviously is not correct ;)

Anyway.

- Running the latest zigbee2mqtt version: 1.10.1
- Running a CC2530 + CC2591 [router](https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.htmll)
- Running a [CC2531](https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.htmll) connected to HASS
`zigbee2mqtt:info  2020-02-19 22:49:38: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}'`

BarBaar44, 2020-02-20:
And a debug log:

`zigbee2mqtt:debug 2020-02-20 18:10:27: Received MQTT message on 'zigbee2mqtt/0x00158d000394c113/set' with data 'OPEN'
zigbee2mqtt:debug 2020-02-20 18:10:27: Publishing 'set' 'state' to '0x00158d000394c113'
zigbee2mqtt:debug 2020-02-20 18:10:36: Received MQTT message on 'zigbee2mqtt/0x00158d000314188b/set' with data 'CLOSE'
zigbee2mqtt:debug 2020-02-20 18:10:36: Publishing 'set' 'state' to '0x00158d000314188b'
zigbee2mqtt:error 2020-02-20 18:10:37: Publish 'set' 'state' to '0x00158d000394c113' failed: 'Error: Write 0x00158d000394c113/1 genAnalogOutput({"85":{"value":100,"type":57}}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 55311 - 1 - 3 - 13 - 4 after 10000ms)'
zigbee2mqtt:debug 2020-02-20 18:10:37: Error: Write 0x00158d000394c113/1 genAnalogOutput({"85":{"value":100,"type":57}}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 55311 - 1 - 3 - 13 - 4 after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:150:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
zigbee2mqtt:info  2020-02-20 18:10:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00158d000394c113' failed: 'Error: Write 0x00158d000394c113/1 genAnalogOutput({\"85\":{\"value\":100,\"type\":57}}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":true}) failed (Error: Timeout - 55311 - 1 - 3 - 13 - 4 after 10000ms)'","meta":{"friendly_name":"0x00158d000394c113"}}'
zigbee2mqtt:error 2020-02-20 18:10:46: Publish 'set' 'state' to '0x00158d000314188b' failed: 'Error: Write 0x00158d000314188b/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 29453 - 1 - 4 - 13 - 4 after 10000ms)'
zigbee2mqtt:debug 2020-02-20 18:10:46: Error: Write 0x00158d000314188b/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"timeout":10000,"defaultResponseTimeout":15000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 29453 - 1 - 4 - 13 - 4 after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:150:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.10.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
zigbee2mqtt:info  2020-02-20 18:10:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00158d000314188b' failed: 'Error: Write 0x00158d000314188b/1 genAnalogOutput({\"85\":{\"value\":0,\"type\":57}}, {\"timeout\":10000,\"defaultResponseTimeout\":15000,\"manufacturerCode\":null,\"disableDefaultResponse\":true}) failed (Error: Timeout - 29453 - 1 - 4 - 13 - 4 after 10000ms)'","meta":{"friendly_name":"0x00158d000314188b"}}'`

Koenkk, 2020-02-20:
Please try with the latest dev branch. Yesterday I've pushed a fix to rediscover the route on timeout.

BarBaar44, 2020-02-21:
Does seem to fix something. However, the current status of the curtain is still incorrect. In other words: toggeling the curtain doesn't work...

Below the output of a toggle, but nothing happens with the curtain.

`zigbee2mqtt:info  2020-02-21 10:13:20: MQTT publish: topic 'zigbee2mqtt/0x00158d00028287f1', payload '{"linkquality":99,"battery":78,"voltage":2962,"click":"single"}'
zigbee2mqtt:info  2020-02-21 10:13:20: MQTT publish: topic 'zigbee2mqtt/0x00158d00028287f1', payload '{"linkquality":99,"battery":78,"voltage":2962,"click":""}'
zigbee2mqtt:debug 2020-02-21 10:13:20: Received MQTT message on 'zigbee2mqtt/0x00158d000314188b/set' with data 'CLOSE'
zigbee2mqtt:debug 2020-02-21 10:13:20: Publishing 'set' 'state' to '0x00158d000314188b'`

BarBaar44, 2020-02-22:
See this one as well: https://github.com/Koenkk/zigbee2mqtt/issues/2875

BarBaar44, 2020-02-23:
OK.. upgraded to source routing, and repaired all devices. Looks like that fixed the issue for now..

closed 2020-02-23 by BarBaar44