Koenkk/zigbee2mqtt #2927: Router not joining after changing channel 11 -> 15

iz3man, 2020-02-10:
My Zigbee network was working fine.
Wanted to add a KONKE presence sensor and the supported devices page says they only work on channel 15,20,25. So I set channel to 15 which works fine with my wifi setup.
I had to re-pair all sensors (was aware of that), but my CC2591 routers don't reconnect.
What do I need to do? They both are running router-cc2530-cc2591-std.hex from 2019-02.
Thanks!

Ton1965, 2020-02-11:
You probably need to reset your routers. It is done by toggling power three times in short duration.

CodeFinder2, 2020-02-11:
As a workaround, have you tried reflashing them router? I also got difficulties in repairing a CC2530 + CC2591 using the off > 2s on > off cycles. Reflashing causes the router to enter pairing mode at start. (I know this is tedious and shouldn't be 'the solution here'.)

iz3man, 2020-02-11:
No. I wasn't aware of the reset procedure. And it seems that's what i was looking/asking for. So thanks a lot for that. I will try it and let you know!

CodeFinder2, 2020-02-11:
Ah oki. Please close the issue then if it fixes your problem. 😉

iz3man, 2020-02-11:
Thanks a lot. That worked perfectly fine!
Off, On for 2s, Off, On for 2s, Off, On for 2s, Off, On .... Wait ...

closed 2020-02-11 by iz3man

CodeFinder2, 2020-02-11:
By the way, all this is also stated in the docs so you should also take a look there ;-)

https://www.zigbee2mqtt.io/devices/CC2530.ROUTER.html

iz3man, 2020-02-11:
Shame on me :(