Koenkk/zigbee2mqtt #2929: Ikea Trädfri Buttons trigger my Innr SP 120 Smart Plug

Ahrimaan, 2020-02-10:
Hi,

i have a weird problem
I have paired :
TRADFRI Remote Control https://www.zigbee2mqtt.io/devices/E1524_E1810.html
TRADFRI ON/OFF https://www.zigbee2mqtt.io/devices/E1766.html
TRADFRI Motion Sensor 
https://www.zigbee2mqtt.io/devices/E1525_E1745.html

INNR Sp120 Smart Plug: https://www.zigbee2mqtt.io/devices/SP_120.html

I DO NOT RUN Node-RED yet, so the devices are not able to trigger anything but:

When i click the Button on the Remote or the On/Off from Tradfri, my Smart Plug is going on and off.
HOW is that even possible ??

Any Ideas ?

CodeFinder2, 2020-02-10:
You may have (accidentally?) bound your remote to your smart plug. Try re-pairing (possibly keeping the plug unconnected just to be sure 😂).

Zigbee allows direct connections between certain if supported by the vendor / device.

ryanbeaton, 2020-02-11:
HI @Ahrimaan 
It's noted on the device page for the remote:
By default this remote is bound to the default bind group which you first have to unbind it from. This can be done by sending to zigbee2mqtt/bridge/unbind/[DEVICE_FRIENLDY_NAME]] payload default_bind_group.

That's why it will be turning on and off your switch.

Ahrimaan, 2020-02-11:
> You may have (accidentally?) bound your remote to your smart plug. Try re-pairing (possibly keeping the plug unconnected just to be sure 😂).
> 

Oh man how stupid ..... yeah that was the problem, i´ve reseted the Plug and now all works normal.
Thanks !

closed 2020-02-11 by Ahrimaan